dariuszseweryn/RxAndroidBle #599: [Question] Keeping GATT connection open

mattinger, 2019-07-17:
### Summary

I was trying to determine from the API if it's possible to keep a gatt connection open outside of the lifecycle of the establishConnection observable.  Or is the connection closed once there are no more subscribers.  I'm basically trying to wrap things up in kotlin coroutines:

```kotlin
suspend fun RxBleDevice.connectGatt(autoConnect: Boolean): RxBleConnection =
    suspendCancellableCoroutine<RxBleConnection> { continuation ->
        val disposable = establishConnection(autoConnect).subscribe(
            { conn -> continuation.resume(conn) },
            { throwable -> continuation.resumeWithException(throwable) })

        continuation.invokeOnCancellation {
            if (!disposable.isDisposed) {
                disposable.dispose()
            }
        }
    }
```

I'm just unsure if the connection would even be usable once this coroutine resumed.

### Library version
`1.10.1`

#### Preconditions

dariuszseweryn, 2019-07-17:
> This leads me to believe that the connection can stay open as long as the returned disposable is open.

Correct. Just have in mind that this `Disposable` is the only handle to explicitly disconnect your peripheral. If you need to cancel the connection at any point you should return `Pair<RxBleConnection, Disposable>` from the coroutine.

Or even make the `RxBleConnection` a `CoroutineContext`/`CoroutineScope` somehow as if the connection will ever be broken (and that may happen asynchronously at any moment) all further actions on the `RxBleConnection` will fail. I am not familiar enough with coroutines to give a concrete example, I just think it should be feasible.

dariuszseweryn, 2019-07-24:
If there are no more questions I will close this thread.
Feel free to share your solution as it may be helpful for some.

Best regards

closed 2019-07-24 by dariuszseweryn