dariuszseweryn/RxAndroidBle #602: After reconnection, first read operation stalls on "QUEUED ServiceDiscoveryOperation(201690754)"

starkeyvsk, 2019-07-26:
I am running into issues with reconnection.

The first time my app connects to the device, it works just fine. The first read operation  i issue to RxAndroidBLE logs the following to LogCat:

```
2019-07-26 17:08:36.957 D/[(DeviceImpl.kt:248)$getDeviceSide]: Discovering Side of Device
2019-07-26 17:08:37.772 D/RxBle#ConnectionOperationQueue: QUEUED   ServiceDiscoveryOperation(226361603)
2019-07-26 17:08:37.780 D/RxBle#ConnectionOperationQueue: STARTED  ServiceDiscoveryOperation(226361603)
2019-07-26 17:08:37.788 I/RxBle#ConnectionOperationQueue: RUNNING  ServiceDiscoveryOperation{MAC='XX:XX:XX:11:10:B2'}
2019-07-26 17:08:37.796 D/BluetoothGatt: discoverServices() - device: XX:XX:XX:11:10:B2
2019-07-26 17:08:37.819 D/BluetoothGatt: onSearchComplete() = Device=XX:XX:XX:11:10:B2 Status=0
2019-07-26 17:08:37.824 I/RxBle#GattCallback: MAC='XX:XX:XX:11:10:B2'     onServicesDiscovered(), status=0
2019-07-26 17:08:37.831 V/RxBle#ServiceDiscoveryOperation: Preparing services description
2019-07-26 17:08:37.892 V/RxBle#ServiceDiscoveryOperation: --------------- ====== Printing peripheral content ====== ---------------
.
.
... the app proceeds with the connection and works just fine.
```

The library then proceeds to work fine.

The I reboot the bluetooth peripheral device and once RxAndroidBLE finds the device and my initialization process gets its connection state callback telling it to initialize, the same block of code stalls with this:

```
2019-07-26 17:09:27.612 D/[(RxConnection.kt:159)$invoke]: Device XX:XX:XX:11:10:B2 connection state change detected: Connected
2019-07-26 17:09:28.117 D/[(RxConnection.kt:94)$accept]: connectToDevice on next. Request MTU next
2019-07-26 17:09:28.121 D/RxBle#ConnectionOperationQueue: QUEUED   MtuRequestOperation(263262052)
2019-07-26 17:09:28.125 D/RxBle#ConnectionOperationQueue: STARTED  MtuRequestOperation(263262052)
2019-07-26 17:09:28.127 I/RxBle#ConnectionOperationQueue: RUNNING  MtuRequestOperation{MAC='XX:XX:XX:11:10:B2', mtu=153}
2019-07-26 17:09:28.134 D/BluetoothGatt: configureMTU() - device: XX:XX:XX:11:10:B2 mtu: 153
2019-07-26 17:09:28.217 D/BluetoothGatt: onConfigureMTU() - Device=XX:XX:XX:11:10:B2 mtu=153 status=0
2019-07-26 17:09:28.219 I/RxBle#GattCallback: MAC='XX:XX:XX:11:10:B2'             onMtuChanged(), status=0, value=153
2019-07-26 17:09:28.223 D/[(RxConnection.kt:97)$accept]: Returned MTU size: 153
2019-07-26 17:09:28.225 D/[(RxConnection.kt:99)$accept]: Current MTU service: 150
2019-07-26 17:09:28.228 D/[(DeviceImpl.kt:138)$invoke]: 98137447 Connection State Updated XX:XX:XX:11:10:B2: CONNECTED
2019-07-26 17:09:28.229 D/[(DeviceImpl.kt:182)$handleConnected]: Starting connection/initialization process for  XX:XX:XX:11:10:B2
2019-07-26 17:09:28.242 D/RxBle#ConnectionOperationQueue: QUEUED   ServiceDiscoveryOperation(201690754)
2019-07-26 17:09:34.018 D/[(DeviceImpl.kt:248)$getSide]: Discovering Side of Device
.
.
.... and nothing after this.....
```

I don't know why the library stops doing anything after this point, and I am not sure where to even start looking.

dariuszseweryn, 2019-07-29:
Hello @starkeyvsk 

It is hard for me to say what is happening since you have not provided any code. I see no logs of operations that had finished either though they should. I would expect:
```
D/RxBle#ConnectionOperationQueue: FINISHED MtuRequestOperation(239106113) in 29 ms
```
So either an operation stalls or you have edited the logs too aggressively.

dariuszseweryn, 2019-08-02:
Any news @starkeyvsk ?

starkeyvsk, 2019-08-02:
It looks like we managed to stall our communications queue (which is queuing some compound operations) through a combination of blocking calls . We removed those calls and it is working now. Therefore I am not sure if there is anything that this library needs to do differently. 

What's odd is I think you are correct about missing a FINISH call for the MTU operation, but IIRC the callback for onSuccess somehow fired anyway. So im not sure if we actually manages to stall the Libraries internal queue or the queueing mechanism we have on top of it.

We can close this out. Thanks for being so responsive Darius.

dariuszseweryn, 2019-08-05:
Probably threads used for callbacks got locked so no new callback could go through. There is nothing library can do if this happens. Never use blocking calls as threading is hard™. ;)

closed 2019-08-05 by dariuszseweryn