dariuszseweryn/RxAndroidBle #605: Scanned device names are not updated unless scan is restarted

seljabali, 2019-08-23:
### Summary
Nearby ble device names, if changed after a scan is started, is not reflected in scanned results accordingly. When the scan is restarted however, it show its updated name accordingly. 

### Library version
`1.10.0`

#### Preconditions
- Get a device whose name can change on demand.

#### Steps to reproduce actual result
<br />1. Start device with name `foo` & mac address `X`
<br />2. Start a ble scan
<br />3. Note `foo` found amongst devices scanned
<br />4. Change device name name to `bar`

#### Actual result
<br /> Device with mac address `X` is found with name `foo`

#### Expected result
<br /> Device with mac address `X` is found with name `bar`

#### Minimum code snippet reproducing the issue
```
RxBle.get()
	.scanBleDevices(
	    ScanSettings.Builder()
	        .setScanMode(ScanSettings.SCAN_MODE_LOW_LATENCY)
	        .setCallbackType(ScanSettings.CALLBACK_TYPE_ALL_MATCHES)
	        .build()
	)
	.subscribeOn(Schedulers.io())
	.observeOn(AndroidSchedulers.mainThread())
	.subscribe({ bleDevice ->
		// print found device names + mac address
	}, { throwable -> 

	})
```

dariuszseweryn, 2019-08-26:
The data is [taken directly from the OS API](https://github.com/Polidea/RxAndroidBle/blob/c2c4b19c85cf348647f9d6b1522fddbbb8d75cd0/rxandroidble/src/main/java/com/polidea/rxandroidble2/internal/RxBleDeviceImpl.java#L93).

You can prove that the device name is being actually changed in the advertisement data by adding logs as requested by the template. Then it could be technically cached by the library.

dariuszseweryn, 2019-08-27:
Maybe `LeScanCallback` returns a new `BluetoothDevice` object on each advertisement scanned and the library's implementation uses a single/first instance it got.

Alternatively the API 21 scan behaves differently from the legacy one.