dariuszseweryn/RxAndroidBle #608: setCallbackType

houyuehai, 2019-09-06:
Hi！
When I set ScanSettings. CALLBACK_TYPE_ALL_MATCHES, I can find Bluetooth. When I set ScanSettings. CALLBACK_TYPE_FIRST_MATCH, I couldn't find it. And "java.lang.Security Exception: Need BLUETOOTH_PRIVILEGED permission" appears in the log. And I won't have this problem in MI 6x (android 8.1), which is the problem in Huaweihonr 9x (android 9.1).
log：
2019-09-06 16:36:30.964 9391-9444/com.hst.radarflow W/Binder: Caught a RuntimeException from the binder stub implementation.
    java.lang.SecurityException: Need BLUETOOTH_PRIVILEGED permission: Neither user 10129 nor current process has android.permission.BLUETOOTH_PRIVILEGED.
        at android.os.Parcel.createException(Parcel.java:1953)
        at android.os.Parcel.readException(Parcel.java:1921)
        at android.os.Parcel.readException(Parcel.java:1871)
        at android.bluetooth.IBluetoothGatt$Stub$Proxy.startScan(IBluetoothGatt.java:1024)
        at android.bluetooth.le.BluetoothLeScanner$BleScanCallbackWrapper.onScannerRegistered(BluetoothLeScanner.java:518)
        at android.bluetooth.le.IScannerCallback$Stub.onTransact(IScannerCallback.java:57)
        at android.os.Binder.execTransact(Binder.java:739)

houyuehai, 2019-09-06:
ScanFilter scanFilter = new ScanFilter.Builder().setServiceUuid(ParcelUuid.fromString("0000b350-d6d8-c7ec-bdf0-eab1bfc6bcbc")).build();

houyuehai, 2019-09-06:
.flatMap(aBoolean -> rxBleClient.scanBleDevices(new ScanSettings.Builder()//开始le蓝牙扫描
                            .setScanMode(ScanSettings.SCAN_MODE_LOW_LATENCY)
                            .build(),scanFilter))
                    .take(16000, TimeUnit.MILLISECONDS)//扫描时间
                    .toList()

dariuszseweryn, 2019-09-13:
This looks like a model specific bug. As far as I know the library cannot do anything to prevent this situation. Feel free to comment/reopen if you think otherwise

Best regards

closed 2019-09-13 by dariuszseweryn

MBach, 2021-05-05:
Bumping this thread!
With Huawei P30, I have the exact same problem when I set  `callbackType`

My working code on Samsung S10:
```
const scanOptions: ScanOptions = { scanMode: ScanMode.LowLatency, callbackType: ScanCallbackType.FirstMatch }
```
but NOT on Huawei P30.

So, I have made a little modification by adding `react-native-device-info` package:

```
import { getManufacturer } from 'react-native-device-info'
// ...
    let scanOptions: ScanOptions = { scanMode: ScanMode.LowLatency }
    const m = await getManufacturer()
    if (m !== 'HUAWEI') {
      scanOptions.callbackType = ScanCallbackType.FirstMatch
    }
```
Yeah, kind of ugly, right?

MBach, 2021-05-05:
I mean, I have this problem in the other repository https://github.com/Polidea/react-native-ble-plx