dariuszseweryn/RxAndroidBle #610: Not able to combine discoverServices() and writeCharacteristic() at once

manishpatelgt, 2019-09-16:
I am trying to combine  discoverServices() and  writeCharacteristic() at once but not able to do this.

I am using below

`implementation "com.polidea.rxandroidble2:rxandroidble:1.10.1"`

```
final Disposable disposable =  bleDevice
                            .establishConnection(false)
                            .flatMap(connection ->
                                    Observable.combineLatest(
                                            connection.writeCharacteristic(Consts.ALERT_LEVEL_UUID, Utils.getTestONInputBytes()).doOnSuccess(bytes -> onWriteSuccess()),
                                            connection.discoverServices().doOnSuccess(services -> swapScanResult(services)),
                                            Pair::new))
                            .observeOn(AndroidSchedulers.mainThread())
                            .subscribe();

                    compositeDisposable.add(disposable);
```

I am getting compile time error like

> error: no suitable method found for zip(Single<byte[]>,Single<RxBleDeviceServices>,Pair::new)
> method Observable.<T#1,R#1>zip(Iterable<? extends ObservableSource<? extends T#1>>,Function<? super Object[],? extends R#1>) is not applicable
> (cannot infer type-variable(s) T#1,R#1
> (actual and formal argument lists differ in length))
> method Observable.<T#2,R#2>zip(ObservableSource<? extends ObservableSource<? extends T#2>>,Function<? super Object[],? extends R#2>) is not applicable
> (cannot infer type-variable(s) T#2,R#2
> (actual and formal argument lists differ in length))
> method Observable.<T1#1,T2#1,R#3>zip(ObservableSource<? extends T1#1>,ObservableSource<? extends T2#1>,BiFunction<? super T1#1,? super T2#1,? extends R#3>) is not applicable
> (cannot infer type-variable(s) T1#1,T2#1,R#3
> (argument mismatch; Single<byte[]> cannot be converted to ObservableSource<? extends T1#1>))
> method Observable.<T1#2,T2#2,R#4>zip(ObservableSource<? extends T1#2>,ObservableSource<? extends T2#2>,BiFunction<? super T1#2,? super T2#2,? extends R#4>,boolean) is not applicable
> (cannot infer type-variable(s) T1#2,T2#2,R#4
> (actual and formal argument lists differ in length))
> method Observable.<T1#3,T2#3,R#5>zip(ObservableSource<? extends T1#3>,ObservableSource<? extends T2#3>,BiFunction<? super T1#3,? super T2#3,? extends R#5>,boolean,int) is not applicable
> (cannot infer type-variable(s) T1#3,T2#3,R#5
> (actual and formal argument lists differ in length))
> method Observable.<T1#4,T2#4,T3#1,R#6>zip(ObservableSource<? extends T1#4>,ObservableSource<? extends T2#4>,ObservableSource<? extends T3#1>,Function3<? super T1#4,? super T2#4,? super T3#1,? extends R#6>) is not applicable
> (cannot infer type-variable(s) T1#4,T2#4,T3#1,R#6
> (actual and formal argument lists differ in length))
> method Observable.<T1#5,T2#5,T3#2,T4#1,R#7>zip(ObservableSource<? extends T1#5>,ObservableSource<? extends T2#5>,ObservableSource<? extends T3#2>,ObservableSource<? extends T4#1>,Function4<? super T1#5,? super T2#5,? super T3#2,? super T4#1,? extends R#7>) is not applicable
> (cannot infer type-variable(s) T1#5,T2#5,T3#2,T4#1,R#7
> (actual and formal argument lists differ in length))
> method Observable.<T1#6,T2#6,T3#3,T4#2,T5#1,R#8>zip(ObservableSource<? extends T1#6>,ObservableSource<? extends T2#6>,ObservableSource<? extends T3#3>,ObservableSource<? extends T4#2>,ObservableSource<? extends T5#1>,Function5<? super T1#6,? super T2#6,? super T3#3,? super T4#2,? super T5#1,? extends R#8>) is not applicable
> (cannot infer type-variable(s) T1#6,T2#6,T3#3,T4#2,T5#1,R#8
> (actual and formal argument lists differ in length))
> method Observable.<T1#7,T2#7,T3#4,T4#3,T5#2,T6#1,R#9>zip(ObservableSource<? extends T1#7>,ObservableSource<? extends T2#7>,ObservableSource<? extends T3#4>,ObservableSource<? extends T4#3>,ObservableSource<? extends T5#2>,ObservableSource<? extends T6#1>,Function6<? super T1#7,? super T2#7,? super T3#4,? super T4#3,? super T5#2,? super T6#1,? extends R#9>) is not applicable
> (cannot infer type-variable(s) T1#7,T2#7,T3#4,T4#3,T5#2,T6#1,R#9
> (actual and formal argument lists differ in length))
> method Observable.<T1#8,T2#8,T3#5,T4#4,T5#3,T6#2,T7#1,R#10>zip(ObservableSource<? extends T1#8>,ObservableSource<? extends T2#8>,ObservableSource<? extends T3#5>,ObservableSource<? extends T4#4>,ObservableSource<? extends T5#3>,ObservableSource<? extends T6#2>,ObservableSource<? extends T7#1>,Function7<? super T1#8,? super T2#8,? super T3#5,? super T4#4,? super T5#3,? super T6#2,? super T7#1,? extends R#10>) is not applicable
> (cannot infer type-variable(s) T1#8,T2#8,T3#5,T4#4,T5#3,T6#2,T7#1,R#10
> (actual and formal argument lists differ in length))
> method Observable.<T1#9,T2#9,T3#6,T4#5,T5#4,T6#3,T7#2,T8#1,R#11>zip(ObservableSource<? extends T1#9>,ObservableSource<? extends T2#9>,ObservableSource<? extends T3#6>,ObservableSource<? extends T4#5>,ObservableSource<? extends T5#4>,ObservableSource<? extends T6#3>,ObservableSource<? extends T7#2>,ObservableSource<? extends T8#1>,Function8<? super T1#9,? super T2#9,? super T3#6,? super T4#5,? super T5#4,? super T6#3,? super T7#2,? super T8#1,? extends R#11>) is not applicable
> (cannot infer type-variable(s) T1#9,T2#9,T3#6,T4#5,T5#4,T6#3,T7#2,T8#1,R#11
> (actual and formal argument lists differ in length))
> method Observable.<T1#10,T2#10,T3#7,T4#6,T5#5,T6#4,T7#3,T8#2,T9,R#12>zip(ObservableSource<? extends T1#10>,ObservableSource<? extends T2#10>,ObservableSource<? extends T3#7>,ObservableSource<? extends T4#6>,ObservableSource<? extends T5#5>,ObservableSource<? extends T6#4>,ObservableSource<? extends T7#3>,ObservableSource<? extends T8#2>,ObservableSource<? extends T9>,Function9<? super T1#10,? super T2#10,? super T3#7,? super T4#6,? super T5#5,? super T6#4,? super T7#3,? super T8#2,? super T9,? extends R#12>) is not applicable
> (cannot infer type-variable(s) T1#10,T2#10,T3#7,T4#6,T5#5,T6#4,T7#3,T8#2,T9,R#12
> (actual and formal argument lists differ in length))
> where T#1,R#1,T#2,R#2,T1#1,T2#1,R#3,T1#2,T2#2,R#4,T1#3,T2#3,R#5,T1#4,T2#4,T3#1,R#6,T1#5,T2#5,T3#2,T4#1,R#7,T1#6,T2#6,T3#3,T4#2,T5#1,R#8,T1#7,T2#7,T3#4,T4#3,T5#2,T6#1,R#9,T1#8,T2#8,T3#5,T4#4,T5#3,T6#2,T7#1,R#10,T1#9,T2#9,T3#6,T4#5,T5#4,T6#3,T7#2,T8#1,R#11,T1#10,T2#10,T3#7,T4#6,T5#5,T6#4,T7#3,T8#2,T9,R#12 are type-variables:
> T#1 extends Object declared in method <T#1,R#1>zip(Iterable<? extends ObservableSource<? extends T#1>>,Function<? super Object[],? extends R#1>)
> R#1 extends Object declared in method <T#1,R#1>zip(Iterable<? extends ObservableSource<? extends T#1>>,Function<? super Object[],? extends R#1>)
> T#2 extends Object declared in method <T#2,R#2>zip(ObservableSource<? extends ObservableSource<? extends T#2>>,Function<? super Object[],? extends R#2>)
> R#2 extends Object declared in method <T#2,R#2>zip(ObservableSource<? extends ObservableSource<? extends T#2>>,Function<? super Object[],? extends R#2>)
> T1#1 extends Object declared in method <T1#1,T2#1,R#3>zip(ObservableSource<? extends T1#1>,ObservableSource<? extends T2#1>,BiFunction<? super T1#1,? super T2#1,? extends R#3>)
> T2#1 extends Object declared in method <T1#1,T2#1,R#3>zip(ObservableSource<? extends T1#1>,ObservableSource<? extends T2#1>,BiFunction<? super T1#1,? super T2#1,? extends R#3>)
> R#3 extends Object declared in method <T1#1,T2#1,R#3>zip(ObservableSource<? extends T1#1>,ObservableSource<? extends T2#1>,BiFunction<? super T1#1,? super T2#1,? extends R#3>)
> T1#2 extends Object declared in method <T1#2,T2#2,R#4>zip(ObservableSource<? extends T1#2>,ObservableSource<? extends T2#2>,BiFunction<? super T1#2,? super T2#2,? extends R#4>,boolean)
> T2#2 extends Object declared in method <T1#2,T2#2,R#4>zip(ObservableSource<? extends T1#2>,ObservableSource<? extends T2#2>,BiFunction<? super T1#2,? super T2#2,? extends R#4>,boolean)
> R#4 extends Object declared in method <T1#2,T2#2,R#4>zip(ObservableSource<? extends T1#2>,ObservableSource<? extends T2#2>,BiFunction<? super T1#2,? super T2#2,? extends R#4>,boolean)
[... 27 lines cut ...]
> T3#5 extends Object declared in method <T1#8,T2#8,T3#5,T4#4,T5#3,T6#2,T7#1,R#10>zip(ObservableSource<? extends T1#8>,ObservableSource<? extends T2#8>,ObservableSource<? extends T3#5>,ObservableSource<? extends T4#4>,ObservableSource<? extends T5#3>,ObservableSource<? extends T6#2>,ObservableSource<? extends T7#1>,Function7<? super T1#8,? super T2#8,? super T3#5,? super T4#4,? super T5#3,? super T6#2,? super T7#1,? extends R#10>)
[... cut ...]
> R#12 extends Object declared in method <T1#10,T2#10,T3#7,T4#6,T5#5,T6#4,T7#3,T8#2,T9,R#12>zip(ObservableSource<? extends T1#10>,ObservableSource<? extends T2#10>,ObservableSource<? extends T3#7>,ObservableSource<? extends T4#6>,ObservableSource<? extends T5#5>,ObservableSource<? extends T6#4>,ObservableSource<? extends T7#3>,ObservableSource<? extends T8#2>,ObservableSource<? extends T9>,Function9<? super T1#10,? super T2#10,? super T3#7,? super T4#6,? super T5#5,? super T6#4,? super T7#3,? super T8#2,? super T9,? extends R#12>)

dariuszseweryn, 2019-09-16:
Hello!

Thank you for using the library.

This is a general usage question and as such should be posted on [stackoverflow](https://stackoverflow.com/questions/tagged/rxandroidble) with `rxandroidble` tag—it will be easier to find by other users. Also feel free to check already posted questions as a similar one may be already answered.

Best Regards

closed 2019-09-16 by dariuszseweryn

manishpatelgt, 2019-09-16:
@dariuszseweryn Already posted in [stackoverflow ](https://stackoverflow.com/questions/57945944/how-do-i-connect-multiple-characteristics-at-the-same-time)