dariuszseweryn/RxAndroidBle #611: NotifyClick not working

manishpatelgt, 2019-09-18:
I am trying below code from sample app.

 ```
if (isConnected()) {
            final Disposable disposable = connectionObservable
                    .flatMap(rxBleConnection -> rxBleConnection.setupNotification(characteristicUuid, NotificationSetupMode.QUICK_SETUP))
                    .doOnNext(notificationObservable -> runOnUiThread(this::notificationHasBeenSetUp))
                    .flatMap(notificationObservable -> notificationObservable)
                    .retryWhen(errors -> errors.flatMap(error -> {
                        if (error instanceof BleDisconnectedException) {
                            Log.d("Retry", "Retrying");
                            return Observable.just(null);
                        }
                        return Observable.error(error);
                    }))
                    .observeOn(AndroidSchedulers.mainThread())
                    .subscribe(this::onNotificationReceived, this::onNotificationSetupFailure);

            compositeDisposable.add(disposable);
```

As soon as press Register Notifications button I am always getting error like below

[image]
[image]

How to resolve this ? How do I enable notification for that characteristicUuid ?

dariuszseweryn, 2019-09-18:
I do not know — you did not include logs as issue template asks you to

manishpatelgt, 2019-09-18:
@dariuszseweryn  I am adding logs in 10 min

manishpatelgt, 2019-09-18:
@dariuszseweryn 
Logs 

> t: setCharacteristicNotification() - uuid: 0000ffe1-0000-1000-8000-00805f9b34fb enable: true
> 2019-09-18 14:49:43.371 15811-15811/com.polidea.rxandroidble.sample D/BluetoothGatt: setCharacteristicNotification() - uuid: 0000ffe1-0000-1000-8000-00805f9b34fb enable: false
> 2019-09-18 14:49:43.377 15811-15811/com.polidea.rxandroidble.sample I/ViewRootImpl: jank_removeInvalidNode all the node in jank list is out of time
> 2019-09-18 14:49:43.382 15811-15811/com.polidea.rxandroidble.sample I/System.out: Notifications error: com.polidea.rxandroidble2.exceptions.BleCannotSetCharacteristicNotificationException: Cannot find client characteristic config descriptor (code 2) with characteristic UUID 0000ffe1-0000-1000-8000-00805f9b34fb

dariuszseweryn, 2019-09-18:
`com.polidea.rxandroidble2.exceptions.BleCannotSetCharacteristicNotificationException: Cannot find client characteristic config descriptor (code 2) with characteristic UUID 0000ffe1-0000-1000-8000-00805f9b34fb`
As you can see your peripheral does not conform to the Bluetooth Core Specification. It has no `Client Characteristic Configuration Descriptor`. There are specific modes for setting notifications in such cases. See `NotificationSetupMode`.

manishpatelgt, 2019-09-18:
@dariuszseweryn  How to do this with this sample ?

dariuszseweryn, 2019-09-18:
Change `RxBleConnection.setupNotification()` call to use a proper `NotificationSetupMode`. Or preferably fix the peripheral so it will include `Client Characteristic Configuration Descriptor` according to the Bluetooth Spec

manishpatelgt, 2019-09-18:
@dariuszseweryn  Worked well now. Before I am using **NotificationSetupMode.QUICK_SETUP** and after updating to **NotificationSetupMode.COMPAT**.

Thanks for your little help

closed 2019-09-18 by dariuszseweryn

guodi1014, 2025-02-21:
I have also encountered the same problem, which is indeed an effective solution. It seems that not all device manufacturers will configure according to standards.  Thank you !