dariuszseweryn/RxAndroidBle #614: setServiceUuid not used by library on android 5

balakzde, 2019-09-18:
### Summary
If I use setServiceUuid, I can see all the devices in logcat not the one I specified using uuid. 

This only happens on some devices (for me it is LG G3 with android 5.0). I know there is known issue (https://stackoverflow.com/questions/34065210/android-ble-device-scan-with-filter-is-not-working/34092300)
but this device is not the case. If I use android api to do the same, it does not happen.
```kotlin
val bluetoothManager = context.getSystemService(Context.BLUETOOTH_SERVICE) as BluetoothManager
val bluetoothAdapter = bluetoothManager.adapter
val scanner = bluetoothAdapter!!.bluetoothLeScanner
val filters = ArrayList<ScanFilter>()
filters.add(ScanFilter.Builder().setServiceUuid(ParcelUuid.fromString(SERVICE_UUID)).build())
val scanMode = ScanSettings.SCAN_MODE_BALANCED
scanner!!.startScan(filters, ScanSettings.Builder().setScanMode(scanMode).build(), scanCallback)
```
So I guess my question is, are you doing filtering of service uuid inside your library for android 5 on purpose, becouse it is the only way you can guarantee it will work on all devices or is this a bug?

In addition do you know if it has any impact on performance? It seems like a lot to handle 50 results every second.
### Library version
`1.10.1`

#### Minimum code snippet reproducing the issue
```kotlin
rxBleClient.scanBleDevices(
ScanSettings.Builder().setScanMode(scanMode).setCallbackType(ScanSettings.CALLBACK_TYPE_ALL_MATCHES).build(),
ScanFilter.Builder().setServiceUuid(ParcelUuid.fromString(Constants.SERVICE_UUID)).build()
)
```

#### Logs from the application:
```
09-18 17:24:36.997 20482-20498/cz.quanti.helios_nfc D/BluetoothLeScanner: onScanResult() - ScanResult{mDevice=79:87:A3:67:A2:FD, mScanRecord=ScanRecord [mAdvertiseFlags=26, mServiceUuids=null, mManufacturerSpecificData={76=[16, 5, 19, 28, 122, -15, -52]}, mServiceData={}, mTxPowerLevel=-2147483648, mDeviceName=null], mRssi=-86, mTimestampNanos=94186089737647}
09-18 17:24:37.005 20482-20643/cz.quanti.helios_nfc D/BluetoothLeScanner: onScanResult() - ScanResult{mDevice=50:62:A9:D4:C9:CA, mScanRecord=ScanRecord [mAdvertiseFlags=26, mServiceUuids=null, mManufacturerSpecificData={76=[16, 5, 3, 28, 7, 3, -100]}, mServiceData={}, mTxPowerLevel=-2147483648, mDeviceName=null], mRssi=-75, mTimestampNanos=94186098017283}
09-18 17:24:37.010 20482-20498/cz.quanti.helios_nfc D/BluetoothLeScanner: onScanResult() - ScanResult{mDevice=F0:18:98:A7:38:14, mScanRecord=ScanRecord [mAdvertiseFlags=6, mServiceUuids=null, mManufacturerSpecificData={76=[16, 5, 11, 28, -92, 67, -47]}, mServiceData={}, mTxPowerLevel=-2147483648, mDeviceName=null], mRssi=-77, mTimestampNanos=94186102849782}
09-18 17:24:37.048 20482-20643/cz.quanti.helios_nfc D/BluetoothLeScanner: onScanResult() - ScanResult{mDevice=65:BF:67:DA:95:4C, mScanRecord=ScanRecord [mAdvertiseFlags=26, mServiceUuids=null, mManufacturerSpecificData={76=[16, 5, 1, -104, -59, -72, -67]}, mServiceData={}, mTxPowerLevel=-2147483648, mDeviceName=null], mRssi=-80, mTimestampNanos=94186140194105}
09-18 17:24:37.061 20482-20498/cz.quanti.helios_nfc D/BluetoothLeScanner: onScanResult() - ScanResult{mDevice=90:FD:9F:43:E7:1D, mScanRecord=ScanRecord [mAdvertiseFlags=6, mServiceUuids=[a54e8025-b815-41ba-bf3f-cbf7d06de5b4], mManufacturerSpecificData={65535=[25, 4, 68, 97]}, mServiceData={}, mTxPowerLevel=-2147483648, mDeviceName=2N Access Unit 2.0], mRssi=-77, mTimestampNanos=94186153436761}
09-18 17:24:37.072 20482-20499/cz.quanti.helios_nfc D/BluetoothLeScanner: onScanResult() - ScanResult{mDevice=68:64:4B:04:5F:4B, mScanRecord=ScanRecord [mAdvertiseFlags=26, mServiceUuids=null, mManufacturerSpecificData={76=[9, 6, 3, 115, -64, -88, 84, 13]}, mServiceData={}, mTxPowerLevel=-2147483648, mDeviceName=null], mRssi=-63, mTimestampNanos=94186165239782}
09-18 17:24:37.077 20482-20498/cz.quanti.helios_nfc D/BluetoothLeScanner: onScanResult() - ScanResult{mDevice=5F:F1:59:BC:63:DC, mScanRecord=ScanRecord [mAdvertiseFlags=-1, mServiceUuids=null, mManufacturerSpecificData={76=[7, 25, 1, 6, 32, 0, -11, 15, 6, 1, 0, -50, -32, -100, 92, 72, -30, 126, -48, -49, -8, 82, 24, 30, 107, 68, 98]}, mServiceData={}, mTxPowerLevel=-2147483648, mDeviceName=null], mRssi=-75, mTimestampNanos=94186169709887}
09-18 17:24:37.091 20482-20499/cz.quanti.helios_nfc D/BluetoothLeScanner: onScanResult() - ScanResult{mDevice=47:C0:6A:91:F6:1E, mScanRecord=ScanRecord [mAdvertiseFlags=26, mServiceUuids=null, mManufacturerSpecificData={}, mServiceData={}, mTxPowerLevel=-2147483648, mDeviceName=null], mRssi=-72, mTimestampNanos=94186183499991}
09-18 17:24:37.162 20482-20499/cz.quanti.helios_nfc D/BluetoothLeScanner: onScanResult() - ScanResult{mDevice=76:AD:70:D2:1A:5F, mScanRecord=ScanRecord [mAdvertiseFlags=6, mServiceUuids=null, mManufacturerSpecificData={76=[12, 14, 0, -53, -59, -4, -107, 24, 76, 68, -106, -52, -109, -6, 3, 109]}, mServiceData={}, mTxPowerLevel=-2147483648, mDeviceName=null], mRssi=-81, mTimestampNanos=94186255074053}
09-18 17:24:37.168 20482-20499/cz.quanti.helios_nfc D/BluetoothLeScanner: onScanResult() - ScanResult{mDevice=90:FD:9F:43:E7:1D, mScanRecord=ScanRecord [mAdvertiseFlags=6, mServiceUuids=[a54e8025-b815-41ba-bf3f-cbf7d06de5b4], mManufacturerSpecificData={65535=[25, 4, 68, 97]}, mServiceData={}, mTxPowerLevel=-2147483648, mDeviceName=2N Access Unit 2.0], mRssi=-80, mTimestampNanos=94186261099314}
09-18 17:24:37.181 20482-20498/cz.quanti.helios_nfc D/BluetoothLeScanner: onScanResult() - ScanResult{mDevice=08:8C:1E:21:12:A2, mScanRecord=ScanRecord [mAdvertiseFlags=-1, mServiceUuids=null, mManufacturerSpecificData={6=[1, 9, 32, 2, -116, -124, 103, 112, -102, 86, 56, 6, 29, 44, -52, -90, -103, -16, 28, 58, -96, 4, -84, -111, -123, 126, 46]}, mServiceData={}, mTxPowerLevel=-2147483648, mDeviceName=null], mRssi=-62, mTimestampNanos=94186274415251}
09-18 17:24:37.199 20482-20499/cz.quanti.helios_nfc D/BluetoothLeScanner: onScanResult() - ScanResult{mDevice=60:03:08:A8:29:81, mScanRecord=ScanRecord [mAdvertiseFlags=6, mServiceUuids=null, mManufacturerSpecificData={76=[16, 5, 11, 28, -124, 7, -97]}, mServiceData={}, mTxPowerLevel=-2147483648, mDeviceName=null], mRssi=-74, mTimestampNanos=94186291472491}
09-18 17:24:37.214 20482-20643/cz.quanti.helios_nfc D/BluetoothLeScanner: onScanResult() - ScanResult{mDevice=70:8D:4C:21:3E:6B, mScanRecord=ScanRecord [mAdvertiseFlags=26, mServiceUuids=null, mManufacturerSpecificData={76=[16, 5, 3, 28, 15, 38, -66]}, mServiceData={}, mTxPowerLevel=-2147483648, mDeviceName=null], mRssi=-77, mTimestampNanos=94186306638481}
09-18 17:24:37.249 20482-20499/cz.quanti.helios_nfc D/BluetoothLeScanner: onScanResult() - ScanResult{mDevice=41:55:A9:CF:88:1A, mScanRecord=ScanRecord [mAdvertiseFlags=26, mServiceUuids=null, mManufacturerSpecificData={76=[16, 5, 26, 28, 5, -120, 89]}, mServiceData={}, mTxPowerLevel=-2147483648, mDeviceName=null], mRssi=-82, mTimestampNanos=94186342172907}
09-18 17:24:37.263 20482-20643/cz.quanti.helios_nfc D/BluetoothLeScanner: onScanResult() - ScanResult{mDevice=5F:F1:59:BC:63:DC, mScanRecord=ScanRecord [mAdvertiseFlags=-1, mServiceUuids=null, mManufacturerSpecificData={76=[7, 25, 1, 6, 32, 0, -11, 15, 6, 1, 0, -50, -32, -100, 92, 72, -30, 126, -48, -49, -8, 82, 24, 30, 107, 68, 98]}, mServiceData={}, mTxPowerLevel=-2147483648, mDeviceName=null], mRssi=-79, mTimestampNanos=94186356704158}
09-18 17:24:37.287 20482-20643/cz.quanti.helios_nfc D/BluetoothLeScanner: onScanResult() - ScanResult{mDevice=5C:63:85:AF:50:AE, mScanRecord=ScanRecord [mAdvertiseFlags=-1, mServiceUuids=null, mManufacturerSpecificData={6=[1, 9, 32, 2, 13, -82, -47, -5, -11, 52, 71, 122, -114, -113, -81, 10, 17, 43, -7, 18, 42, 99, 109, -43, -51, -35, -61]}, mServiceData={}, mTxPowerLevel=-2147483648, mDeviceName=null], mRssi=-90, mTimestampNanos=94186379591501}
09-18 17:24:37.302 20482-20498/cz.quanti.helios_nfc D/BluetoothLeScanner: onScanResult() - ScanResult{mDevice=E8:4D:95:B7:AA:81, mScanRecord=ScanRecord [mAdvertiseFlags=6, mServiceUuids=null, mManufacturerSpecificData={456=[2, -110, 68, -116, 30, 14, 108, 0, -86, -48, 14, -109, 0, 103, 46, -106, -113, 19, 67, 81, -79, -128, -55, 118]}, mServiceData={}, mTxPowerLevel=-2147483648, mDeviceName=null], mRssi=-92, mTimestampNanos=94186394779574}
09-18 17:24:37.347 20482-20643/cz.quanti.helios_nfc D/BluetoothLeScanner: onScanResult() - ScanResult{mDevice=76:AD:70:D2:1A:5F, mScanRecord=ScanRecord [mAdvertiseFlags=6, mServiceUuids=null, mManufacturerSpecificData={76=[12, 14, 0, -53, -59, -4, -107, 24, 76, 68, -106, -52, -1 [... cut ...] ], mRssi=-72, mTimestampNanos=94187086334834}
09-18 17:24:38.019 20482-20498/cz.quanti.helios_nfc D/BluetoothLeScanner: onScanResult() - ScanResult{mDevice=5C:63:85:AF:50:AE, mScanRecord=ScanRecord [mAdvertiseFlags=-1, mServiceUuids=null, mManufacturerSpecificData={6=[1, 9, 32, 2, 13, -82, -47, -5, -11, 52, 71, 122, -114, -113, -81, 10, 17, 43, -7, 18, 42, 99, 109, -43, -51, -35, -61]}, mServiceData={}, mTxPowerLevel=-2147483648, mDeviceName=null], mRssi=-89, mTimestampNanos=94187106580094}

```

dariuszseweryn, 2019-09-18:
> are you doing filtering of service uuid inside your library for android 5 on purpose, becouse it is the only way you can guarantee it will work on all devices or is this a bug?

Yes, on 21 <= API < 23 all filtering is emulated in software (see `ScanSetupBuilderImplApi21.java`). If I remember correctly on Lollipop offloaded filtering did not always worked fine and the aim of the library is to get as seamless experience as possible without adding any blacklists/whitelists.

> In addition do you know if it has any impact on performance?

It definitely has an impact on performance. I do not think it is significant though.

balakzde, 2019-09-19:
Fair enough, thanks!

closed 2019-09-19 by balakzde