dariuszseweryn/RxAndroidBle #616: Make Client Characteristic Config configurable

AdrianGlasnapp, 2019-09-19:
Currently NotificationAndIndicationManager uses hardcoded value for Client Characteristic Config (**00002902-0000-1000-8000-00805f9b34fb**). Some devices expose custom descriptors for notifications and because of hardcoded value there is no possibility to register for such notifications. 

Is it reasonable to pass custom descriptor to `setupNotification`?
`
setupNotification(uuid, customDescriptor)
`

dariuszseweryn, 2019-09-19:
I have not seen such a case — could you give more details?

mkonkel, 2019-09-19:
Got the same situation on some custom BT device.

mkonkel, 2019-09-20:
[image]

dariuszseweryn, 2019-09-20:
Ok, so there is a custom descriptor I get it — but what is the reasoning of having such?

mkonkel, 2019-09-20:
dunno, manufacturer decision, I know it's strange but I can't change it.

dariuszseweryn, 2019-09-20:
Answering the question:
> Is it reasonable to pass custom descriptor to `setupNotification`?

No, it is not reasonable to do so. It is against the Core Bluetooth Specification.

Adding such customisation would most probably also need specifying custom on/off data or other behaviours. Afterwards this custom solution would need to be maintained or extended for other against-the-standard solutions.

Current library API allows for creating a flow that would fit your needs — I do not see a reason for introducing new API that promotes breaking specification.

Best Regards

closed 2019-09-20 by dariuszseweryn