dariuszseweryn/RxAndroidBle #617: Module using RxAndroidBle. Want to make it soif the app is backgrounded the connection stays alive.

sai350, 2019-09-19:
I have a module that uses RxAndroidBle. I want to make it so that if the app is backgrounded the connection stays alive. I am thinking about using a foreground service to handle the connection and that your module would just request the connection from the service. If it already exists, then the old one would be returned. If it were the first time, app start up, then a new one would be created. Is it how you would envision it and if so, what's the best way to pass a connection between the service and the module.

RobLewis, 2019-09-19:
I have this exact scenario using a Wi-Fi connection and it works fine. Took me a while to figure out the foreground service and its need for a notification, but eventually got it working. 
Just take care to dispose any streams the app uses when leaving it. Any streams the service uses can stick around until you shut it down (my service queries the peripheral device for updates every few seconds).

RobLewis, 2019-09-19:
P.S. I wrote some code to simplify using Android Services. You might find it helpful: https://gist.github.com/RobLewis/fe7fad0381b761bae2842dc0d2cd8f2c

sai350, 2019-09-19:
Thank You Rob

dariuszseweryn, 2019-10-25:
I have created a Wiki page for background operation — https://github.com/Polidea/RxAndroidBle/wiki/Tutorial:-Background-operation

closed 2019-10-25 by dariuszseweryn