dariuszseweryn/RxAndroidBle #618: com.polidea.rxandroidble2.exceptions.BleDisconnectedException: Disconnected from MAC='XX.XX.XX.XX' with status 8 (GATT_INSUF_AUTHORIZATION or

denghewenai, 2019-09-20:
2019-09-20 17:08:18.949 4789-4823/D/RxBle#ConnectionOperationQueue: Connection operations queue to be terminated (MAC='C4:A6:E6:DD:31:4F')
    com.polidea.rxandroidble2.exceptions.BleDisconnectedException: Disconnected from MAC='XX.XX.XX.XXF' with status 8 (GATT_INSUF_AUTHORIZATION or GATT_CONN_TIMEOUT)
        at com.polidea.rxandroidble2.internal.connection.RxBleGattCallback$2.onConnectionStateChange(RxBleGattCallback.java:80)
        at android.bluetooth.BluetoothGatt$1$4.run(BluetoothGatt.java:262)
        at android.bluetooth.BluetoothGatt.runOrQueueCallback(BluetoothGatt.java:770)
        at android.bluetooth.BluetoothGatt.access$200(BluetoothGatt.java:39)
        at android.bluetooth.BluetoothGatt$1.onClientConnectionState(BluetoothGatt.java:257)
        at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:71)
        at android.os.Binder.execTransact(Binder.java:756)

The exception occurred after sending an instruction.

dariuszseweryn, 2019-09-20:
Hello!

Thank you for using the library.

This is a general usage question and as such should be posted on [stackoverflow](https://stackoverflow.com/questions/tagged/rxandroidble) with `rxandroidble` tag—it will be easier to find by other users. Also feel free to check already posted questions as a similar one may be already answered.

Best Regards

P.S. Most probably your peripheral has crashed and reset — that is why it times out

closed 2019-09-20 by dariuszseweryn

denghewenai, 2019-09-20:
OK,Thanks for your reply.

magicleon94, 2020-12-04:
Hi @denghewenai 
Have you asked on SO? Can you link me the question?

denghewenai, 2020-12-04:
> Hi @denghewenai
> Have you asked on SO? Can you link me the question?

At that time, this question was not raised on Stackoverflow, because it was found that the use of nrf Connect would also be disconnected. This question was subsequently modified by a colleague of embedded hardware.

https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp&hl=zh_HK

gauravdhim1990, 2023-03-31:
I am also facing the same issue. when i connected to the device it's auto disconnected after setup notification. 
What should i do. 
I am using aut-flag false.

Yohannfra, 2023-10-11:
Hi, we have the exact same issue. Like @gauravdhim1990 described when we connect to our device we get this error after the notification setup.