dariuszseweryn/RxAndroidBle #627: Compile errors in v. 1.10.2

RobLewis, 2019-09-25:
When I update my `rxandroidble-parent` project in Android Studio to v. 1.10.2, I get these "cannot resolve symbol" errors in the file `RxBleGattCallback.java`: `bleshadow` (lines 26 & 27), `Inject` (line 59), and `Named` (line 60). 

Everything else compiles OK.

dariuszseweryn, 2019-09-25:
This seems to be related to https://github.com/johnrengelman/shadow/issues/264
I wish there would be a way to achieve it.

This seems to be an IDE / gradle issue. I have changed `settings.gradle` from:
```
include ':rxandroidble', ':dagger-compiler-shadow', ':dagger-library-shadow'
include ':sample', ':sample-kotlin', ':mockrxandroidble'

rootProject.name = 'rxandroidble-parent'
```
to
```
include ':dagger-compiler-shadow', ':dagger-library-shadow', ':rxandroidble'
include ':sample', ':sample-kotlin', ':mockrxandroidble'

rootProject.name = 'rxandroidble-parent'
```
And re-synced the project as IDE has suggested and the issue was gone. Once I did `File -> Invalidate Caches/Restart` the problem reappeared.

RobLewis, 2019-09-25:
So all you did was change the order of the items in the first `include`? 
I don't really understand much about this but I note that the bug in `shadow` is almost 3 years old (or is it a JetBrains bug?). The project history shows that it was worked on as recently as 3 months ago so evidently it is being maintained.

dariuszseweryn, 2019-09-26:
> So all you did was change the order of the items in the first `include`?

Yes but it only helped temporarily.
The project seems to be maintained but it seems that this issue is either complex to fix/workaround or not a big impact from the maintainers perspective.

Now it seems that a simple click on `Sync Project with Gradle Files` button solves the problem?
[image]
Dunno, either way I do not know what can I do on the this repository's side to help with the issue

RobLewis, 2019-10-03:
I've tried updating to latest Android Studio and Gradle, editing `settings.gradle`, syncing project to Gradle files. Nothing changes the errors. 

Does this mean I'm going to get this same problem in other projects using the library?

dariuszseweryn, 2019-10-21:
> Does this mean I'm going to get this same problem in other projects using the library?

No — this errors are visible only in the library project, not in the consumer application projects

closed 2020-05-21 by RobLewis