dariuszseweryn/RxAndroidBle #636: Investigate `ScanFilter.setManufacturerData()` not working

dariuszseweryn, 2019-10-28:
### Summary
It seems that setting
`ScanFitler.Builder().setManufacturerData(/* int manufacturerId */ 0x0123, /* byte[] manufacturerData */ null)` will not pass any advertisements with the specified `manufacturerId`.
More fitting solution would be to pass all advertisements with this id no matter the data.

### Library version
`1.10.3` (not a library issue)

#### Preconditions
Have a peripheral advertising data with `manufacturerId = 0x0123`

#### Steps to reproduce actual result
<br />1. Set `ScanFitler.Builder().setManufacturerData(/* int manufacturerId */ 0x0123, /* byte[] manufacturerData */ null)`
<br />2. Scan peripheral that advertises `manufacturerId = 0x0123`
<br />3. Observe results

#### Actual result
No `ScanResult` is emitted

#### Expected result
A `ScanResult` with the given `manufacturerId` is emitted

#### Minimum code snippet reproducing the issue
```
        val scanSettings = ScanSettings.Builder()
            .setScanMode(ScanSettings.SCAN_MODE_LOW_LATENCY)
            .setCallbackType(ScanSettings.CALLBACK_TYPE_ALL_MATCHES)
            .build()

        val manuId = 0xBEEF
        val scanFilter = ScanFilter.Builder()
            .setManufacturerData(manuId, null)
            .build()

        return rxBleClient.scanBleDevices(scanSettings, scanFilter)
```

#### Logs from the application running with settings:
```java
RxBleClient.updateLogOptions(new LogOptions.Builder()
        .setLogLevel(LogConstants.DEBUG)
        .setMacAddressLogSetting(LogConstants.MAC_ADDRESS_FULL)
        .setUuidsLogSetting(LogConstants.UUIDS_FULL)
        .setShouldLogAttributeValues(true)
        .build()
);
```

```
// no logs as Android OS does not match anything
```

omeryounus, 2020-01-09:
Also setDeviceName(name) and setDeviceMacAddress(macaddress) is not working.

dariuszseweryn, 2020-01-31:
I have seen `setDeviceName(name)` not working because of name being advertised with more than one trailing `0`. As for the MAC address I have not seen this before.

Now testing with Samsung SM-G920F / API 24. When using `.setManufacturerData(manuId, null)` the filter does not seem to work at all.