dariuszseweryn/RxAndroidBle #637: Add scan precondition verifier for Android 10

dariuszseweryn, 2019-10-28:
### Summary
On Android 10 scan needs `ACCESS_FINE_LOCATION` permission.

### Library version
`1.10.3`

#### Preconditions
Android 10 mobile phone to run the test

#### Steps to reproduce actual result
<br />1. Use an app with `ACCESS_COARSE_LOCATION`
<br />2. Try to run a scan
<br />3. Observe result

#### Actual result
Scan seems to be started correctly but a log is shown:
```
E/BluetoothUtils: Permission denial: Need ACCESS_FINE_LOCATION permission to get scan results
```

#### Expected result
A `BleScanException` is emitted with a desirable error code `LOCATION_PERMISSION_MISSING` or `FINE_LOCATION_PERMISSION_MISSING` (TBD)

#### Minimum code snippet reproducing the issue
```
        val scanSettings = ScanSettings.Builder()
            .setScanMode(ScanSettings.SCAN_MODE_LOW_LATENCY)
            .setCallbackType(ScanSettings.CALLBACK_TYPE_ALL_MATCHES)
            .build()

        val scanFilter = ScanFilter.Builder()
            // add custom filters if needed
            .build()

        rxBleClient.scanBleDevices(scanSettings, scanFilter).subscribe()
```

#### Logs from the application running with settings:
```java
RxBleClient.updateLogOptions(new LogOptions.Builder()
        .setLogLevel(LogConstants.DEBUG)
        .setMacAddressLogSetting(LogConstants.MAC_ADDRESS_FULL)
        .setUuidsLogSetting(LogConstants.UUIDS_FULL)
        .setShouldLogAttributeValues(true)
        .build()
);
```

```
2019-10-28 17:29:28.364 8318-8357/com.polidea.rxandroidble.sample I/RxBle#ClientOperationQueue: RUNNING  ScanOperationApi21{ANY_MUST_MATCH -> nativeFilters=[BluetoothLeScanFilter [mDeviceName=null, MAC=null, mUuid=null, mUuidMask=null, mServiceDataUuid=null, mServiceData=null, mServiceDataMask=null, mManufacturerId=-1, mManufacturerData=null, mManufacturerDataMask=null]]}
2019-10-28 17:29:28.365 8318-8380/com.polidea.rxandroidble.sample I/RxBle#QueueOperation: Scan operation is requested to start.
2019-10-28 17:29:28.368 8318-8380/com.polidea.rxandroidble.sample D/BluetoothAdapter: isLeEnabled(): ON
2019-10-28 17:29:28.374 6276-6323/? I/bt_stack: [INFO:gatt_api.cc(946)] GATT_Register 4f95b434-3363-7845-06bf-01ea4105e510
2019-10-28 17:29:28.374 6276-6323/? I/bt_stack: [INFO:gatt_api.cc(966)] allocated gatt_if=7
2019-10-28 17:29:28.375 8318-8345/com.polidea.rxandroidble.sample D/BluetoothLeScanner: onScannerRegistered() - status=0 scannerId=7 mScannerId=0
2019-10-28 17:29:28.380 6276-6294/? E/BluetoothUtils: Permission denial: Need ACCESS_FINE_LOCATION permission to get scan results
```

closed 2019-11-07 by dariuszseweryn

Steppschuh, 2019-11-08:
This should probably be mentioned in the `Permissions` section of the readme.

dariuszseweryn, 2019-11-08:
How would you see it?

Steppschuh, 2019-11-08:
I just would give a heads up to users that they need to request the `ACCESS_FINE_LOCATION` permission on Android 10+.  From the current wording one could think that `ACCESS_COARSE_LOCATION` would be sufficient.

dariuszseweryn, 2019-11-08:
Should be better now

PR #640 (2019-11-07): Fixed location permission check for Android 10
Since Android 10 ‘Manifest.permission.ACCESS_COARSE_LOCATION’ is no longer sufficient to get BLE scan results. ‘Manifest.permission.ACCESS_FINE_LOCATION’ is a must. Adjusted BleScanException.LOCATION_PERMISSION_MISSING javadoc to reflect that.

Closes #637