dariuszseweryn/RxAndroidBle #641: Observable or Single?

valeriyo, 2019-11-07:
This is confusing:
```
  /**
     * @return Observable emitting another observable when the notification setup is complete.
     */
    Observable<Observable<byte[]>> setupNotification(@NonNull UUID characteristicUuid, @NonNull NotificationSetupMode setupMode);
```
Why is this not `Single<Observable<byte[]>>` ?

dariuszseweryn, 2019-11-07:
https://github.com/Polidea/RxAndroidBle#observable-behaviour

valeriyo, 2019-11-07:
Thanks for the link. Nonetheless, it should be `Single<Observable<byte[]>>` - the subscription to the emitted `Observable<byte[]>` should be sufficient to manage resources.

dariuszseweryn, 2019-11-07:
Sometimes there is a use case when there are no subscribers to `Observable<byte[]>` but the user does not want to notify the peripheral about it

valeriyo, 2019-11-07:
Not sure I understand. You are saying that there are use-cases when one would subscribe to the outer `Observable<Observable<byte[]>>` but not to the inner `Observable<byte[]>`?

dariuszseweryn, 2019-11-07:
Yes. Imagine that the inner `Observable<byte[]>` gets subscribed at various times depending on other factors but the BLE notification (as a side effect) is expected to be continuous.

dariuszseweryn, 2019-11-08:
Does it answer your question @valeriyo ? Do you have an idea on how to address this case differently?

valeriyo, 2019-11-09:
Yes, thanks for following up. I'm planning to use `RxAndroidBle` for scanning - because it seems to work very well, but for data (connecting/writing/reading) - I found the `RxAndroidBle` APIs too cumbersome to use (managing observables subscriptions, and the UndeliverableException, which is actually the `RxAndroidBle`'s fault - it shouldn't emit errors when there are no subscribers). So, I'll stick with `nordicsemi` for now. Thanks.

closed 2019-11-09 by valeriyo

dariuszseweryn, 2019-11-12:
> I found the RxAndroidBle APIs too cumbersome to use

I am afraid that I cannot do much with this. You are always welcome to give constructive feedback. Best regards