dariuszseweryn/RxAndroidBle #644: Operation queue blocking

Kilnn, 2019-11-11:
### Summary
When a continuous set of operations is added to the queue, one of the operations in the middle throws an exception. The subsequent operation disposes and causes the QueueReleaseInterface to fail to be released, causing the entire queue to block.

### Library version
`1.10.4`

### Test mobile phones
`Samsung SM-G9300 Android 8.0.0`

### Test Code
```
private void testCode() {
        final RxBleConnection rxBleConnection = mConnector.getRxBleConnection();
        if (rxBleConnection == null) return;

        Observable.fromArray(1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15)
                //Use flatMap instead of concatMap to quickly add operations to the queue
                .flatMapSingle(new Function<Integer, SingleSource<byte[]>>() {
                    @Override
                    public SingleSource<byte[]> apply(Integer integer) throws Exception {
                        if (integer == 5) {
                            //Custom operation throws an exception
                            return rxBleConnection
                                    .queue(new RxBleCustomOperation<byte[]>() {
                                        @NonNull
                                        @Override
                                        public Observable<byte[]> asObservable(BluetoothGatt bluetoothGatt, RxBleGattCallback rxBleGattCallback, Scheduler scheduler) throws Throwable {
                                            Thread.sleep(3000);
                                            throw new NullPointerException();
                                        }
                                    }).firstOrError();
                        } else {
                            //Normal operation
                            return rxBleConnection
                                    .writeCharacteristic(WRITE_CHARACTERISTIC_UUID, new byte[]{integer.byteValue(), 0, 0, 0, 0});
                        }
                    }
                })
                .subscribe(new Consumer<byte[]>() {
                    @Override
                    public void accept(byte[] bytes) throws Exception {
                        Log.d("TestActivity", "bytes:" + Arrays.toString(bytes));
                    }
                }, new Consumer<Throwable>() {
                    @Override
                    public void accept(Throwable throwable) throws Exception {
                        Log.w("TestActivity", "error", throwable);
                        //After the error occurs, continue to repeat the test until the bug is reproduced
                        Thread.sleep(500);
                        testCode();
                    }
                });
    }
```

### Log
```
2019-11-11 15:40:26.357 18776-19610/com.htsmart.wristband2sample D/TestActivity: bytes:[4, 0, 0, 0, 0]
2019-11-11 15:40:26.363 18776-19609/com.htsmart.wristband2sample D/RxBle#ConnectionOperationQueue: FINISHED CharacteristicWriteOperation(47329391) in 592 ms
2019-11-11 15:40:26.367 18776-19609/com.htsmart.wristband2sample D/RxBle#ConnectionOperationQueue: STARTED  (24295941)
2019-11-11 15:40:26.369 18776-19609/com.htsmart.wristband2sample I/RxBle#ConnectionOperationQueue: RUNNING  com.polidea.rxandroidble2.internal.connection.RxBleConnectionImpl$11@172ba05
2019-11-11 15:40:29.378 18776-19609/com.htsmart.wristband2sample D/RxBle#ConnectionOperationQueue: FINISHED (24295941) in 3010 ms
2019-11-11 15:40:29.383 18776-19609/com.htsmart.wristband2sample D/RxBle#ConnectionOperationQueue: STARTED  CharacteristicWriteOperation(16475275)
2019-11-11 15:40:29.383 18776-18888/com.htsmart.wristband2sample D/RxBle#ConnectionOperationQueue: REMOVED  CharacteristicWriteOperation(206967647)
2019-11-11 15:40:29.388 18776-19609/com.htsmart.wristband2sample I/RxBle#ConnectionOperationQueue: RUNNING  CharacteristicWriteOperation{MAC='AB:01:CD:10:A1:C9', characteristic=[uuid='0000ff02-0000-1000-8000-00805f9b34fb', hexValue=[06, 00, 00, 00, 00]]}
2019-11-11 15:40:29.390 18776-18888/com.htsmart.wristband2sample D/RxBle#ConnectionOperationQueue: REMOVED  CharacteristicWriteOperation(85015793)
2019-11-11 15:40:29.400 18776-18888/com.htsmart.wristband2sample D/RxBle#ConnectionOperationQueue: REMOVED  CharacteristicWriteOperation(54115943)
2019-11-11 15:40:29.404 18776-18888/com.htsmart.wristband2sample D/RxBle#ConnectionOperationQueue: REMOVED  CharacteristicWriteOperation(176312249)
2019-11-11 15:40:29.408 18776-18888/com.htsmart.wristband2sample D/RxBle#ConnectionOperationQueue: REMOVED  CharacteristicWriteOperation(239281341)
2019-11-11 15:40:29.412 18776-18888/com.htsmart.wristband2sample D/RxBle#ConnectionOperationQueue: REMOVED  CharacteristicWriteOperation(191492481)
[... 24 lines cut ...]
2019-11-11 15:40:29.972 18776-18888/com.htsmart.wristband2sample D/RxBle#ConnectionOperationQueue: QUEUED   CharacteristicWriteOperation(92986246)
2019-11-11 15:40:29.975 18776-18888/com.htsmart.wristband2sample D/RxBle#ConnectionOperationQueue: QUEUED   CharacteristicWriteOperation(145407092)
2019-11-11 15:40:29.977 18776-18888/com.htsmart.wristband2sample D/RxBle#ConnectionOperationQueue: QUEUED   CharacteristicWriteOperation(11086354)
2019-11-11 15:40:29.981 18776-18888/com.htsmart.wristband2sample D/RxBle#ConnectionOperationQueue: QUEUED   CharacteristicWriteOperation(267463648)
2019-11-11 15:40:29.984 18776-18888/com.htsmart.wristband2sample D/RxBle#ConnectionOperationQueue: QUEUED   CharacteristicWriteOperation(99341662)
2019-11-11 15:40:29.986 18776-18888/com.htsmart.wristband2sample D/RxBle#ConnectionOperationQueue: QUEUED   CharacteristicWriteOperation(23930380)
2019-11-11 15:40:29.988 18776-18888/com.htsmart.wristband2sample D/RxBle#ConnectionOperationQueue: QUEUED   CharacteristicWriteOperation(236710250)
2019-11-11 15:40:29.991 18776-18888/com.htsmart.wristband2sample D/RxBle#ConnectionOperationQueue: QUEUED   CharacteristicWriteOperation(13407992)
2019-11-11 15:40:29.993 18776-18888/com.htsmart.wristband2sample E/RxBle#ObservableCreate: QueueOperation terminated with an unexpected exception
    java.lang.NullPointerException
        at com.htsmart.wristband2sample.testgatt.TestGattLayerActivity$11$1.asObservable(TestGattLayerActivity.java:172)
        at com.polidea.rxandroidble2.internal.connection.RxBleConnectionImpl$11.protectedRun(RxBleConnectionImpl.java:330)
        at com.polidea.rxandroidble2.internal.QueueOperation$1.subscribe(QueueOperation.java:41)
        at io.reactivex.internal.operators.observable.ObservableCreate.subscribeActual(ObservableCreate.java:40)
        at io.reactivex.Observable.subscribe(Observable.java:12267)
        at io.reactivex.internal.operators.observable.ObservableSubscribeOn$SubscribeTask.run(ObservableSubscribeOn.java:96)
        at io.reactivex.internal.schedulers.ScheduledDirectTask.call(ScheduledDirectTask.java:38)
        at io.reactivex.internal.schedulers.ScheduledDirectTask.call(ScheduledDirectTask.java:26)
        at java.util.concurrent.FutureTask.run(FutureTask.java:266)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1162)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:636)
        at java.lang.Thread.run(Thread.java:764)

```
In the log, the last started operation `CharacteristicWriteOperation(16475275)` did not end normally.

Kilnn, 2019-11-11:
The error occurred in FIFORunnableEntry, and the Observable dispose in onSubscribe, causing ObservableOnSubscribe#subscribe to fail to execute.

Like this:

```
Observable
                .create(new ObservableOnSubscribe<Integer>() {
                    @Override
                    public void subscribe(ObservableEmitter<Integer> emitter) throws Exception {
                        Log.d("TestActivity", "subscribe");
                    }
                })
                .subscribeOn(Schedulers.io())
                .unsubscribeOn(Schedulers.io())
                .subscribe(new Observer<Integer>() {
                    @Override
                    public void onSubscribe(Disposable d) {
                        Log.d("TestActivity", "onSubscribe");
                        d.dispose();
                    }

                    @Override
                    public void onNext(Integer integer) {

                    }

                    @Override
                    public void onError(Throwable e) {

                    }

                    @Override
                    public void onComplete() {

                    }
                });
```

dariuszseweryn, 2019-11-12:
Hello,
Thank you for the report. I am afraid that the example code you provided violates the contract of `RxBleCustomOperation`. Would you mind using something that does conform to it? Like:
```
new RxBleCustomOperation<byte[]>() {
    @NonNull
    @Override
    public Observable<byte[]> asObservable(BluetoothGatt bluetoothGatt, RxBleGattCallback rxBleGattCallback, Scheduler scheduler) throws Throwable {
        return Observable.error<byte[]>(new NullPointerException()).delaySubscription(3000, TimeUnit.MILLISECONDS, scheduler)
    }
}
```

Kilnn, 2019-11-12:
I used the way you recommended to construct `RxBleCustomOperation` . After running for a while, the same bug will occur.

dariuszseweryn, 2019-11-19:
Yup, confirmed. I was trying to understand and solve the problem during past few days. I think I have worked around it — you can check branch `fix/race_stalled_queue` and see if it helps

Kilnn, 2019-11-20:
Using `Scheduler.scheduleDirect` instead of `subscriptionOn(Scheduler)`, I tested for an hour and no bug occurred.
Well ! Thank you for your help!

PR #650 (2019-11-22): Fixes stalled operation queue in a rare race situation
`.subscribeOn()` operator seems to be not actually subscribing to the upstream if downstream is already disposed. It seems impossible to track whether the upstream was subscribed or not and therefore side-effects for releasing a queue cannot be reliably run. Getting rid of the operator but running the subscription on the `Scheduler`’s worker directly makes the operation being always run. It’s the operation’s responsibility to release the queue at a proper moment (when the `BluetoothGatt` is no longer blocked).

Fixes #644
closed 2019-11-22 by dariuszseweryn