dariuszseweryn/RxAndroidBle #645: Please deprecate ScanFilters in the RxBleClient.scanBleDevices(ScanSettings scanSettings, ScanFilter... scanFilters) method

valeriyo, 2019-11-12:
Apparently, `ScanFilters` doesn't work on some devices. For example, filtering by Service UUID doesn't work on Samsung Galaxy S6 (as documented by https://stackoverflow.com/questions/34065210/android-ble-device-scan-with-filter-is-not-working - and I confirmed this issue myself today, after spending 2+ hours troubleshooting). The workaround is to remove all `ScanFilters` and filter results "manually".

RxAndroidBle shouldn't expose broken OS functionality, when a workaround is so apparent. Please remove `ScanFilters` altogether, and let developers filter results with an Rx `.filter` operator - which is going to work 100%.

Thanks.

dariuszseweryn, 2019-11-12:
Could you disclose more details about the issue you encountered? I.e.
- what is the BLE advertisement's payload you want to match?
- what is the exact setup of `ScanSettings` and `ScanFilter`s you use?

valeriyo, 2019-11-21:
Sure - just a basic filter by service UUID - nothing fancy:
```
      rxBleClient.scanBleDevices(
           ScanSettings.Builder()
               .setScanMode(SCAN_MODE_LOW_LATENCY)
               .build(),
           ScanFilter.Builder()
              .setServiceUuid(ParcelUuid(serviceUuid))
              .build()
      )
```
- this works on many devices, but on Galaxy S6 there are no results.

dariuszseweryn, 2019-11-21:
I had run a quick check with Samsung SM-G920F / API 24 seems to work as intended. I would need more info. Unfortunately the linked SO post is not very consistent in terms of the actual situation with what the peripheral is advertising and what is the bug.

```
rxBleClient.scanBleDevices(
    ScanSettings.Builder().setScanMode(ScanSettings.SCAN_MODE_LOW_LATENCY).build(),
    ScanFilter.Builder().setServiceUuid(ParcelUuid.fromString("6E400000-B5A3-F393-E0A9-E50E24DCCA9E")).build()
)
    .observeOn(AndroidSchedulers.mainThread())
    .subscribe(
        { Log.e("SUCCESS", "OK!") },
        { Log.e("FAILURE", "NOK!", it) }
    )
```
```
2019-11-21 14:56:23.981 19454-19475/com.polidea.rxandroidble.sample I/RxBle#ClientOperationQueue: RUNNING  ScanOperationApi21{ANY_MUST_MATCH -> nativeFilters=[BluetoothLeScanFilter [mDeviceName=null, MAC=null, mUuid=6e400000-b5a3-f393-e0a9-e50e24dcca9e, mUuidMask=null, mServiceDataUuid=null, mServiceData=null, mServiceDataMask=null, mManufacturerId=-1, mManufacturerData=null, mManufacturerDataMask=null]]}
2019-11-21 14:56:23.983 19454-19789/com.polidea.rxandroidble.sample I/RxBle#QueueOperation: Scan operation is requested to start.
2019-11-21 14:56:24.642 19454-19454/com.polidea.rxandroidble.sample I/RxBle#Client: ScanResult{bleDevice=RxBleDeviceImpl{MAC='D3:4D:73:E8:41:1F', name=Queued_Write}, rssi=-46, timestampNanos=267733097276126, callbackType=CALLBACK_TYPE_ALL_MATCHES, scanRecord=[03, 19, 00, 00, 02, 01, 06, 11, 07, 9E, CA, DC, 24, 0E, E5, A9, E0, 93, F3, A3, B5, 00, 00, 40, 6E, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00]}
2019-11-21 14:56:24.643 19454-19454/com.polidea.rxandroidble.sample E/SUCCESS: OK!
2019-11-21 14:56:25.202 19454-19454/com.polidea.rxandroidble.sample I/RxBle#Client: ScanResult{bleDevice=RxBleDeviceImpl{MAC='D3:4D:73:E8:41:1F', name=Queued_Write}, rssi=-46, timestampNanos=267733679303001, callbackType=CALLBACK_TYPE_ALL_MATCHES, scanRecord=[03, 19, 00, 00, 02, 01, 06, 11, 07, 9E, CA, DC, 24, 0E, E5, A9, E0, 93, F3, A3, B5, 00, 00, 40, 6E, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00]}
2019-11-21 14:56:25.202 19454-19454/com.polidea.rxandroidble.sample E/SUCCESS: OK!
2019-11-21 14:56:25.595 19454-19454/com.polidea.rxandroidble.sample I/RxBle#Client: ScanResult{bleDevice=RxBleDeviceImpl{MAC='D3:4D:73:E8:41:1F', name=Queued_Write}, rssi=-47, timestampNanos=267734069510418, callbackType=CALLBACK_TYPE_ALL_MATCHES, scanRecord=[03, 19, 00, 00, 02, 01, 06, 11, 07, 9E, CA, DC, 24, 0E, E5, A9, E0, 93, F3, A3, B5, 00, 00, 40, 6E, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00]}
2019-11-21 14:56:25.596 19454-19454/com.polidea.rxandroidble.sample E/SUCCESS: OK!
2019-11-21 14:56:25.842 19454-19789/com.polidea.rxandroidble.sample I/RxBle#CancellableDisposable: Scan operation is requested to stop.
```

dariuszseweryn, 2019-12-02:
Since I have no more actionable info and disabling the whole feature would be a big loss for the whole user base — I am closing until more info is available.

closed 2019-12-02 by dariuszseweryn

dariuszseweryn, 2019-12-06:
Accidentally I have recreated the issue while looking for UUID `43210100-5678-9012-3456-789012345678` — it may be worth reinvestigating

dariuszseweryn, 2020-01-31:
A small update:
I get inconsistent results – Today I spent an hour testing scan filter with Service UUIDs:
- `43210100-5678-9012-3456-789012345678`
- `43210200-5678-9012-3456-789012345678`
- `6E400001-B5A3-F393-E0A9-E50E24DCCA9E`

They worked fine on the same device as before (Samsung SM-G920F / API 24).