dariuszseweryn/RxAndroidBle #648: crash in stopLeScan because "BT Adapter is not turned ON"

valeriyo, 2019-11-21:
With RxAndroidBle v1.10.4 - any advice on how to deal with this crash?

```
E/e: Uncaught exception in thread Thread[pool-1-thread-1,5,main]
    java.lang.IllegalStateException: BT Adapter is not turned ON
        at android.bluetooth.le.BluetoothLeUtils.checkAdapterStateOn(BluetoothLeUtils.java:136)
        at android.bluetooth.le.BluetoothLeScanner.stopScan(BluetoothLeScanner.java:266)
        at com.polidea.rxandroidble2.internal.util.RxBleAdapterWrapper.stopLeScan(RxBleAdapterWrapper.java:75)
        at com.polidea.rxandroidble2.internal.operations.ScanOperationApi21.stopScan(ScanOperationApi21.java:113)
        at com.polidea.rxandroidble2.internal.operations.ScanOperationApi21.stopScan(ScanOperationApi21.java:28)
        at com.polidea.rxandroidble2.internal.operations.ScanOperation$1.cancel(ScanOperation.java:39)
        at io.reactivex.internal.disposables.CancellableDisposable.dispose(CancellableDisposable.java:48)
        at io.reactivex.internal.disposables.DisposableHelper.dispose(DisposableHelper.java:124)
        at io.reactivex.internal.operators.observable.ObservableCreate$CreateEmitter.dispose(ObservableCreate.java:121)
        at io.reactivex.internal.disposables.DisposableHelper.dispose(DisposableHelper.java:124)
        at io.reactivex.internal.operators.observable.ObservableSubscribeOn$SubscribeOnObserver.dispose(ObservableSubscribeOn.java:73)
        at io.reactivex.internal.operators.observable.ObservableUnsubscribeOn$UnsubscribeObserver$DisposeTask.run(ObservableUnsubscribeOn.java:95)
        at io.reactivex.internal.schedulers.ScheduledDirectTask.call(ScheduledDirectTask.java:38)
        at io.reactivex.internal.schedulers.ScheduledDirectTask.call(ScheduledDirectTask.java:26)
        at java.util.concurrent.FutureTask.run(FutureTask.java:266)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
        at java.lang.Thread.run(Thread.java:919)
```

dariuszseweryn, 2019-11-21:
What Android version? 10?

valeriyo, 2019-11-21:
Android 10, Pixel 3XL

dariuszseweryn, 2019-11-21:
Please check branch `fix/scan_stop_illegal_state_exception`
This is some new behaviour which was not observed in API < 29

While I am grateful for your reports, please try to stick to the template — it really helps me with dealing with them.

Best Regards

valeriyo, 2019-11-21:
Looks like the fix should do the trick - thanks!

BTW, I couldn't find any recent changes that could have caused this in https://android.googlesource.com/platform/frameworks/base/+/refs/heads/master/core/java/android/bluetooth/le/BluetoothLeScanner.java - assuming, that's the correct source.

valeriyo, 2019-11-21:
When is the next release planned?

dariuszseweryn, 2019-11-22:
> BTW, I couldn't find any recent changes that could have caused this in https://android.googlesource.com/platform/frameworks/base/+/refs/heads/master/core/java/android/bluetooth/le/BluetoothLeScanner.java - assuming, that's the correct source.

I dunno. Technically it should be crashing since Android 5.0. I dunno why it does not nor I see any such logs anywhere. I have just tested on Pixel with Android 10 and the app did not crash. Anyway I will incorporate this fix and will try to make a release Today (no guarantee though)

dariuszseweryn, 2019-11-22:
Are you sure that this is actually a crash or a log from a default thread error handler?

valeriyo, 2019-11-22:
I'm not sure - this was reported to me by a developer who's using our SDK in their app. Either way, it's a good fix to have, since it's going to avoid an exception from happening (from their point of view, an incorrect usage of their API). Thanks!

PR #649 (2019-11-25): Added check BluetoothAdapter.isEnabled() before calling BluetoothLeScanner.stopScan(ScanCallback)
Apparently on some versions of Android OS it is possible to get BluetoothLeScanner even if bluetooth is off. In this situation calling any methods/functions on the scanner will fail with an IllegalStateException due to adapter being off. Previously the scanner was not possible to obtain when the adapter was off or the exception was not thrown.

Fixes #648
closed 2019-11-25 by dariuszseweryn