dariuszseweryn/RxAndroidBle #652: button read is not enabled for Temperature Measurement value

fukemy, 2019-12-03:
Hello, first thanks for best library. I have a question:
 In class CharacteristicOperationExampleActivity, i connected to Temperature Measurement value(00002A1C-0000-1000-8000-00805f9b34fb), but after connected button read is not enabled, it's hard to know to me, can you help? Thanks

dariuszseweryn, 2019-12-03:
Hello. The characteristic mentioned by you must have no `READ` property
Java:
```
readButton.setEnabled(hasProperty(characteristic, BluetoothGattCharacteristic.PROPERTY_READ));
```
Kotlin:
```
read.isEnabled = hasProperty(BluetoothGattCharacteristic.PROPERTY_READ)
```
If you expect to read it you should ask the developer of the peripheral to fix the configuration. Or maybe your expectation is wrong? Maybe you should check notifications?
Anyway — this does not seem to be a problem with the library.
Best Regards

closed 2019-12-03 by dariuszseweryn

fukemy, 2019-12-03:
hello, let me explain first

Here is my device : 
[image]
This device were using to get temperature from baby who wear it.
Here is other app used code to get temperature from this device:
[image]

In other code saw this const val is using to get temperature: 
```
val HT_MEASUREMENT_CHARACTERISTIC_UUID = UUID.fromString("00002A1C-0000-1000-8000-00805f9b34fb")
val HT_SERVICE_UUID = UUID.fromString("00001809-0000-1000-8000-00805f9b34fb")
```

So i swear that device can using to get temperature, but i dont know why button read is not enabled, can you explain more about READ property, thanks

dariuszseweryn, 2019-12-04:
I believe that it is possible to get data from the device.
> So i swear that device can using to get temperature, but i dont know why button read is not enabled, can you explain more about READ property, thanks

Let me repeat what is added in the issue template you have deleted when posting this issue:
> Disclaimer
**Before you will post a new issue to the RxAndroidBle library consider if you are filling a bug/feature request or a general question about:**
>
> Usage of RxJava
**Usage of Bluetooth Low Energy**
Usage of RxAndroidBle
Weird behaviour of the peripheral you are using
Other not directly related to a new feature request or a bug
**If any of the above points seems to match your use case — consider using Google** or creating a question on www.stackoverflow.com with a tag rxandroidble where it will be easier to access by other people with similar questions. Issues that are not bugs or feature requests will be closed.

There are plenty of pages that describe what is BLE, characteristics, properties etc. They will give you much better understanding than me trying to quickly get you up to speed.

fukemy, 2019-12-05:
Hi, after 1 day to find out problem, i tried to use BluetoothLeGatt from Google, it's work with the same UUID 
```
/**
     * Enables or disables notification on a give characteristic.
     *
     * @param characteristic Characteristic to act on.
     * @param enabled If true, enable notification.  False otherwise.
     */
    public void setCharacteristicNotification(BluetoothGattCharacteristic characteristic,
                                              boolean enabled) {
        if (mBluetoothAdapter == null || mBluetoothGatt == null) {
            Log.w(TAG, "BluetoothAdapter not initialized");
            return;
        }
        mBluetoothGatt.setCharacteristicNotification(characteristic, enabled);

        // This is specific to Heart Rate Measurement.
        if (HT_MEASUREMENT_CHARACTERISTIC_UUID.equals(characteristic.getUuid())) {
            BluetoothGattDescriptor descriptor = characteristic.getDescriptor(
                    UUID.fromString(SampleGattAttributes.CLIENT_CHARACTERISTIC_CONFIG));
            descriptor.setValue(BluetoothGattDescriptor.ENABLE_INDICATION_VALUE);
            mBluetoothGatt.writeDescriptor(descriptor);
        }
    }
```

In above case i just enable indication for HT_MEASUREMENT_CHARACTERISTIC_UUID, then i can get temperature from my device. 
So, in my case, i need to register indication to get temperature value, now it's work, thanks for your help 👍

dariuszseweryn, 2019-12-05:
Glad to hear that you have solved the problem 👍