dariuszseweryn/RxAndroidBle #661: BleAlreadyConnectedException throws when try to reconnect after disconnecting

tidimito, 2020-01-08:
I have the functionality when user can disconnect from connected BLE device and then connect to other one or to the same device. 
The source code of my service is:
```
public class BluetoothLeService extends Service {

    private final static long CONNECTION_DELAY_IN_SECONDS = 6L;

    private RxBleClient rxBleClient = App.getRxBleClient();
    private RxBleDevice targetBluetoothDevice = null;

    // Observables
    private Observable<RxBleConnection> connectionObservable = null;
    private Observable<RxBleConnection.RxBleConnectionState> connectionStateObservable = null;

    private Disposable internalConnectionStateDisposable = null;
    private Disposable internalConnectionDisposable = null;

    // Scan
    private Observable<ScanResult> scanObservable = null;
    private Disposable scanDisposable = null;

    private int connectionAttempts  = 0;

    public synchronized void scan() {
        if (scanDisposable != null) {
             return;
        }

        scanObservable = rxBleClient.scanBleDevices(
                new ScanSettings.Builder()
                        .setScanMode(ScanSettings.SCAN_MODE_LOW_LATENCY)
                        .setCallbackType(ScanSettings.CALLBACK_TYPE_ALL_MATCHES)
                        .build(),
                new ScanFilter.Builder().setServiceUuid(new ParcelUuid(SOME_UUID_1)).build(),
                new ScanFilter.Builder().setServiceUuid(new ParcelUuid(SOME_UUID_2)).build()
        )
                .subscribeOn(Schedulers.io())
                .observeOn(AndroidSchedulers.mainThread())
                .doFinally(() -> {
                    scanDisposable.dispose();
                    scanDisposable = null;
                });

        scanDisposable = scanObservable.subscribe(
                scanResult -> {
                   ...
                   handleDeviceFound(scanResult.getBleDevice());
                },
                Timber::e
        );
    }

    public void stopScan() {
        if (scanDisposable != null) {
            scanDisposable.dispose();
            scanDisposable = null;
        }
    }

    private synchronized void connect(@NonNull final String address, boolean autoConnect) {
        if (targetBluetoothDevice != null) {
            disconnect();

            synchronized (this) {
                try {
                    wait(5000L);
                } catch (InterruptedException ignored) { }
            }
        }

        targetBluetoothDevice = rxBleClient.getBleDevice(address);

        connectionStateObservable = targetBluetoothDevice.observeConnectionStateChanges()
                .mergeWith(disconnectSubject)
                .compose(ReplayingShare.instance())
                .observeOn(AndroidSchedulers.mainThread());

        // Create an internal connection state subscription
        internalConnectionStateDisposable = connectionStateObservable.subscribe(
                (RxBleConnection.RxBleConnectionState state) -> Timber.d("Connection state changed to: %s", state.toString()));

        // Establish the connection with retry logic
        connectionObservable = targetBluetoothDevice.establishConnection(autoConnect)
                .doOnNext(rxBleConnection -> {
                    ...
                    onConnectionEstablished();
                })
                .compose(ReplayingShare.instance())
                .observeOn(AndroidSchedulers.mainThread());

        // Create a subscription to this connection observable
        internalConnectionDisposable = connectionObservable.subscribe(
                                                                  this::logConnection, 
                                                                  this::retryOnFailure);
    }

    public void disconnect() {
        if (targetBluetoothDevice == null) {
            return;
        }

        internalConnectionStateDisposable.dispose();
        internalConnectionDisposable.dispose();

        targetBluetoothDevice = null;
    }

    private synchronized void retryOnFailure(Throwable throwable) {
        if (targetBluetoothDevice == null) {
            return;
        }
        connectionAttempts++;

        Timer connectionTimer = new Timer();
        TimerTask timerTask = new TimerTask() {
            @Override
            public void run() {
                if (targetBluetoothDevice == null) return;
                connect(targetBluetoothDevice.getMacAddress(), connectionAttempts > 2);
            }
        };
        connectionTimer.schedule(timerTask, 500L);
    }
}
```
When user want to change BLE-device I have the following function calls sequence
```
stopScan();
disconnect();
scan()
//User chooses new device
connect(MAC, false);
```
and sometimes when the App tries to reconnect to the same device I receive
```
com.polidea.rxandroidble2.exceptions.BleAlreadyConnectedException: Already connected to device with MAC address DC.......
```
Maybe I use the api in a wrong way? Or I should add something?

dariuszseweryn, 2020-01-08:
Hello, could you conform to the bug report template? Most probably the library is right and you did not disconnected from the peripheral

tidimito, 2020-01-08:
Yes, I think it isn't disconnected because after some attempts device connected successfully. But what is the correct way to wait/understand that ble-device is disconnected and we can reconnect?

dariuszseweryn, 2020-01-08:
`RxBleDevice#establishConnection` observable will either emit an error or it will be disposed

tidimito, 2020-01-08:
Thank you. Could you please answer some more questions?
What if I dispose internalConnectionDisposable but don't dispose internalConnectionStateDisposable? Will ble-device be correct disconnected and will I receive disconnected state without memory leaks and other side effects?

dariuszseweryn, 2020-01-08:
That should be the case.

dariuszseweryn, 2020-01-10:
If that is all I close this issue

closed 2020-01-10 by dariuszseweryn