dariuszseweryn/RxAndroidBle #662: RxBleDevice.observeConnectionStateChanges gets stuck after N reconnections

kanat, 2020-01-10:
## Steps

1. Start listening connection states:
```kotlin
.flatMapObservable { rxBleDevice -> rxBleDevice.observeConnectionStateChanges() }
.observeOn(scheduler)
.doOnSubscribe { SLog.d(TAG, "[listenConnection] subscribed(%s)", address) }
.doOnNext { SLog.v(TAG, "[listenConnection] next(%s): %s", address, it) }
.doOnDispose { SLog.v(TAG, "[listenConnection] disposed(%s)", address) }
.doFinally { SLog.v(TAG, "[listenConnection] finished(%s)", address) }
```

2. Get connected to `ble device`
3. Turn off `ble device`
4. Get disconnected (`RxBleConnectionState.DISCONNECTED`)
5. Schedule reconnection in 2 seconds
6. Repeat from stage 4.

## Actual result
After `N` attempts of reconnections `observeConnectionStateChanges()` emits nothing.

## Expected result
After `N` attempts of reconnections `observeConnectionStateChanges()` still emits `RxBleConnectionState` events.

### Investigation:
My `Disposable` from the stage **1** is still alive because I didn't see such a message in the `Logcat`:
```kotlin 
SLog.v(TAG, "[listenConnection] disposed(%s)", address)
``` 
Finally, I've figured out that starting at some connection retry the `subscribers` array inside `BehaviorRelay` is empty.

#### Screenshots:
- `BehaviorRelay` has subscribers:
[image]

- `BehaviorRelay` has **NO** subscribers:
[image]

## Logs:
```
2020-01-10 12:07:03.515 4088-4198/ru.starlinex.lib.testapp W/SleManager: (sle-1-t-1:3203) [scheduleReconnect] address: CA:5F:21:15:CB:74
2020-01-10 12:07:05.519 4088-4198/ru.starlinex.lib.testapp I/SleManager: (sle-1-t-1:3203) [scheduleReconnect] time to connect: CA:5F:21:15:CB:74
2020-01-10 12:07:05.541 4088-4201/ru.starlinex.lib.testapp D/RxBle#ClientOperationQueue: (RxBleThread-1:3206) QUEUED   ConnectOperation(267538281)
2020-01-10 12:07:05.543 4088-4193/ru.starlinex.lib.testapp D/RxBle#ClientOperationQueue: (Thread-6:3201) STARTED  ConnectOperation(267538281)
2020-01-10 12:07:05.545 4088-4193/ru.starlinex.lib.testapp I/RxBle#ClientOperationQueue: (Thread-6:3201) RUNNING  ConnectOperation{MAC='CA:5F:21:15:CB:74', autoConnect=false}
2020-01-10 12:07:05.549 4088-4202/ru.starlinex.lib.testapp V/RxBle#BleConnectionCompat: (pool-1-thread-1:3207) Connecting without reflection
2020-01-10 12:07:05.549 4088-4198/ru.starlinex.lib.testapp V/SleManager: (sle-1-t-1:3203) [listenConnection] next(CA:5F:21:15:CB:74): RxBleConnectionState{CONNECTING}
2020-01-10 12:07:05.551 4088-4202/ru.starlinex.lib.testapp D/BluetoothAdapter: STATE_ON
2020-01-10 12:07:05.555 4088-4202/ru.starlinex.lib.testapp D/BluetoothGatt: connect() - device: CA:5F:21:15:CB:74, auto: false
2020-01-10 12:07:05.555 4088-4202/ru.starlinex.lib.testapp D/BluetoothAdapter: isSecureModeEnabled
2020-01-10 12:07:05.557 4088-4202/ru.starlinex.lib.testapp D/BluetoothGatt: registerApp()
2020-01-10 12:07:05.557 4088-4202/ru.starlinex.lib.testapp D/BluetoothGatt: registerApp() - UUID=5a585778-6475-4751-8e43-8a5a8c644b34
2020-01-10 12:07:05.563 4088-4112/ru.starlinex.lib.testapp D/BluetoothGatt: onClientRegistered() - status=0 clientIf=10
2020-01-10 12:07:10.591 4088-4112/ru.starlinex.lib.testapp D/BluetoothGatt: onClientConnectionState() - status=133 clientIf=10 device=CA:5F:21:15:CB:74
2020-01-10 12:07:10.601 4088-4112/ru.starlinex.lib.testapp I/RxBle#GattCallback: (Binder:4088_2:3192) MAC='CA:5F:21:15:CB:74'  onConnectionStateChange(), status=133, value=0
2020-01-10 12:07:10.606 4088-4112/ru.starlinex.lib.testapp D/RxBle#ConnectionOperationQueue: (Binder:4088_2:3192) Connection operations queue to be terminated (MAC='CA:5F:21:15:CB:74')
    com.polidea.rxandroidble2.exceptions.BleDisconnectedException: Disconnected from MAC='CA:5F:21:15:CB:74' with status 133 (GATT_ERROR)
        at com.polidea.rxandroidble2.internal.connection.RxBleGattCallback$2.onConnectionStateChange(RxBleGattCallback.java:81)
        at android.bluetooth.BluetoothGatt$1$4.run(BluetoothGatt.java:313)
        at android.bluetooth.BluetoothGatt.runOrQueueCallback(BluetoothGatt.java:876)
        at android.bluetooth.BluetoothGatt.access$200(BluetoothGatt.java:43)
        at android.bluetooth.BluetoothGatt$1.onClientConnectionState(BluetoothGatt.java:308)
        at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:71)
        at android.os.Binder.execTransact(Binder.java:739)
2020-01-10 12:07:10.608 4088-4200/ru.starlinex.lib.testapp V/RxBle#Executors: (pool-3-thread-1:3205) Terminated (MAC='CA:5F:21:15:CB:74')
2020-01-10 12:07:10.612 4088-4112/ru.starlinex.lib.testapp D/RxBle#ClientOperationQueue: (Binder:4088_2:3192) QUEUED   DisconnectOperation(99379484)
2020-01-10 12:07:10.616 4088-4193/ru.starlinex.lib.testapp D/RxBle#ClientOperationQueue: (Thread-6:3201) FINISHED ConnectOperation(267538281) in 5072 ms
2020-01-10 12:07:10.617 4088-4193/ru.starlinex.lib.testapp D/RxBle#ClientOperationQueue: (Thread-6:3201) STARTED  DisconnectOperation(99379484)
2020-01-10 12:07:10.620 4088-4193/ru.starlinex.lib.testapp I/RxBle#ClientOperationQueue: (Thread-6:3201) RUNNING  DisconnectOperation{MAC='CA:5F:21:15:CB:74'}
2020-01-10 12:07:10.622 4088-4202/ru.starlinex.lib.testapp D/BluetoothManager: getConnectionState()
[... 90 lines cut ...]
2020-01-10 12:07:26.996 4088-4202/ru.starlinex.lib.testapp D/BluetoothGatt: registerApp()
2020-01-10 12:07:26.996 4088-4202/ru.starlinex.lib.testapp D/BluetoothGatt: registerApp() - UUID=dbbf89c4-c2b2-4926-b07c-236250b8e46e
2020-01-10 12:07:27.003 4088-4112/ru.starlinex.lib.testapp D/BluetoothGatt: onClientRegistered() - status=0 clientIf=10
2020-01-10 12:07:32.032 4088-4112/ru.starlinex.lib.testapp D/BluetoothGatt: onClientConnectionState() - status=133 clientIf=10 device=CA:5F:21:15:CB:74
2020-01-10 12:07:32.040 4088-4112/ru.starlinex.lib.testapp I/RxBle#GattCallback: (Binder:4088_2:3192) MAC='CA:5F:21:15:CB:74'  onConnectionStateChange(), status=133, value=0
2020-01-10 12:07:32.045 4088-4112/ru.starlinex.lib.testapp D/RxBle#ConnectionOperationQueue: (Binder:4088_2:3192) Connection operations queue to be terminated (MAC='CA:5F:21:15:CB:74')
    com.polidea.rxandroidble2.exceptions.BleDisconnectedException: Disconnected from MAC='CA:5F:21:15:CB:74' with status 133 (GATT_ERROR)
        at com.polidea.rxandroidble2.internal.connection.RxBleGattCallback$2.onConnectionStateChange(RxBleGattCallback.java:81)
        at android.bluetooth.BluetoothGatt$1$4.run(BluetoothGatt.java:313)
        at android.bluetooth.BluetoothGatt.runOrQueueCallback(BluetoothGatt.java:876)
        at android.bluetooth.BluetoothGatt.access$200(BluetoothGatt.java:43)
        at android.bluetooth.BluetoothGatt$1.onClientConnectionState(BluetoothGatt.java:308)
        at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:71)
        at android.os.Binder.execTransact(Binder.java:739)
2020-01-10 12:07:32.047 4088-4200/ru.starlinex.lib.testapp V/RxBle#Executors: (pool-3-thread-1:3205) Terminated (MAC='CA:5F:21:15:CB:74')
2020-01-10 12:07:32.051 4088-4112/ru.starlinex.lib.testapp D/RxBle#ClientOperationQueue: (Binder:4088_2:3192) QUEUED   DisconnectOperation(167738852)
2020-01-10 12:07:32.054 4088-4193/ru.starlinex.lib.testapp D/RxBle#ClientOperationQueue: (Thread-6:3201) FINISHED ConnectOperation(191452689) in 5080 ms
2020-01-10 12:07:32.056 4088-4193/ru.starlinex.lib.testapp D/RxBle#ClientOperationQueue: (Thread-6:3201) STARTED  DisconnectOperation(167738852)
2020-01-10 12:07:32.057 4088-4193/ru.starlinex.lib.testapp I/RxBle#ClientOperationQueue: (Thread-6:3201) RUNNING  DisconnectOperation{MAC='CA:5F:21:15:CB:74'}
2020-01-10 12:07:32.058 4088-4202/ru.starlinex.lib.testapp D/BluetoothManager: getConnectionState()
2020-01-10 12:07:32.058 4088-4202/ru.starlinex.lib.testapp D/BluetoothManager: getConnectedDevices
2020-01-10 12:07:32.063 4088-4202/ru.starlinex.lib.testapp D/BluetoothGatt: close()
2020-01-10 12:07:32.070 4088-4202/ru.starlinex.lib.testapp D/BluetoothGatt: unregisterApp() - mClientIf=10
2020-01-10 12:07:32.079 4088-4193/ru.starlinex.lib.testapp D/RxBle#ClientOperationQueue: (Thread-6:3201) FINISHED DisconnectOperation(167738852) in 21 ms
```

dariuszseweryn, 2020-01-10:
I see that the relay is properly informed about the state change. Which would point to the subscriber being absent — what does it have to do with the library?

kanat, 2020-01-10:
@dariuszseweryn the library uses `BehaviorRelay` under the hood. 3rd party apps work directly with `RxBleDevice`. My point is that the library should either keep emitting `RxBleConnectionStates` (no matter what is hidden behind) because I didn't dispose my `Observable` or notify me that `observeConnectionStateChanges` has been disposed.

So, currently, I'm still subscribed to `observeConnectionStateChanges`, but get no connection state changes.

dariuszseweryn, 2020-01-10:
I made a simple test just now that does nothing more than repeatedly connects/disconnects a device (SensorTag in this case). It running several minutes now already and seems to work just fine.
```
        AtomicInteger ai = new AtomicInteger();
        scanDisposable = rxBleClient.scanBleDevices(
                new ScanSettings.Builder()
                        .setScanMode(ScanSettings.SCAN_MODE_LOW_LATENCY)
                        .setCallbackType(ScanSettings.CALLBACK_TYPE_ALL_MATCHES)
                        .build(),
                new ScanFilter.Builder()
//                            .setDeviceAddress("B4:99:4C:34:DC:8B")
                        // add custom filters if needed
                        .build()
        )
                .filter(scanResult -> "SensorTag".equals(scanResult.getBleDevice().getName()))
                .take(1)
                .map(ScanResult::getBleDevice)
                .flatMap(rxBleDevice -> Observable.merge(
                        rxBleDevice.observeConnectionStateChanges(),
                        rxBleDevice.establishConnection(false)
                                .take(1)
                                .ignoreElements()
                                .doOnComplete(() -> Log.i("STATE", "Connections: " + ai.getAndIncrement()))
                                .repeatUntil(() -> false)
                                .toObservable()
                ))
                .doFinally(this::dispose)
                .subscribe(state -> Log.i("STATE", state.toString()), this::onScanFailure);
```
Logs:
```
2020-01-10 17:51:20.708 15983-16469/com.polidea.rxandroidble.sample I/STATE: RxBleConnectionState{DISCONNECTING}
2020-01-10 17:51:20.736 15983-16469/com.polidea.rxandroidble.sample I/STATE: RxBleConnectionState{DISCONNECTED}
2020-01-10 17:51:20.740 15983-16469/com.polidea.rxandroidble.sample I/STATE: RxBleConnectionState{CONNECTING}
2020-01-10 17:51:20.911 15983-16472/com.polidea.rxandroidble.sample I/STATE: RxBleConnectionState{CONNECTED}
2020-01-10 17:51:20.913 15983-16472/com.polidea.rxandroidble.sample I/STATE: Connections: 1172
2020-01-10 17:51:20.933 15983-16469/com.polidea.rxandroidble.sample I/STATE: RxBleConnectionState{DISCONNECTING}
2020-01-10 17:51:20.964 15983-16469/com.polidea.rxandroidble.sample I/STATE: RxBleConnectionState{DISCONNECTED}
2020-01-10 17:51:20.966 15983-16469/com.polidea.rxandroidble.sample I/STATE: RxBleConnectionState{CONNECTING}
2020-01-10 17:51:21.145 15983-16472/com.polidea.rxandroidble.sample I/STATE: RxBleConnectionState{CONNECTED}
2020-01-10 17:51:21.148 15983-16472/com.polidea.rxandroidble.sample I/STATE: Connections: 1173
2020-01-10 17:51:21.176 15983-16469/com.polidea.rxandroidble.sample I/STATE: RxBleConnectionState{DISCONNECTING}
2020-01-10 17:51:21.224 15983-16469/com.polidea.rxandroidble.sample I/STATE: RxBleConnectionState{DISCONNECTED}
2020-01-10 17:51:21.227 15983-16469/com.polidea.rxandroidble.sample I/STATE: RxBleConnectionState{CONNECTING}
2020-01-10 17:51:21.399 15983-16472/com.polidea.rxandroidble.sample I/STATE: RxBleConnectionState{CONNECTED}
2020-01-10 17:51:21.400 15983-16472/com.polidea.rxandroidble.sample I/STATE: Connections: 1174
2020-01-10 17:51:21.420 15983-16469/com.polidea.rxandroidble.sample I/STATE: RxBleConnectionState{DISCONNECTING}
2020-01-10 17:51:21.464 15983-16469/com.polidea.rxandroidble.sample I/STATE: RxBleConnectionState{DISCONNECTED}
2020-01-10 17:51:21.469 15983-16469/com.polidea.rxandroidble.sample I/STATE: RxBleConnectionState{CONNECTING}
2020-01-10 17:51:21.634 15983-16472/com.polidea.rxandroidble.sample I/STATE: RxBleConnectionState{CONNECTED}
2020-01-10 17:51:21.636 15983-16472/com.polidea.rxandroidble.sample I/STATE: Connections: 1175
2020-01-10 17:51:21.655 15983-16469/com.polidea.rxandroidble.sample I/STATE: RxBleConnectionState{DISCONNECTING}
2020-01-10 17:51:21.725 15983-16469/com.polidea.rxandroidble.sample I/STATE: RxBleConnectionState{DISCONNECTED}
2020-01-10 17:51:21.730 15983-16469/com.polidea.rxandroidble.sample I/STATE: RxBleConnectionState{CONNECTING}
2020-01-10 17:51:21.911 15983-16472/com.polidea.rxandroidble.sample I/STATE: RxBleConnectionState{CONNECTED}
```
I cannot reproduce the issue you are facing. I would need to get a minimalistic example code so I could test myself. Or if you think you have found a bug in `RxRelay` you could raise a bug there.

kanat, 2020-01-10:
Ok, let me create some simple app to reproduce :)
Btw, this behavior is unstable, it takes me about 5-15 minutes to reproduce.

dariuszseweryn, 2020-01-10:
I have altered my test to check with retries:
```
        AtomicInteger ai = new AtomicInteger();
        AtomicInteger ai2 = new AtomicInteger();
        scanDisposable = rxBleClient.scanBleDevices(
                new ScanSettings.Builder()
                        .setScanMode(ScanSettings.SCAN_MODE_LOW_LATENCY)
                        .setCallbackType(ScanSettings.CALLBACK_TYPE_ALL_MATCHES)
                        .build(),
                new ScanFilter.Builder()
//                            .setDeviceAddress("B4:99:4C:34:DC:8B")
                        // add custom filters if needed
                        .build()
        )
                .filter(scanResult -> "SensorTag".equals(scanResult.getBleDevice().getName()))
                .take(1)
                .map(ScanResult::getBleDevice)
                .flatMap(rxBleDevice -> Observable.merge(
                        rxBleDevice.observeConnectionStateChanges(),
                        rxBleDevice.establishConnection(false)
                                .take(1)
                                .ignoreElements()
                                .doOnComplete(() -> Log.i("STATE", "Connections: " + ai.getAndIncrement()))
                                .repeatUntil(() -> false)
                                .retryWhen(throwableFlowable -> throwableFlowable.doOnNext(t -> Log.i("STATE", "Retries: " + ai2.getAndIncrement())))
                                .toObservable()
                ))
                .doFinally(this::dispose)
                .subscribe(state -> Log.i("STATE", state.toString()), this::onScanFailure);
```

Logs:
```
2020-01-10 21:51:34.119 7298-7882/com.polidea.rxandroidble.sample I/STATE: RxBleConnectionState{DISCONNECTING}
2020-01-10 21:51:34.135 7298-7882/com.polidea.rxandroidble.sample I/STATE: RxBleConnectionState{DISCONNECTED}
2020-01-10 21:51:34.140 7298-7882/com.polidea.rxandroidble.sample I/STATE: RxBleConnectionState{CONNECTING}
2020-01-10 21:51:39.338 7298-14762/com.polidea.rxandroidble.sample I/STATE: Retries: 851
2020-01-10 21:51:39.358 7298-7882/com.polidea.rxandroidble.sample I/STATE: RxBleConnectionState{DISCONNECTING}
2020-01-10 21:51:39.375 7298-7882/com.polidea.rxandroidble.sample I/STATE: RxBleConnectionState{DISCONNECTED}
2020-01-10 21:51:39.380 7298-7882/com.polidea.rxandroidble.sample I/STATE: RxBleConnectionState{CONNECTING}
2020-01-10 21:51:44.579 7298-7310/com.polidea.rxandroidble.sample I/STATE: Retries: 852
2020-01-10 21:51:44.590 7298-7882/com.polidea.rxandroidble.sample I/STATE: RxBleConnectionState{DISCONNECTING}
2020-01-10 21:51:44.601 7298-7882/com.polidea.rxandroidble.sample I/STATE: RxBleConnectionState{DISCONNECTED}
2020-01-10 21:51:44.604 7298-7882/com.polidea.rxandroidble.sample I/STATE: RxBleConnectionState{CONNECTING}
2020-01-10 21:51:49.785 7298-7884/com.polidea.rxandroidble.sample I/STATE: Retries: 853
2020-01-10 21:51:49.798 7298-7882/com.polidea.rxandroidble.sample I/STATE: RxBleConnectionState{DISCONNECTING}
2020-01-10 21:51:49.809 7298-7882/com.polidea.rxandroidble.sample I/STATE: RxBleConnectionState{DISCONNECTED}
2020-01-10 21:51:49.813 7298-7882/com.polidea.rxandroidble.sample I/STATE: RxBleConnectionState{CONNECTING}
2020-01-10 21:51:54.979 7298-16144/com.polidea.rxandroidble.sample I/STATE: Retries: 854
```
Still looks good...

dariuszseweryn, 2020-01-16:
@kanat Any news?

kanat, 2020-01-16:
@dariuszseweryn, sorry guys. An urgent release came up, so I didn't have time to continue with this. I'll try to return to this issue next week.

dariuszseweryn, 2020-01-31:
Since quite few days has passed, there are is no more info nor I can recreate the issue — I am closing. Feel free to comment if new data will be available and I could reopen.

Best regards

kanat, 2020-02-06:
@dariuszseweryn, sorry for the late reply.
Here is the sample app: https://github.com/kanat/rxandroidble_sample
It took me about 20 minutes to reproduce.
Just open the app and let it work until the **DISCONNECTED** state gets frozen. 

[image]

### Logs:
```
2020-02-06 20:25:44.791 27207-27419/com.sample.rxandroidble2 D/BluetoothGatt: onClientConnectionState() - status=133 clientIf=10 device=CA:5F:21:15:CB:74

...

2020-02-06 20:41:48.578 27207-27249/com.sample.rxandroidble2 D/RxBle#SampleApp: [connect] no args
2020-02-06 20:41:48.581 27207-27246/com.sample.rxandroidble2 D/RxBle#ClientOperationQueue: QUEUED   ConnectOperation(201709189)
2020-02-06 20:41:48.583 27207-27244/com.sample.rxandroidble2 D/RxBle#ClientOperationQueue: STARTED  ConnectOperation(201709189)
2020-02-06 20:41:48.583 27207-27244/com.sample.rxandroidble2 I/RxBle#ClientOperationQueue: RUNNING  ConnectOperation{MAC='CA:5F:21:15:CB:74', autoConnect=false}
2020-02-06 20:41:48.588 27207-27247/com.sample.rxandroidble2 V/RxBle#BleConnectionCompat: Connecting without reflection
2020-02-06 20:41:48.588 27207-27249/com.sample.rxandroidble2 D/RxBle#SampleApp: [listenConnection] emitted: RxBleConnectionState{CONNECTING}
2020-02-06 20:41:48.589 27207-27247/com.sample.rxandroidble2 D/BluetoothAdapter: STATE_ON
2020-02-06 20:41:48.595 27207-27247/com.sample.rxandroidble2 D/BluetoothGatt: connect() - device: CA:5F:21:15:CB:74, auto: false
2020-02-06 20:41:48.595 27207-27247/com.sample.rxandroidble2 D/BluetoothAdapter: isSecureModeEnabled
2020-02-06 20:41:48.597 27207-27247/com.sample.rxandroidble2 D/BluetoothGatt: registerApp()
2020-02-06 20:41:48.597 27207-27247/com.sample.rxandroidble2 D/BluetoothGatt: registerApp() - UUID=796b0a6f-d13c-4e47-9739-842108fb5549
2020-02-06 20:41:48.603 27207-27901/com.sample.rxandroidble2 D/BluetoothGatt: onClientRegistered() - status=0 clientIf=10
2020-02-06 20:41:53.619 27207-27901/com.sample.rxandroidble2 D/BluetoothGatt: onClientConnectionState() - status=133 clientIf=10 device=CA:5F:21:15:CB:74
2020-02-06 20:41:53.623 27207-27901/com.sample.rxandroidble2 I/RxBle#GattCallback: MAC='CA:5F:21:15:CB:74'  onConnectionStateChange(), status=133, value=0
2020-02-06 20:41:53.628 27207-27901/com.sample.rxandroidble2 D/RxBle#ConnectionOperationQueue: Connection operations queue to be terminated (MAC='CA:5F:21:15:CB:74')
    com.polidea.rxandroidble2.exceptions.BleDisconnectedException: Disconnected from MAC='CA:5F:21:15:CB:74' with status 133 (GATT_ERROR)
        at com.polidea.rxandroidble2.internal.connection.RxBleGattCallback$2.onConnectionStateChange(RxBleGattCallback.java:81)
        at android.bluetooth.BluetoothGatt$1$4.run(BluetoothGatt.java:313)
        at android.bluetooth.BluetoothGatt.runOrQueueCallback(BluetoothGatt.java:876)
        at android.bluetooth.BluetoothGatt.access$200(BluetoothGatt.java:43)
        at android.bluetooth.BluetoothGatt$1.onClientConnectionState(BluetoothGatt.java:308)
        at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:71)
        at android.os.Binder.execTransact(Binder.java:739)
2020-02-06 20:41:53.629 27207-27245/com.sample.rxandroidble2 V/RxBle#Executors: Terminated (MAC='CA:5F:21:15:CB:74')
2020-02-06 20:41:53.630 27207-27249/com.sample.rxandroidble2 E/RxBle#SampleApp: [connect] failed: com.polidea.rxandroidble2.exceptions.BleDisconnectedException: Disconnected from MAC='CA:5F:21:15:CB:74' with status 133 (GATT_ERROR)
    com.polidea.rxandroidble2.exceptions.BleDisconnectedException: Disconnected from MAC='CA:5F:21:15:CB:74' with status 133 (GATT_ERROR)
        at com.polidea.rxandroidble2.internal.connection.RxBleGattCallback$2.onConnectionStateChange(RxBleGattCallback.java:81)
        at android.bluetooth.BluetoothGatt$1$4.run(BluetoothGatt.java:313)
        at android.bluetooth.BluetoothGatt.runOrQueueCallback(BluetoothGatt.java:876)
        at android.bluetooth.BluetoothGatt.access$200(BluetoothGatt.java:43)
        at android.bluetooth.BluetoothGatt$1.onClientConnectionState(BluetoothGatt.java:308)
        at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:71)
        at android.os.Binder.execTransact(Binder.java:739)
2020-02-06 20:41:53.630 27207-27901/com.sample.rxandroidble2 D/RxBle#ClientOperationQueue: QUEUED   DisconnectOperation(262111208)
2020-02-06 20:41:53.631 27207-27244/com.sample.rxandroidble2 D/RxBle#ClientOperationQueue: FINISHED ConnectOperation(201709189) in 5049 ms
2020-02-06 20:41:53.632 27207-27244/com.sample.rxandroidble2 D/RxBle#ClientOperationQueue: STARTED  DisconnectOperation(262111208)
2020-02-06 20:41:53.632 27207-27244/com.sample.rxandroidble2 I/RxBle#ClientOperationQueue: RUNNING  DisconnectOperation{MAC='CA:5F:21:15:CB:74'}
2020-02-06 20:41:53.633 27207-27247/com.sample.rxandroidble2 D/BluetoothManager: getConnectionState()
2020-02-06 20:41:53.633 27207-27247/com.sample.rxandroidble2 D/BluetoothManager: getConnectedDevices
2020-02-06 20:41:53.634 27207-27249/com.sample.rxandroidble2 D/RxBle#SampleApp: [listenConnection] emitted: RxBleConnectionState{DISCONNECTING}
2020-02-06 20:41:53.646 27207-27247/com.sample.rxandroidble2 D/BluetoothGatt: close()
2020-02-06 20:41:53.651 27207-27247/com.sample.rxandroidble2 D/BluetoothGatt: unregisterApp() - mClientIf=10
2020-02-06 20:41:53.655 27207-27244/com.sample.rxandroidble2 D/RxBle#ClientOperationQueue: FINISHED DisconnectOperation(262111208) in 23 ms
[... cut ...]
2020-02-06 20:42:00.744 27207-27247/com.sample.rxandroidble2 D/BluetoothManager: getConnectionState()
2020-02-06 20:42:00.745 27207-27247/com.sample.rxandroidble2 D/BluetoothManager: getConnectedDevices
2020-02-06 20:42:00.751 27207-27247/com.sample.rxandroidble2 D/BluetoothGatt: close()
2020-02-06 20:42:00.756 27207-27247/com.sample.rxandroidble2 D/BluetoothGatt: unregisterApp() - mClientIf=10
2020-02-06 20:42:00.764 27207-27244/com.sample.rxandroidble2 D/RxBle#ClientOperationQueue: FINISHED DisconnectOperation(73425920) in 19 ms
```

kanat, 2020-02-06:
I'm able to reproduce such behavior on Samsung S10 (SM-973F) with Android 9 onboard.

dariuszseweryn, 2020-02-19:
This is a true Heisenbug. I did reproduce this issue twice without any meaningful logs to debug. When added more logs I could not reproduce it any longer. I did encountered few `UndeliverableExceptions` though. Other than that I have run the program for quite a few nights already. I am running on Samsung S6 with Android 7. I will check if I can lay my hands on some newer Galaxy S phone.

dariuszseweryn, 2020-02-24:
I have added quite a lot of logs to the library and I am running tests... Unfortunately I do not have a good reproduction ratio — currently my test is running for almost 91 hours straight without issues.
```
2020-02-24 10:52:28.029 901-5067/? I/RxBle#GattCallback: MAC='CA:5F:21:15:CB:74'  onConnectionStateChange(), status=133, value=0
2020-02-24 10:52:28.036 901-5067/? D/RxBle#ConnectionOperationQueue: Connection operations queue to be terminated (MAC='CA:5F:21:15:CB:74')
    com.polidea.rxandroidble2.exceptions.BleDisconnectedException: Disconnected from MAC='CA:5F:21:15:CB:74' with status 133 (GATT_ERROR)
        at com.polidea.rxandroidble2.internal.connection.RxBleGattCallback$2.onConnectionStateChange(RxBleGattCallback.java:81)
        at android.bluetooth.BluetoothGatt$1.onClientConnectionState(BluetoothGatt.java:228)
        at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:70)
        at android.os.Binder.execTransact(Binder.java:573)
2020-02-24 10:52:28.037 901-984/? E/RxBle#SampleApp: [connect] failed: com.polidea.rxandroidble2.exceptions.BleDisconnectedException: Disconnected from MAC='CA:5F:21:15:CB:74' with status 133 (GATT_ERROR)
    com.polidea.rxandroidble2.exceptions.BleDisconnectedException: Disconnected from MAC='CA:5F:21:15:CB:74' with status 133 (GATT_ERROR)
        at com.polidea.rxandroidble2.internal.connection.RxBleGattCallback$2.onConnectionStateChange(RxBleGattCallback.java:81)
        at android.bluetooth.BluetoothGatt$1.onClientConnectionState(BluetoothGatt.java:228)
        at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:70)
        at android.os.Binder.execTransact(Binder.java:573)
2020-02-24 10:52:28.039 901-5067/? D/RxBle#ClientOperationQueue: QUEUED   DisconnectOperation(136225217)
2020-02-24 10:52:28.042 901-960/? D/RxBle#ClientOperationQueue: FINISHED ConnectOperation(192496026) in 5210 ms
2020-02-24 10:52:28.043 901-960/? D/RxBle#ClientOperationQueue: STARTED  DisconnectOperation(136225217)
2020-02-24 10:52:28.045 901-960/? I/RxBle#ClientOperationQueue: RUNNING  DisconnectOperation{MAC='CA:5F:21:15:CB:74'}
2020-02-24 10:52:28.047 901-980/? I/RxBle#DisconnectOperation: New state => RxBleConnectionState{DISCONNECTING}/138144915
2020-02-24 10:52:28.048 901-980/? I/RxBle#Device: CONN_STATE_CHANGE next RxBleConnectionState{DISCONNECTING} 138144915
2020-02-24 10:52:28.048 901-980/? D/BluetoothManager: getConnectionState()
2020-02-24 10:52:28.048 901-980/? D/BluetoothManager: getConnectedDevices
2020-02-24 10:52:28.049 901-984/? D/RxBle#SampleApp: [listenConnection] emitted: RxBleConnectionState{DISCONNECTING}
2020-02-24 10:52:28.050 901-984/? D/RxBle#SampleApp: [updateUI] schedule: RxBleConnectionState{DISCONNECTING}
2020-02-24 10:52:28.053 901-980/? D/BluetoothGatt: close()
2020-02-24 10:52:28.055 901-980/? D/BluetoothGatt: unregisterApp() - mClientIf=6
2020-02-24 10:52:28.058 901-980/? I/RxBle#DisconnectOperation: New state => RxBleConnectionState{DISCONNECTED}/138144915
2020-02-24 10:52:28.059 901-980/? I/RxBle#Device: CONN_STATE_CHANGE next RxBleConnectionState{DISCONNECTED} 138144915
2020-02-24 10:52:28.060 901-984/? D/RxBle#SampleApp: [listenConnection] emitted: RxBleConnectionState{DISCONNECTED}
2020-02-24 10:52:28.061 901-984/? D/RxBle#SampleApp: [updateUI] schedule: RxBleConnectionState{DISCONNECTED}
2020-02-24 10:52:28.061 901-960/? D/RxBle#ClientOperationQueue: FINISHED DisconnectOperation(136225217) in 18 ms
2020-02-24 10:52:28.061 901-984/? D/RxBle#SampleApp: [scheduleReconnect] no args
2020-02-24 10:52:30.065 901-984/? D/RxBle#SampleApp: [scheduleReconnect] emitted: 0
2020-02-24 10:52:30.065 901-984/? D/RxBle#SampleApp: [connect] call
2020-02-24 10:52:30.088 901-984/? D/RxBle#SampleApp: [connect] no args
2020-02-24 10:52:30.090 901-984/? I/RxBle#SampleApp: [connect] undeliverables: 14 / 90:53:39
2020-02-24 10:52:30.097 901-979/? D/RxBle#ClientOperationQueue: QUEUED   ConnectOperation(265710066)
2020-02-24 10:52:30.100 901-960/? D/RxBle#ClientOperationQueue: STARTED  ConnectOperation(265710066)
2020-02-24 10:52:30.103 901-960/? I/RxBle#ClientOperationQueue: RUNNING  ConnectOperation{MAC='CA:5F:21:15:CB:74', autoConnect=false}
2020-02-24 10:52:30.106 901-980/? I/RxBle#ConnectOperation: New state => RxBleConnectionState{CONNECTING}/138144915
2020-02-24 10:52:30.108 901-980/? I/RxBle#Device: CONN_STATE_CHANGE next RxBleConnectionState{CONNECTING} 138144915
2020-02-24 10:52:30.110 901-984/? D/RxBle#SampleApp: [listenConnection] emitted: RxBleConnectionState{CONNECTING}
2020-02-24 10:52:30.111 901-984/? D/RxBle#SampleApp: [updateUI] schedule: RxBleConnectionState{CONNECTING}
2020-02-24 10:52:30.118 901-980/? D/BluetoothGatt: connect() - device: CA:5F:21:15:CB:XX, auto: false
2020-02-24 10:52:30.118 901-980/? D/BluetoothAdapter: isSecureModeEnabled
2020-02-24 10:52:30.123 901-980/? D/BluetoothGatt: registerApp()
2020-02-24 10:52:30.123 901-980/? D/BluetoothGatt: registerApp() - UUID=59042749-9652-4380-addd-221ce6318b9c
2020-02-24 10:52:30.232 901-16574/? D/BluetoothGatt: onClientRegistered() - status=0 clientIf=6
2020-02-24 10:52:35.275 901-5940/? D/BluetoothGatt: onClientConnectionState() - status=133 clientIf=6 device=CA:5F:21:15:CB:XX
2020-02-24 10:52:35.289 901-5940/? I/RxBle#GattCallback: MAC='CA:5F:21:15:CB:74'  onConnectionStateChange(), status=133, value=0
[... cut ...]
2020-02-24 10:52:35.325 901-984/? D/RxBle#SampleApp: [scheduleReconnect] no args
2020-02-24 10:52:35.328 901-960/? D/RxBle#ClientOperationQueue: FINISHED DisconnectOperation(177522681) in 18 ms
```
I wonder if lack of RxJava error handler is not a potential culprit.
```
RxJavaPlugins.setErrorHandler {
    if (it is UndeliverableException && it.cause is BleException) {
        Timber.tag(TAG).e("Undeliverable!")
        undeliverables.incrementAndGet()
        return@setErrorHandler
    }
    throw it
}
```

dariuszseweryn, 2020-04-03:
@kanat Any news perhaps?

closed 2020-06-09 by dariuszseweryn