dariuszseweryn/RxAndroidBle #665: BLE device update  info after connection

dononcharles, 2020-01-22:
Hi RxAndroidBle Team,

_Sorry for my question which is not bug neither a feature request._

My problem description:
I am using your library to scan ble devices and it working. But i've seen that the fullname of ble device is cut, for exmple: DENV-1253563623 and i only get DENV-12. I think is something regarding the transmission which is limited on 8 bytes.
I ve also seen that if i make a paired connection with my ble device, after that, the fullname is showing but i've done this connection with my PC bluetooth system and i now i want to scan and before showing the device information to the client, i will make a background connection et get the new name. 

Now my question is, how can i make a connection  > bleDevice.establishConnection(false)
and after that get the new name of the device?

`private fun autoConnectDevice(itSc: ScanResult) {
        itSc.bleDevice.establishConnection(false)
            .flatMapSingle  { it.discoverServices() }
            .observeOn(AndroidSchedulers.mainThread())
            .doFinally { dispose() }
            .subscribe({
                Log.e("tes", it.toString())
                onConnectionReceived()
            }, {
                onConnectionFailure(it)
            }).let {
                connectionDisposable = it
            }
    }`

Hope you will understand my question.
Thanks for any help

dariuszseweryn, 2020-01-22:
Hello,

As you have concluded this is not the best place to ask this question as this is not a bug nor a feature request. To give you some hints about what may be happening so you could move foreward:
1. Most probably the advertisement packet contains other data and there is simply no room for the full name (`DENV-1253563623`) — in this case only a partial may be sent (`DENV-12`)
2. Full name may be sent in a EIR / Scan Response packet. Some Android phones do not return EIR data when offloaded filtering is used
3. There is a Name Characteristic defined in the Bluetooth Specification which can be read after connecting to the peripheral
4. How does your PC handle the situation – I do not know

Hope this will help you. Good luck

closed 2020-01-22 by dariuszseweryn

dononcharles, 2020-01-22:
Thanks for your reply. 
As for the PC connection, i do not know the pc handle it.
But am interesting in your third 3.point.

Can u help to make that. just after connected to the device et get this info?
Thanks

dariuszseweryn, 2020-01-22:
Simply connect and read the characteristic if it is available. Google should have the answer about the characteristic UUID. Making an issue on github is not the place to seek knowledge about how to use RxJava or BLE. Unfortunately I do not have enough time to help everyone but I try to give some hints what can you google for.

Best regards

dononcharles, 2020-01-22:
ok.thanks.

Best regards

dononcharles, 2020-01-22:
Hi, 
Please, i looking around to find the solution for my problem but u ve said something which if possible, can u explain me:

**Some Android phones do not return EIR data when offloaded filtering is used** what do u mean by **when offloaded filtering is used**

Thanks for your consideration time

dariuszseweryn, 2020-01-22:
https://lmgtfy.com/?q=android+ble+offloaded+filtering — probably the first result will have something about it in the official docs
basically any filtering not done by your code

dononcharles, 2020-01-24:
Hi Darius,
Hope u are doing well. After long search i found solution to get the fullname of the device.  I think for the first time am getting only advertising info which send only short name of the device. so now i need to loop on a specific service to get the full name.
I would not like to disturb you but i think u can have some idea to give me to solve another problem i am getting. 
After connected to the devices, and got the fullname, i would like to close all the connection to devices. i am connecting to many devices. but when i close the app and start the scan again, i am getting error that i am already connected to the device.

Here is my code, peharps you can guide me. Sorry for once more disturbing

`private fun onScanToggleClick() {
        if (isScanning) {
            if (scanDisposable != null) scanDisposable?.dispose()
        } else {
            if (rxBleClient.isScanRuntimePermissionGranted) {
                scanBleDevices()
                    .observeOn(AndroidSchedulers.mainThread())
                    .flatMap { scanResult ->
                        //scanResult.bleDevice.establishConnection(false)
                        prepareConnectionObservable(scanResult)
                    }
                    .flatMapSingle { cx ->
                        Log.e("TAG", "1")
                        cx.discoverServices()
                            .flatMap { services ->
                                Log.e("TAG", "2")
                                services.getService(UUID.fromString(AllGattServices.GENERIC_ACCESS_UUID))
                                    .map { service ->
                                        Log.e("TAG", "3")
                                        service.getCharacteristic(
                                            UUID.fromString(
                                                AllGattCharacteristics.DEVICE_NAME
                                            )
                                        )
                                    }
                                    .flatMap {

                                            characteristic ->
                                        Log.e("TAG", "4")
                                        cx.readCharacteristic(characteristic)

                                    }
                            }
                    }
                    .doFinally { 
                        dispose()
                    }
                    .subscribe({
                        Log.e("TAG", it.toString())
                        if (it != null && it.isNotEmpty()) {
                            val stringBuilder = StringBuilder(it.size)
                            for (byteChar in it) stringBuilder.append(
                                String.format(
                                    "%02X ",
                                    byteChar
                                )
                            )
                            Log.e("FULLNAME---", String(it) + "\n" + stringBuilder.toString())
                        }
                    }, {
                        Log.e(TAG, "CONNECTION TO DEVICE ERROR >> ${it.message}")
                        onScanFailure(it)
                    })

                    .let { scanDisposable = it }
            } else {
                hasClickedScan = true
                requestLocationPermission(rxBleClient)
            }
        }
        updateButtonUIState()

    }`

dariuszseweryn, 2020-01-24:
You need to dispose the connection disposable.

dononcharles, 2020-01-24:
yes. i create a function which make it 
fun dispose(){scanDisposable.dispose}

when i connected to the devices et close the connection et reconnect again, it telling me i've already connected to them. And i remark that after a certain minutes, the connection is close.

So what i want to do, is to check with the adresse mac, if the state connection is active and disconnected them.  But i do not know how to deal with. I search a lot but nothing

dononcharles, 2020-01-24:
Precision: for only one device, there is no problem