dariuszseweryn/RxAndroidBle #671: How do i get RSSI even after connected to BLE ?

manishpatelgt, 2020-03-08:
I am using 

`   implementation "com.polidea.rxandroidble2:rxandroidble:1.11.0"`

I want to get RSSI even after connected to BLE device. I saw after connected to BLE device when i scan again i don't see my connected ble device appears in the scan list. 

So is that possible to get RSSI for connected BLE device using this lib?

dariuszseweryn, 2020-03-09:
> So is that possible to get RSSI for connected BLE device using this lib?

Yes. Check `RxBleConnection` API

closed 2020-03-09 by dariuszseweryn

labteleco, 2020-03-17:
How to use it?

dariuszseweryn, 2020-03-17:
There is an API without any parameters. I am not sure what you ask about.

manishpatelgt, 2020-03-18:
This is what I am looking for

[RssiPeriodicExampleActivity](https://github.com/Polidea/RxAndroidBle/blob/master/sample/src/main/java/com/polidea/rxandroidble2/sample/example5_rssi_periodic/RssiPeriodicExampleActivity.java)

thanks @dariuszseweryn