dariuszseweryn/RxAndroidBle #672: Weird behavior establishConnection with RxJava#timeout operator.

BadClyx, 2020-03-17:
**Describe the bug**
Hi everyone, I'm getting a weird bug which I do not understand at all. I'm not really familiar with Rx and the behavior I'm observing is the exact opposite from what I learned. I hope you could help me.

**To Reproduce**
Here is the code I'm using for trying to get a TimeoutException if it has not been able to connect with 10 seconds.

```
val device = ZTBleManager.client.getBleDevice("AA:BB:CC:DD:EE:FF")
        
        device.observeConnectionStateChanges()
            .observeOn(AndroidSchedulers.mainThread())
            .subscribe { onConnectionStateChange(it) }
            .let { connectionStateChangesDisposable = it }

        device.establishConnection(false)
            .timeout(10, TimeUnit.SECONDS)
            .observeOn(AndroidSchedulers.mainThread())
            .doFinally { connection = null }
            .subscribe({ onConnectionSuccess(it) }, { onConnectionFailed(it) })
            .let { connection = it }

```
Here is the logcat output I get, As you can see the onConnectionSucess is called (which means an item has been emitted
 and should not timeout) but you see that I got the error too.

```D/BluetoothGatt: connect() - device: D8:E4:A2:F1:6A:5F, auto: false
    registerApp()
    registerApp() - UUID=3d05cead-2aed-4d03-9c79-08837da40896
D/BluetoothGatt: onClientRegistered() - status=0 clientIf=5
D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=5 device=D8:E4:A2:F1:6A:5F
D/onConnectionSuccess: Connected com.polidea.rxandroidble2.internal.connection.RxBleConnectionImpl@fa91090
D/BluetoothGatt: onConnectionUpdated() - Device=D8:E4:A2:F1:6A:5F interval=6 latency=0 timeout=500 status=0
D/BluetoothGatt: onConnectionUpdated() - Device=D8:E4:A2:F1:6A:5F interval=36 latency=0 timeout=500 status=0
D/BluetoothGatt: onConnectionUpdated() - Device=D8:E4:A2:F1:6A:5F interval=156 latency=0 timeout=400 status=0
E/onConnectionFailed: The source did not signal an event for 10 seconds and has been terminated.
D/BluetoothGatt: cancelOpen() - device: D8:E4:A2:F1:6A:5F
D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=5 device=D8:E4:A2:F1:6A:5F
D/BluetoothGatt: close()
```

**Expected behavior**
In this situation I should not get the Timeout.

**Smartphone**
 - Device: One Plus 7 Pro
 - OS: Android 10
 - Library version: 1.11.0

dariuszseweryn, 2020-03-17:
Read `.timeout()` Javadoc which is a part of RxJava library. It should explain all to you.

closed 2020-03-17 by dariuszseweryn

BadClyx, 2020-03-17:
Excuse me if it is Rx things and not appropriate to issues. I read the doc carefully. 

"The Timeout operator allows you to abort an Observable with an onError termination if that Observable fails to emit any items during a specified span of time." from the Rx Documentation.

Isn't the case ? Where establishConnection emits a RxBleConnection..

What am I missunderstanding, please I really stuck on that. Can you point me to the right direction maybe ?

dariuszseweryn, 2020-03-17:
> if that Observable fails to emit any items during a specified span of time

It is exactly as it is described:

1. Subscription
2. `RxBleConnection` emission
3. 10 seconds
4. `TimeoutException` error emission

You want to unsubscribe from `.timeout()` after the first emission from `.establishConnection()` but keep `.establishConnection()` subscribed to sustain the connection. Check `.publish()` operator — it may be handy