dariuszseweryn/RxAndroidBle #673: SetupNotification after updateBlePriority seems to fail.

aygalinc, 2020-03-19:
**Describe the bug**
I establish connection to a BLE device if I found appropriate GATT characteristic & I try to update connection priority with updateBlePriority.

And if I try to setup Notif on some characteristic, i got an error (see log). 
connection
        .setupNotification(readCharacteristicUUID)
```
D/Main: [main]BLE scan network : starting
D/BLECoapService: [main]QR code to validate : https://go2se.com/ref=EPG400L/sn=PPL20083000110
D/BLECoapService: [main]Identify : ref=EPG400L/sn=PPL20083000110
D/BLECoapService: [main]CRC16 : 0xe783
    [main]Checksum : 0x2c96
D/BLECoapService: [DefaultDispatcher-worker-1]Sending new BLE step : INIT
D/BLECoapService: [DefaultDispatcher-worker-1]QR code to validate : https://go2se.com/ref=EPG400L/sn=PPL20083000110
    [DefaultDispatcher-worker-1]Identify : ref=EPG400L/sn=PPL20083000110
    [DefaultDispatcher-worker-1]CRC16 : 0xe783
    [DefaultDispatcher-worker-1]Checksum : 0x2c96
D/Main: [main]BLE step received : INIT
D/BLECoapService: [DefaultDispatcher-worker-1]Setting BTP Fragment payload size to 179
D/RxBle#Log: Received new options (LogOptions{logLevel=null, macAddressLogSetting=null, uuidLogSetting=null, shouldLogAttributeValues=null, shouldLogScannedPeripherals=null, logger=com.polidea.rxandroidble2.internal.RxBleLog$2@ff8fd09}) and merged with old setup: LoggerSetup{logLevel=3, macAddressLogSetting=2147483647, uuidLogSetting=2147483647, shouldLogAttributeValues=false, shouldLogScannedPeripherals=true, logger=com.polidea.rxandroidble2.internal.RxBleLog$1@812ab0e}. New setup: LoggerSetup{logLevel=3, macAddressLogSetting=2147483647, uuidLogSetting=2147483647, shouldLogAttributeValues=false, shouldLogScannedPeripherals=true, logger=com.polidea.rxandroidble2.internal.RxBleLog$2@ff8fd09}
D/RxBle#Client: Received new options (LogOptions{logLevel=3, macAddressLogSetting=null, uuidLogSetting=null, shouldLogAttributeValues=null, shouldLogScannedPeripherals=null, logger=null}) and merged with old setup: LoggerSetup{logLevel=3, macAddressLogSetting=2147483647, uuidLogSetting=2147483647, shouldLogAttributeValues=false, shouldLogScannedPeripherals=true, logger=com.polidea.rxandroidble2.internal.RxBleLog$2@ff8fd09}. New setup: LoggerSetup{logLevel=3, macAddressLogSetting=2147483647, uuidLogSetting=2147483647, shouldLogAttributeValues=false, shouldLogScannedPeripherals=true, logger=com.polidea.rxandroidble2.internal.RxBleLog$2@ff8fd09}
D/BLECoapService: [DefaultDispatcher-worker-1]Sending new BLE step : SCANNING
D/Main: [main]BLE step received : SCANNING
D/RxBle#ClientOperationQueue: QUEUED   ScanOperationApi21(164279837)
D/RxBle#ClientOperationQueue: STARTED  ScanOperationApi21(164279837)
I/RxBle#ClientOperationQueue: RUNNING  ScanOperationApi21{}
I/RxBle#QueueOperation: Scan operation is requested to start.
D/RxBle#ScanOperationApi21: No library side filtering —> debug logs of scanned devices disabled
D/BluetoothAdapter: isLeEnabled(): ON
D/BluetoothLeScanner: onScannerRegistered() - status=0 scannerId=7 mScannerId=0
D/RxBle#ClientOperationQueue: FINISHED ScanOperationApi21(164279837) in 30 ms
I/RxBle#Client: ScanResult{bleDevice=RxBleDeviceImpl{MAC='XX:XX:XX:XX:XX:XX', name=null}, rssi=-44, timestampNanos=3824946457981, callbackType=CALLBACK_TYPE_ALL_MATCHES, scanRecord=[...]}
I/RxBle#Client: ScanResult{bleDevice=RxBleDeviceImpl{MAC='XX:XX:XX:XX:XX:XX', name=null}, rssi=-40, timestampNanos=3825882320834, callbackType=CALLBACK_TYPE_ALL_MATCHES, scanRecord=[...]}
I/RxBle#Client: ScanResult{bleDevice=RxBleDeviceImpl{MAC='XX:XX:XX:XX:XX:XX', name=null}, rssi=-39, timestampNanos=3826183576177, callbackType=CALLBACK_TYPE_ALL_MATCHES, scanRecord=[...]}
I/RxBle#Client: ScanResult{bleDevice=RxBleDeviceImpl{MAC='XX:XX:XX:XX:XX:XX', name=null}, rssi=-42, timestampNanos=3826403010678, callbackType=CALLBACK_TYPE_ALL_MATCHES, scanRecord=[...]}
I/RxBle#Client: ScanResult{bleDevice=RxBleDeviceImpl{MAC='XX:XX:XX:XX:XX:XX', name=null}, rssi=-41, timestampNanos=3826540074129, callbackType=CALLBACK_TYPE_ALL_MATCHES, scanRecord=[...]}
I/RxBle#Client: ScanResult{bleDevice=RxBleDeviceImpl{MAC='XX:XX:XX:XX:XX:XX', name=null}, rssi=-94, timestampNanos=3826548933921, callbackType=CALLBACK_TYPE_ALL_MATCHES, scanRecord=[...]}
I/RxBle#Client: ScanResult{bleDevice=RxBleDeviceImpl{MAC='XX:XX:XX:XX:XX:XX', name=null}, rssi=-36, timestampNanos=3826620406845, callbackType=CALLBACK_TYPE_ALL_MATCHES, scanRecord=[...]}
I/RxBle#Client: ScanResult{bleDevice=RxBleDeviceImpl{MAC='XX:XX:XX:XX:XX:XX', name=null}, rssi=-46, timestampNanos=3827142998876, callbackType=CALLBACK_TYPE_ALL_MATCHES, scanRecord=[...]}
I/RxBle#Client: ScanResult{bleDevice=RxBleDeviceImpl{MAC='XX:XX:XX:XX:XX:XX', name=null}, rssi=-40, timestampNanos=3827197521642, callbackType=CALLBACK_TYPE_ALL_MATCHES, scanRecord=[...]}
I/RxBle#Client: ScanResult{bleDevice=RxBleDeviceImpl{MAC='XX:XX:XX:XX:XX:XX', name=null}, rssi=-37, timestampNanos=3827456226095, callbackType=CALLBACK_TYPE_ALL_MATCHES, scanRecord=[...]}
I/RxBle#Client: ScanResult{bleDevice=RxBleDeviceImpl{MAC='XX:XX:XX:XX:XX:XX', name=null}, rssi=-44, timestampNanos=3827692606067, callbackType=CALLBACK_TYPE_ALL_MATCHES, scanRecord=[...]}
I/RxBle#Client: ScanResult{bleDevice=RxBleDeviceImpl{MAC='XX:XX:XX:XX:XX:XX', name=null}, rssi=-43, timestampNanos=3827930343069, callbackType=CALLBACK_TYPE_ALL_MATCHES, scanRecord=[...]}
I/RxBle#Client: ScanResult{bleDevice=RxBleDeviceImpl{MAC='XX:XX:XX:XX:XX:XX', name=null}, rssi=-42, timestampNanos=3828411712076, callbackType=CALLBACK_TYPE_ALL_MATCHES, scanRecord=[...]}
I/RxBle#Client: ScanResult{bleDevice=RxBleDeviceImpl{MAC='XX:XX:XX:XX:XX:XX', name=null}, rssi=-40, timestampNanos=3828546156673, callbackType=CALLBACK_TYPE_ALL_MATCHES, scanRecord=[...]}
I/RxBle#Client: ScanResult{bleDevice=RxBleDeviceImpl{MAC='XX:XX:XX:XX:XX:XX', name=null}, rssi=-39, timestampNanos=3829030671773, callbackType=CALLBACK_TYPE_ALL_MATCHES, scanRecord=[...]}
I/RxBle#Client: ScanResult{bleDevice=RxBleDeviceImpl{MAC='XX:XX:XX:XX:XX:XX', name=null}, rssi=-41, timestampNanos=3829090233758, callbackType=CALLBACK_TYPE_ALL_MATCHES, scanRecord=[...]}
I/zygote: Do full code cache collection, code=125KB, data=113KB
I/zygote: After code cache collection, code=119KB, data=73KB
I/RxBle#Client: ScanResult{bleDevice=RxBleDeviceImpl{MAC='XX:XX:XX:XX:XX:XX', name=null}, rssi=-41, timestampNanos=3829763728513, callbackType=CALLBACK_TYPE_ALL_MATCHES, scanRecord=[...]}
I/RxBle#Client: ScanResult{bleDevice=RxBleDeviceImpl{MAC='XX:XX:XX:XX:XX:XX', name=null}, rssi=-40, timestampNanos=3830112193235, callbackType=CALLBACK_TYPE_ALL_MATCHES, scanRecord=[...]}
I/RxBle#Client: ScanResult{bleDevice=RxBleDeviceImpl{MAC='XX:XX:XX:XX:XX:XX', name=null}, rssi=-40, timestampNanos=3830181794856, callbackType=CALLBACK_TYPE_ALL_MATCHES, scanRecord=[...]}
I/RxBle#Client: ScanResult{bleDevice=RxBleDeviceImpl{MAC='XX:XX:XX:XX:XX:XX', name=null}, rssi=-37, timestampNanos=3830988614989, callbackType=CALLBACK_TYPE_ALL_MATCHES, scanRecord=[...]}
I/RxBle#Client: ScanResult{bleDevice=RxBleDeviceImpl{MAC='XX:XX:XX:XX:XX:XX', name=null}, rssi=-47, timestampNanos=3831402831905, callbackType=CALLBACK_TYPE_ALL_MATCHES, scanRecord=[...]}
I/RxBle#Client: ScanResult{bleDevice=RxBleDeviceImpl{MAC='XX:XX:XX:XX:XX:XX', name=null}, rssi=-42, timestampNanos=3831900687163, callbackType=CALLBACK_TYPE_ALL_MATCHES, scanRecord=[...]}
I/RxBle#Client: ScanResult{bleDevice=RxBleDeviceImpl{MAC='XX:XX:XX:XX:XX:XX', name=null}, rssi=-39, timestampNanos=3831910114612, callbackType=CALLBACK_TYPE_ALL_MATCHES, scanRecord=[...]}
I/RxBle#Client: ScanResult{bleDevice=RxBleDeviceImpl{MAC='XX:XX:XX:XX:XX:XX', name=null}, rssi=-91, timestampNanos=3831911077581, callbackType=CALLBACK_TYPE_ALL_MATCHES, scanRecord=[...]}
I/RxBle#Client: ScanResult{bleDevice=RxBleDeviceImpl{MAC='XX:XX:XX:XX:XX:XX', name=null}, rssi=-92, timestampNanos=3831920368936, callbackType=CALLBACK_TYPE_ALL_MATCHES, scanRecord=[...]}
I/RxBle#Client: ScanResult{bleDevice=RxBleDeviceImpl{MAC='XX:XX:XX:XX:XX:XX', name=null}, rssi=-92, timestampNanos=3831942714823, callbackType=CALLBACK_TYPE_ALL_MATCHES, scanRecord=[...]}
I/RxBle#Client: ScanResult{bleDevice=RxBleDeviceImpl{MAC='XX:XX:XX:XX:XX:XX', name=null}, rssi=-92, timestampNanos=3831951248418, callbackType=CALLBACK_TYPE_ALL_MATCHES, scanRecord=[...]}
I/RxBle#Client: ScanResult{bleDevice=RxBleDeviceImpl{MAC='XX:XX:XX:XX:XX:XX', name=null}, rssi=-46, timestampNanos=3832516035297, callbackType=CALLBACK_TYPE_ALL_MATCHES, scanRecord=[...]}
[... cut ...]
        at android.bluetooth.BluetoothGatt.runOrQueueCallback(BluetoothGatt.java:725)
        at android.bluetooth.BluetoothGatt.-wrap0(Unknown Source:0)
        at android.bluetooth.BluetoothGatt$1.onDescriptorWrite(BluetoothGatt.java:533)
        at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:172)
        at android.os.Binder.execTransact(Binder.java:697)
I/RxBle#GattCallback: MAC='XX:XX:XX:XX:XX:XX'        onDescriptorWrite(), status=1, value=[uuid='...']
```

**Additional context**
Add any other context about the problem here.

dariuszseweryn, 2020-03-19:
It seems that the handle is invalid. Could you show the code? Is your device bonded? Does it change its services/characteristics?

aygalinc, 2020-03-19:
No bonding requested. 

We call this : 
'''
 btpConnectionDisposable.add(device
          .establishBTPConnection(false, options.isNeedBonding(), options.getOperationTimeout(),
              options.getUnit(),
              BtpUUIDNetworkReference.COAP_BTP_PRIMARY_SERVICE_UUID.getUuid(),
              BtpUUIDNetworkReference.COAP_BTP_WRITE_CHARACTERISTIC_UUID.getUuid(),
              BtpUUIDNetworkReference.COAP_BTP_READ_CHARACTERISTIC_UUID.getUuid(), true)
          .flatMap(btpConnectionBis -> btpConnectionBis
              .updateBlePriority(options.getConnectionPriority())
              .andThen(Observable.just(btpConnectionBis)))
          .subscribe(
              btpConnection -> {
                Timber.tag(LogReference.COAP_BTP_TAG)
                    .d("CoAP endpoint succeed with BTP Connection");
                endpoint.setBtpConnection(btpConnection);
                emitter.onNext(endpoint);
              }....
'''

which delegate connection to our fragmentation layer : 

'''
public Observable<BtpConnection> establishBTPConnection(boolean autoConnect, boolean needBonding,
      long operationTimeout, @NonNull TimeUnit unit, UUID serviceUUID, UUID writeUUID,
      UUID readUUID, boolean isWriteWithBtpEnable) {
    Timber.tag(LogReference.BTP_TAG).d("Try to establish BTP connection");
    return bondWithDevice(needBonding)
        .andThen(
            device.establishConnection(autoConnect, new Timeout(operationTimeout, unit))
                .flatMap(
                    connectedDevice -> discoverBtpService(connectedDevice, serviceUUID, writeUUID,
                        readUUID,
                        isWriteWithBtpEnable))
        );
'''

which check our fragmentation service : 
'''
return Observable.defer(() ->
        connection.discoverServices()
            .flatMapObservable(
                serviceDiscover -> {
                  if (serviceUUID == null || writeUUID == null || readUUID == null) {
                    return Observable
                        .error(new BtpConnectionException("One of the characteristic is null !"));
                  }
                  List<BluetoothGattService> gattService = serviceDiscover
                      .getBluetoothGattServices();
                  BluetoothGattCharacteristic writeCharacteristic = checkExistingGATTService(
                      gattService, serviceUUID, writeUUID, readUUID);
                  if (writeCharacteristic != null) {
                    return Observable.just(
                        new BtpConnectionImpl(connection, writeCharacteristic, readUUID,
                            isWriteWithBtpEnable, options.getFragmentSize()));
                  } else {
                    return Observable.error(new BtpConnectionException(
                        "Cannot find all Btp GATT services & characteristics"));
                  }
                }
            )
    );
'''

and when we create a BTPConnection , we create a multicast stream : 
'''
 mBtpReadResultObservable = connection
        .setupNotification(readCharacteristicUUID)
        .doOnNext(ignored -> Timber.tag(LogReference.BTP_TAG)
            .d("Notifications has been set up successfully "))
        .flatMap(notifObservable -> notifObservable)
        .map(arrayOfByte -> {
          Timber.tag(LogReference.BTP_TAG).d("BTP Read Action triggered, fragment received");
          BtpFragment fragment = new BtpFragment(arrayOfByte);
          return frameBuffer.addFragment(fragment);
        })
        .share(); // Multicast the read stream
'''

That we serve to upper layer at some point. 
So when upper layer subscribe to this stream for the first time it seems to raise an error. 

Checking service not modify anything : 
'''
private static BluetoothGattCharacteristic checkExistingGATTService(
      List<BluetoothGattService> gattServices, UUID serviceUUID, UUID writeUUID, UUID readUUID) {
    boolean btpBleServiceFind = false;
    boolean btpWriteServiceFind = false;
    boolean btpReadServiceFind = false;

    BluetoothGattCharacteristic writeCharacteristic = null;

    for (BluetoothGattService service : gattServices) {
      if (serviceUUID.equals(service.getUuid())) {
        Timber.tag(LogReference.BTP_TAG).d("Btp Ble primary GATT service find");
        btpBleServiceFind = true;
        for (BluetoothGattCharacteristic btpCharacteristic : service.getCharacteristics()) {
          if (readUUID.equals(btpCharacteristic.getUuid())) {
            Timber.tag(LogReference.BTP_TAG).d("Btp Read characteristic find");
            btpReadServiceFind = true;
          }
          if (writeUUID.equals(btpCharacteristic.getUuid())) {
            Timber.tag(LogReference.BTP_TAG).d("Btp Write characteristic find");
            writeCharacteristic = btpCharacteristic;
            btpWriteServiceFind = true;
          }
        }
      }
    }

    if (btpBleServiceFind && btpReadServiceFind && btpWriteServiceFind) {
      return writeCharacteristic;
    }

    return null;
  }
'''

aygalinc, 2020-03-19:
Should I update connectionPriority before service discovery ?

dariuszseweryn, 2020-03-19:
You can try. I wonder if this works if you do not change connection priority

aygalinc, 2020-03-19:
I will try and keep you informed.

dariuszseweryn, 2020-03-19:
Please also give info on what phone / OS are you testing

aygalinc, 2020-03-19:
A moto G5 & android 8.1

aygalinc, 2020-03-22:
After changing my fragmentation layer that sends an extra empty I cannot reproduce it. 
So maybe it was located on edge device side. 
Thx for the quick answer.

closed 2020-03-22 by aygalinc