dariuszseweryn/RxAndroidBle #674: "Read and write combined" example broken

ChrisSDC, 2020-03-19:
**Describe the bug**
The "[Read and write combined](https://github.com/Polidea/RxAndroidBle#read-and-write-combined)" example appears to be build-broken.

[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Copy/paste example
2. Set up appropriate variables

**Setup:**
 - Library version: 1.5.0 & 1.11.0
 - Android Studio version: 3.6.1
 - JDK version 1_8 (Java 8)

**Additional context**
I'm very new to ReactiveX so this could be a very simple syntax issue or even a setup issue on my part (apologies if that's the case), but a fix will be useful for other noobs such as myself.

dariuszseweryn, 2020-03-19:
Yup. It should be `.doOnSuccess()`. Will fix Today

dariuszseweryn, 2020-03-24:
@ChrisSDC Thank you for this report. It is now fixed.

PR #675 (2020-03-24): Fixed Combined Read and Write snippet
It was a leftover from RxJava 1 times.
Fixes #674
closed 2020-03-24 by dariuszseweryn