dariuszseweryn/RxAndroidBle #677: Trying to write to multiple characteristics

Tharunreddych, 2020-03-27:
**Describe the bug**
Trying to write to multiple characteristics

**To Reproduce**
I was following this example where they mentioned you only get rxBleConnection in the fist flatMap
`mConnObservable.flatMap(rxBleConnection -> {
        return rxBleConnection.writeCharacteristic(SSID, mSsidView.getText().toString().getBytes())
                .flatMap(ssidBytes -> rxBleConnection.writeCharacteristic(SSID2, mPassPhrase.getText().toString().getBytes())
                .flatMap(ssid2Bytes -> rxBleConnection.writeCharacteristic(SSID3, mSecurityModeSpinner.getSelectedItem().toString().getBytes())));
    })
            .observeOn(AndroidSchedulers.mainThread())
            .subscribe(ssid3Bytes -> {
               //do something
            }, this::onError, this::onComplete);`

**Expected behavior**
I should be able to write to multiple characteristics, but this is resulting in an error.
`no instance(s) of type variable(s) R exist so that Single<R> conforms to ObservableSource<? extends R>`

**Smartphone (please complete the following information):**
Generic

Based on me searching google, I think that the interfaces have been changed with the rxandroidble2 and so the example might be old. It would be great if you someone can help me out.

RobLewis, 2020-03-28:
Where is the example you're following?

Tharunreddych, 2020-03-29:
Hi,
I was trying to follow this example: 

https://stackoverflow.com/questions/43667549/rxandroidble-multiple-characteristic-notifications-and-read-write

dariuszseweryn, 2020-04-03:
The compiler shows you what is the problem. It is not a library issue. I have edited the answer in SO link you posted above so it should be now compatible with RxJava2 library version.

Best Regards

closed 2020-04-03 by dariuszseweryn