dariuszseweryn/RxAndroidBle #678: Unable to get response while using readCharacteristic method

harrisdas1, 2020-04-02:
**Describe the bug**
I'm using this library to read characteristic data (characteristic is readable) but unable to read that, an exception is thrown

**To Reproduce**
1. Connect to a BLE device
2. Try to read characteristic

**Expected behavior**
Successfully read data from the characteristic.
 
**Device Information**
 - Device: [Google Pixel 2, Moto G6]
 - OS: [ Android 10, Android 9]
 - Library version: [1.11.0]

**Code snippet**

```

    private final CompositeDisposable compositeDisposable = new CompositeDisposable();
    private Observable<RxBleConnection> rxBleConnectionObservable;
    private PublishSubject<Boolean> disconnectTriggerSubject = PublishSubject.create();

    @Override
    public void connect(String macAddress, ConnectCallback callback) {
        try {
            RxBleDevice bleDevice = bleClient.getBleDevice(macAddress);

            rxBleConnectionObservable = bleDevice.establishConnection(false)
                    .takeUntil(disconnectTriggerSubject)
                    .compose(ReplayingShare.instance());

            Disposable connectionDisposable = rxBleConnectionObservable
                    .flatMapSingle(RxBleConnection::discoverServices)
                    .subscribe(response ->
                    {
                        Log.d(TAG, "connection successful");
                    }, error -> {
                        disposeConnectionObservable();
                        Log.e(TAG, "connection Failed", error);
                    });
            compositeDisposable.add(connectionDisposable);

        } catch (Exception e) {
            Log.d(TAG, "connection exception: " + e.getMessage());
            disposeConnectionObservable();
        }

    }
    @Override
    public void readCharacteristic() {
        Log.d(TAG, "readCharacteristic start ");
        Disposable disposable2 = rxBleConnectionObservable
                .firstOrError()
                .flatMap(connection ->
                        connection.readCharacteristic(USER_CHARACTERISTIC_RETURN_STATUS_UUID))
                .doFinally(() -> {
                    Log.d(TAG, "readCharacteristic disposed");
                }).subscribe(data -> {
                    Log.d(TAG, "readCharacteristic data: " + Arrays.toString(data));
                }, error -> {
                    Log.d(TAG, "readCharacteristic Failed" + error.getMessage());
                });
        compositeDisposable.add(disposable2);
    }

  private void disposeConnectionObservable() {
        Log.d(TAG, "dispose connection");
       compositeDisposable.clear();
}
```
```
D/TCConnection: readCharacteristic start 
D/RxBle#ConnectionOperationQueue: QUEUED   CharacteristicReadOperation(68247933)
D/RxBle#ConnectionOperationQueue: STARTED  CharacteristicReadOperation(68247933)
I/RxBle#ConnectionOperationQueue: RUNNING  CharacteristicReadOperation{MAC='80:6F:B0:17:D8:D9', characteristic=[uuid='df863e4c-4e2c-43e5-ac66-5b9c6d76b141']}
D/TCConnection: readCharacteristic FailedGATT exception from MAC address 80:6F:B0:17:D8:D9, with type BleGattOperation{description='CHARACTERISTIC_READ'}
    readCharacteristic disposed
D/RxBle#ConnectionOperationQueue: FINISHED CharacteristicReadOperation(68247933) in 30008 ms
W/BluetoothGatt: Unhandled exception in callback
    java.lang.NullPointerException: Attempt to get length of null array
        at java.lang.StringFactory.newStringFromChars(StringFactory.java:253)
        at java.lang.String.valueOf(String.java:2938)
        at com.polidea.rxandroidble2.internal.logger.LoggerUtil.bytesToHex(LoggerUtil.java:22)
        at com.polidea.rxandroidble2.internal.logger.LoggerUtil$AttributeLogWrapper.toString(LoggerUtil.java:225)
        at java.util.Formatter$FormatSpecifier.printString(Formatter.java:2978)
        at java.util.Formatter$FormatSpecifier.print(Formatter.java:2855)
        at java.util.Formatter.format(Formatter.java:2524)
        at java.util.Formatter.format(Formatter.java:2459)
        at java.lang.String.format(String.java:2870)
        at com.polidea.rxandroidble2.internal.RxBleLog.formatString(RxBleLog.java:166)
        at com.polidea.rxandroidble2.internal.RxBleLog.throwShade(RxBleLog.java:214)
        at com.polidea.rxandroidble2.internal.RxBleLog.i(RxBleLog.java:186)
        at com.polidea.rxandroidble2.internal.logger.LoggerUtil.logCallback(LoggerUtil.java:102)
        at com.polidea.rxandroidble2.internal.connection.RxBleGattCallback$2.onCharacteristicRead(RxBleGattCallback.java:109)
        at android.bluetooth.BluetoothGatt$1$6.run(BluetoothGatt.java:394)
        at android.bluetooth.BluetoothGatt.runOrQueueCallback(BluetoothGatt.java:780)
        at android.bluetooth.BluetoothGatt.access$200(BluetoothGatt.java:41)
        at android.bluetooth.BluetoothGatt$1.onCharacteristicRead(BluetoothGatt.java:388)
        at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:246)
        at android.os.Binder.execTransactInternal(Binder.java:1021)
        at android.os.Binder.execTransact(Binder.java:994)
D/BluetoothGatt: onClientConnectionState() - status=22 clientIf=8 device=80:6F:B0:17:D8:D9
I/RxBle#GattCallback: MAC='80:6F:B0:17:D8:D9'  onConnectionStateChange(), status=22, value=0
D/RxBle#ConnectionOperationQueue: Connection operations queue to be terminated (MAC='80:6F:B0:17:D8:D9')
    com.polidea.rxandroidble2.exceptions.BleDisconnectedException: Disconnected from MAC='80:6F:B0:17:D8:D9' with status 22 (GATT_CONN_TERMINATE_LOCAL_HOST)
        at com.polidea.rxandroidble2.internal.connection.RxBleGattCallback$2.onConnectionStateChange(RxBleGattCallback.java:81)
        at android.bluetooth.BluetoothGatt$1$4.run(BluetoothGatt.java:272)
        at android.bluetooth.BluetoothGatt.runOrQueueCallback(BluetoothGatt.java:780)
        at android.bluetooth.BluetoothGatt.access$200(BluetoothGatt.java:41)
        at android.bluetooth.BluetoothGatt$1.onClientConnectionState(BluetoothGatt.java:267)
        at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:192)
        at android.os.Binder.execTransactInternal(Binder.java:1021)
        at android.os.Binder.execTransact(Binder.java:994)
V/RxBle#Executors: Terminated (MAC='80:6F:B0:17:D8:D9')
D/TCConnection: dispose connection
D/com.trasnscore.sample.ui.tagssearch.TagsSearchFragment: onConnectionError Disconnected from MAC='80:6F:B0:17:D8:D9' with status 22 (GATT_CONN_TERMINATE_LOCAL_HOST)
E/TCConnection: onService subscribe Failed
    com.polidea.rxandroidble2.exceptions.BleDisconnectedException: Disconnected from MAC='80:6F:B0:17:D8:D9' with status 22 (GATT_CONN_TERMINATE_LOCAL_HOST)
        at com.polidea.rxandroidble2.internal.connection.RxBleGattCallback$2.onConnectionStateChange(RxBleGattCallback.java:81)
        at android.bluetooth.BluetoothGatt$1$4.run(BluetoothGatt.java:272)
        at android.bluetooth.BluetoothGatt.runOrQueueCallback(BluetoothGatt.java:780)
        at android.bluetooth.BluetoothGatt.access$200(BluetoothGatt.java:41)
        at android.bluetooth.BluetoothGatt$1.onClientConnectionState(BluetoothGatt.java:267)
        at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:192)
        at android.os.Binder.execTransactInternal(Binder.java:1021)
        at android.os.Binder.execTransact(Binder.java:994)
D/RxBle#ClientOperationQueue: QUEUED   DisconnectOperation(58773868)
D/RxBle#ClientOperationQueue: STARTED  DisconnectOperation(58773868)
I/RxBle#ClientOperationQueue: RUNNING  DisconnectOperation{MAC='80:6F:B0:17:D8:D9'}
D/BluetoothGatt: close()
    unregisterApp() - mClientIf=8
D/RxBle#ClientOperationQueue: FINISHED DisconnectOperation(58773868) in 9 ms

```

This exception is thrown after 30 seconds.

dariuszseweryn, 2020-04-02:
Hmmm... Indeed seems to be a bug. I have not encountered a case where characteristic value is `null`. I would like to check the log after the fix will be released.

closed 2020-04-02 by dariuszseweryn

dariuszseweryn, 2020-04-02:
This is now fixed in version `1.11.1`. Please attach the logs after the fix so I could learn on what exact scenario you have faced

harrisdas1, 2020-04-03:
Thanks for replying, Here is the recent logs

```
D/TCConnection: readProperty start 
D/RxBle#ConnectionOperationQueue: QUEUED   CharacteristicReadOperation(190480201)
D/RxBle#ConnectionOperationQueue: STARTED  CharacteristicReadOperation(190480201)
I/RxBle#ConnectionOperationQueue: RUNNING  CharacteristicReadOperation{MAC='80:6F:B0:17:D8:D9', characteristic=[uuid='df863e4c-4e2c-43e5-ac66-5b9c6d76b141']}
D/TCConnection: readCharacteristic FailedGATT exception from MAC address 80:6F:B0:17:D8:D9, with type BleGattOperation{description='CHARACTERISTIC_READ'}
D/TCConnection: readCharacteristic disposed
D/RxBle#ConnectionOperationQueue: FINISHED CharacteristicReadOperation(190480201) in 30006 ms
I/RxBle#GattCallback: MAC='80:6F:B0:17:D8:D9'     onCharacteristicRead(), status=133, value=[uuid='df863e4c-4e2c-43e5-ac66-5b9c6d76b141', hexValue=null]
D/BluetoothGatt: onClientConnectionState() - status=22 clientIf=7 device=80:6F:B0:17:D8:D9
I/RxBle#GattCallback: MAC='80:6F:B0:17:D8:D9'  onConnectionStateChange(), status=22, value=0
D/RxBle#ConnectionOperationQueue: Connection operations queue to be terminated (MAC='80:6F:B0:17:D8:D9')
    com.polidea.rxandroidble2.exceptions.BleDisconnectedException: Disconnected from MAC='80:6F:B0:17:D8:D9' with status 22 (GATT_CONN_TERMINATE_LOCAL_HOST)
        at com.polidea.rxandroidble2.internal.connection.RxBleGattCallback$2.onConnectionStateChange(RxBleGattCallback.java:81)
        at android.bluetooth.BluetoothGatt$1$4.run(BluetoothGatt.java:272)
        at android.bluetooth.BluetoothGatt.runOrQueueCallback(BluetoothGatt.java:780)
        at android.bluetooth.BluetoothGatt.access$200(BluetoothGatt.java:41)
        at android.bluetooth.BluetoothGatt$1.onClientConnectionState(BluetoothGatt.java:267)
        at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:192)
        at android.os.Binder.execTransactInternal(Binder.java:1021)
        at android.os.Binder.execTransact(Binder.java:994)
D/TCConnection: dispose connection
D/com.trasnscore.sample.ui.tagssearch.TagsSearchFragment: onConnectionError Disconnected from MAC='80:6F:B0:17:D8:D9' with status 22 (GATT_CONN_TERMINATE_LOCAL_HOST)
V/RxBle#Executors: Terminated (MAC='80:6F:B0:17:D8:D9')
E/TCConnection: onService subscribe Failed
    com.polidea.rxandroidble2.exceptions.BleDisconnectedException: Disconnected from MAC='80:6F:B0:17:D8:D9' with status 22 (GATT_CONN_TERMINATE_LOCAL_HOST)
        at com.polidea.rxandroidble2.internal.connection.RxBleGattCallback$2.onConnectionStateChange(RxBleGattCallback.java:81)
        at android.bluetooth.BluetoothGatt$1$4.run(BluetoothGatt.java:272)
        at android.bluetooth.BluetoothGatt.runOrQueueCallback(BluetoothGatt.java:780)
        at android.bluetooth.BluetoothGatt.access$200(BluetoothGatt.java:41)
        at android.bluetooth.BluetoothGatt$1.onClientConnectionState(BluetoothGatt.java:267)
        at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:192)
        at android.os.Binder.execTransactInternal(Binder.java:1021)
        at android.os.Binder.execTransact(Binder.java:994)
D/RxBle#ClientOperationQueue: QUEUED   DisconnectOperation(203602315)
D/RxBle#ClientOperationQueue: STARTED  DisconnectOperation(203602315)
I/RxBle#ClientOperationQueue: RUNNING  DisconnectOperation{MAC='80:6F:B0:17:D8:D9'}
D/BluetoothGatt: close()
D/BluetoothGatt: unregisterApp() - mClientIf=7
D/RxBle#ClientOperationQueue: FINISHED DisconnectOperation(203602315) in 11 ms
```

dariuszseweryn, 2020-04-03:
So I guess the library bug is fixed. I cannot really tell you why your peripheral does not respond to a read request.

PR #679 (2020-04-02): Fixed logging attributes with null value
This situation seems to happen when a characteristic is first read with a non-success status.

Fixes #678