dariuszseweryn/RxAndroidBle #680: Establish connection to an already connected device

githubMoko, 2020-04-03:
I'm in the process of replacing the ble code of an app of mine (standard Android libraries) with RxAndroidBle.
There is a particular case that it seems to be not covered by this library.
I got a brodcast from the OS that a device is connected from the standard BLE panel of Android.
I got the BluetoothDevice but i don't have the GattCallback since it was connected outside of my app. So i just call device.connectGatt to get my callback. This just works and i can link my callback to the device.
In RxAndroidBLE running  rxBleClient.getBleDevice(device.address).establishConnection throws an already connection exception and i can't retrieve rxBleConnection.
I'm not saying that this is a bug, but i need to find a workaround to get past this limitation.

dariuszseweryn, 2020-04-03:
The library works in scope of the application. If the OS broadcasts that a device is connected and library emits `BleAlreadyConnectedException` it means that _your app_ has already connected to the peripheral or at least an approach was made that was not cancelled.

githubMoko, 2020-04-03:
Ok clear! thanks

closed 2020-04-03 by dariuszseweryn