dariuszseweryn/RxAndroidBle #682: How to read full length of data when my data contain 0x0a and 0x0d?

BennyLWK, 2020-04-14:
**Describe the bug**
I am able to connect to the device correctly and it can receive the characteristic correctly without any errors. But it gets something wrong when the notify data contain 0x0a and 0x0d.

For example, the device sends the data as 0b710a6000179c21346fb23feec651bc to the mobile app. The app will receive the message as below. The app split the data into 2 data.

Change = 0B71
Change = 6000179C21346FB23FEEC651BC

**Expected behavior**
Change = 0B710A6000179C21346FB23FEEC651BC0D

**Smartphone:**
 - Device: Samsung Galaxy A70
 - OS: Android 10
 - Library version:  1.11.1

**Additional context**
[image]

dariuszseweryn, 2020-04-14:
Have you tried for instance `nRF Connect` app and check if it behaves the same way? I suppose so. If so then it is a problem with your peripheral.

BennyLWK, 2020-04-15:
Hi @dariuszseweryn thanks for your reply. I tried nRF Connect app before and it behaves the same way. Let me check with my hardware. Is there any application for testing other than nRF Connect?

dariuszseweryn, 2020-04-15:
> Is there any application for testing other than nRF Connect?
Any application that covers your peripheral is good for testing. They all rely on Android OS BLE stack implementation anyway.

I do not get it — you have tested two applications that gave you the same result but you still filled a bug here?

BennyLWK, 2020-04-15:
Hi @dariuszseweryn, since I not familiar with hardware and my hardware colleague insists on the hardware no issue. So I try to find out more info about BLE. Thanks for your support.

closed 2020-04-15 by BennyLWK