dariuszseweryn/RxAndroidBle #683: Not possible to start scan

ramlok, 2020-04-21:
**Describe the bug**
Some phones(Huawei P30, ..) running Android 10 have problem to start BLE scan. 

**To Reproduce**
just call 
`rxBleClient.scanBleDevices(
                                ScanSettings.Builder()
                                    .setScanMode(ScanSettings.SCAN_MODE_BALANCED)
                                    .build(),
                                ScanFilter.Builder()
                                    .setServiceUuid(ParcelUuid(UUID.fromString(Constants.UUID_SERVICE_BATTERY)))
                                    .build()
                            )`

**Expected behavior**
A clear and concise description of what you expected to happen.

**Smartphone (please complete the following information):**
 - Huawei P30
 - OS Android 10
 - Library version: 1.11.1

**Logs from the application when bug occurs (this will greatly help in quick understanding the problem)**
No logs

**Additional context**
When the scan is stareted an Bluetooth is tunred off and on again the scan will deliver ScanResults.
The same applies when during scan phone is locked and unlocked
https://stackoverflow.com/questions/61315872/rxble-start-scan-does-not-work-on-android-10-huawei-p30

dariuszseweryn, 2020-04-21:
Hello,

Thank you for using the library.
Your post seems to be a bug report but it lacks some crucial information. Could you edit to contain information according to the [issue template](https://github.com/Polidea/RxAndroidBle/blob/master/.github/ISSUE_TEMPLATE.md)?

Here you state that scan does not start but on stackoverflow it does not start immediately but only after certain conditions happen. Please add reproduction steps and logs from the application — maybe then it will be possible to help you.

Best regards

ramlok, 2020-04-21:
Thank you for quick resposne.
As you can see in Additional commetn, there is exactly the same description as in stackoverflow.
Repro step:
1. turn on BT and location
2. start application
3. start scan

in this case the state is READY, but no scan result are delivered.
There are no logs from appliaction containing BLE scan issue(I do not have access to Huawei P30 at the moment). 
What is maybe important, I am using rxble library **INSIDE** my library

dariuszseweryn, 2020-04-21:
Without logs I cannot be sure if `RxAndroidBle` is called at all.

dariuszseweryn, 2020-04-28:
@ramlok I am closing this issue as I do not have any actionable info. Feel free to add new things here and I will reopen.

closed 2020-04-28 by dariuszseweryn

arcode-pl, 2020-10-23:
I have too this Bug ;( on Huawei P30 Lite

z3ntu, 2020-12-18:
The service uuid filter seems broken on Huawei P30 Lite with latest RxAndroidBle (1.11.1)