dariuszseweryn/RxAndroidBle #684: Background Proccess

chandrabezzo, 2020-04-22:
When I connect, write, and notify using backgorund service, connection running well but when I closed the application connection between service and BLE device lost.

dariuszseweryn, 2020-04-22:
I believe you want to learn about how Android OS manages processes, when it closes them and what are foreground services.

chandrabezzo, 2020-04-22:
I was learn about Android background proccess using Work Manager, IntentService, and Service. When I use GPS on background proccess, the GPS still connected with service. But when I use Bluetooth on background proccess using this library to connect BLE device, connection lost when application closed.

dariuszseweryn, 2020-04-22:
It all depends on how you implement the service. If the service is started and in foreground then the connection should persist until the OS kills the service process.
If you close the application all application processes are killed thus ending the connection (I may be oversimplicating in some details here)

chandrabezzo, 2020-04-23:
That resolved when I used foreground service, thank you.

closed 2020-04-23 by chandrabezzo