dariuszseweryn/RxAndroidBle #686: .establishConnection() simply disconnects

Omievee, 2020-04-25:
Hi There,

I'm retrieving the bluetooth device from my receiver & then passing it along extracting the MAC address as such:
 
`    fun handleBTD(exampleBT: BluetoothDevice) {
        println("MAC ADDRESS --- ${exampleBT?.address}")
        blueDisp =
            rx.getBleDevice(exampleBT.address)
                .establishConnection(true, Timeout(5, TimeUnit.MINUTES)) // <-- autoConnect flag
                .doOnError {
                    println("error... ${it.printStackTrace()}")
                }
                .subscribe(
                    { rxBleConnection ->
                        println("connection? $rxBleConnection")
                    },
                    { throwable ->
                        println("ERROR : ${throwable.printStackTrace()}")
                    }
                )

        blueDisp?.dispose()
    }`

but as soon as this function rungs, the debugger tells me its been disconnected w/ 
`com.polidea.rxandroidble2.exceptions.BleDisconnectedException: Disconnected from MAC='XX:XX:XX:XX:XX:XX' with status -1 (UNKNOWN)`

 - Device: Pixel 4XL
 - OS: Android 10
 - Library version: latest library version 1.11.1

Full Log:
`D/RxBle#ClientOperationQueue: QUEUED   ConnectOperation(54373468)
D/RxBle#ClientOperationQueue: STARTED  ConnectOperation(54373468)
I/RxBle#ClientOperationQueue: RUNNING  ConnectOperation{MAC='AC:63:BE:C0:AE:92', autoConnect=true}
D/RxBle#ClientOperationQueue: FINISHED ConnectOperation(54373468) in 1 ms
D/RxBle#ConnectionOperationQueue: Connection operations queue to be terminated (MAC='AC:63:BE:C0:AE:92')
    com.polidea.rxandroidble2.exceptions.BleDisconnectedException: Disconnected from MAC='AC:63:BE:C0:AE:92' with status -1 (UNKNOWN)
        at com.polidea.rxandroidble2.internal.serialization.ConnectionOperationQueueImpl.onConnectionUnsubscribed(ConnectionOperationQueueImpl.java:162)
        at com.polidea.rxandroidble2.internal.connection.ConnectorImpl$1$1.run(ConnectorImpl.java:65)
        at io.reactivex.internal.operators.observable.ObservableDoFinally$DoFinallyObserver.runFinally(ObservableDoFinally.java:142)
        at io.reactivex.internal.operators.observable.ObservableDoFinally$DoFinallyObserver.dispose(ObservableDoFinally.java:98)
        at io.reactivex.internal.disposables.DisposableHelper.dispose(DisposableHelper.java:124)
        at io.reactivex.internal.operators.observable.ObservableSubscribeOn$SubscribeOnObserver.dispose(ObservableSubscribeOn.java:73)
        at io.reactivex.internal.operators.observable.ObservableUnsubscribeOn$UnsubscribeObserver$DisposeTask.run(ObservableUnsubscribeOn.java:95)
        at io.reactivex.internal.schedulers.ScheduledDirectTask.call(ScheduledDirectTask.java:38)
        at io.reactivex.internal.schedulers.ScheduledDirectTask.call(ScheduledDirectTask.java:26)
        at java.util.concurrent.FutureTask.run(FutureTask.java:266)
        at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:301)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
        at java.lang.Thread.run(Thread.java:919)
D/RxBle#ClientOperationQueue: QUEUED   DisconnectOperation(202317882)
D/RxBle#ClientOperationQueue: STARTED  DisconnectOperation(202317882)
I/RxBle#ClientOperationQueue: RUNNING  DisconnectOperation{MAC='AC:63:BE:C0:AE:92'}
W/RxBle#QueueOperation: Disconnect operation has been executed but GATT instance was null - considering disconnected.
D/RxBle#ClientOperationQueue: FINISHED DisconnectOperation(202317882) in 1 ms`
**Additional context**

I'm trying to connect to a basic Amazon Echo speaker. Not sure if this matters.

closed 2020-04-25 by Omievee

dariuszseweryn, 2020-04-26:
Hello @Omievee — could you describe why you have closed this issue? What was the cause?

tank777, 2020-07-22:
@dariuszseweryn  Same issue Here is the log 
```
com.polidea.rxandroidble2.exceptions.BleDisconnectedException: Disconnected from MAC='FC:F5:C4:0D:E4:86' with status -1 (UNKNOWN)
        at com.polidea.rxandroidble2.internal.serialization.ConnectionOperationQueueImpl.onConnectionUnsubscribed(ConnectionOperationQueueImpl.java:162)
        at com.polidea.rxandroidble2.internal.connection.ConnectorImpl$1$1.run(ConnectorImpl.java:65)
        at io.reactivex.internal.operators.observable.ObservableDoFinally$DoFinallyObserver.runFinally(ObservableDoFinally.java:142)
        at io.reactivex.internal.operators.observable.ObservableDoFinally$DoFinallyObserver.dispose(ObservableDoFinally.java:98)
        at io.reactivex.internal.disposables.DisposableHelper.dispose(DisposableHelper.java:124)
        at io.reactivex.internal.operators.observable.ObservableSubscribeOn$SubscribeOnObserver.dispose(ObservableSubscribeOn.java:73)
        at io.reactivex.internal.operators.observable.ObservableUnsubscribeOn$UnsubscribeObserver$DisposeTask.run(ObservableUnsubscribeOn.java:95)
        at io.reactivex.internal.schedulers.ScheduledDirectTask.call(ScheduledDirectTask.java:38)
        at io.reactivex.internal.schedulers.ScheduledDirectTask.call(ScheduledDirectTask.java:26)
        at java.util.concurrent.FutureTask.run(FutureTask.java:266)
        at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:301)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
        at java.lang.Thread.run(Thread.java:764)
```
Step to reproduce the bug. 
1. Connect with device 
2. Call dispose
3. Now again try to connect same device. 

Note: It will work fine if we will close the application and re open it.

Omievee, 2020-07-22:
@tank777 For me it was using the library set up wrong. Reading through the repo & seeing the examples on how to use it helped me figure out my problem. Im not sure what the unknown -1 error is but id suggest reading through the examples as I did.

dariuszseweryn, 2020-07-23:
```
D/RxBle#ConnectionOperationQueue: Connection operations queue to be terminated (MAC='AC:63:BE:C0:AE:92')
    com.polidea.rxandroidble2.exceptions.BleDisconnectedException: Disconnected from MAC='AC:63:BE:C0:AE:92' with status -1 (UNKNOWN)
        at com.polidea.rxandroidble2.internal.serialization.ConnectionOperationQueueImpl.onConnectionUnsubscribed(ConnectionOperationQueueImpl.java:162)
        at com.polidea.rxandroidble2.internal.connection.ConnectorImpl$1$1.run(ConnectorImpl.java:65)
        at io.reactivex.internal.operators.observable.ObservableDoFinally$DoFinallyObserver.runFinally(ObservableDoFinally.java:142)
        at io.reactivex.internal.operators.observable.ObservableDoFinally$DoFinallyObserver.dispose(ObservableDoFinally.java:98)
        at io.reactivex.internal.disposables.DisposableHelper.dispose(DisposableHelper.java:124)
        at io.reactivex.internal.operators.observable.ObservableSubscribeOn$SubscribeOnObserver.dispose(ObservableSubscribeOn.java:73)
        at io.reactivex.internal.operators.observable.ObservableUnsubscribeOn$UnsubscribeObserver$DisposeTask.run(ObservableUnsubscribeOn.java:95)
        at io.reactivex.internal.schedulers.ScheduledDirectTask.call(ScheduledDirectTask.java:38)
        at io.reactivex.internal.schedulers.ScheduledDirectTask.call(ScheduledDirectTask.java:26)
        at java.util.concurrent.FutureTask.run(FutureTask.java:266)
        at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:301)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
        at java.lang.Thread.run(Thread.java:919)
```
This is a debug info only. No need to panic. :)
If it would be received in subscription (with unknown reason and no other underlying cause) then I suppose it could be a bug.