dariuszseweryn/RxAndroidBle #690: RxJava 3.x support

AndHain, 2020-05-18:
The [RxJava docs](https://github.com/ReactiveX/RxJava) mention the following:

> The 2.x version is in maintenance mode and will be supported only through bugfixes until February 28, 2021. No new features, behavior changes or documentation adjustments will be accepted or applied to 2.x.

Are there any plans for e.g. rxandroidble3?

Or will the rxandroidble2 be compatible with the RxJava 3.x version?

dariuszseweryn, 2020-05-21:
There are plans to have `rxandroidble3` — these were blocked by dependencies which only had been updated [a day after your post](https://github.com/JakeWharton/RxRelay/releases/tag/rxrelay-3.0.0).

`rxandroidble2` cannot be compatible with RxJava 3 since the plans for ABI compatibility between RxJava 2 and 3 were canceled.

RobLewis, 2020-05-21:
If I understand, RxJava3 is mostly backward-compatible with RxJava2, right? So hopefully it won't be a huge job to migrate RxAndroidBle.

dariuszseweryn, 2020-05-22:
I have started work on migrating to RxJava 3 when it was a release candidate at the end of July 2019. AFAIR it was working quite well up to a point where dependencies were not updated correctly. I do not expect huge amount of work now.

GabrielAraujo, 2020-07-06:
@dariuszseweryn do you have any early access dependency for this? I'm migrating my project and I would just like to build it even if the feature who uses this dependency is a little glitched

dariuszseweryn, 2020-07-08:
@GabrielAraujo There is only `rxjava3` branch now which was not updated since RC times. I do not have enough time to finish it yet — so no, there is nothing workable now and you would need to use [a bridge](https://github.com/akarnokd/RxJavaBridge) for now.

Maragues, 2021-02-25:
👋🏻  [RxJava 2.2.21](https://github.com/ReactiveX/RxJava/releases/tag/v2.2.21) is End of Life release for RxJava2

FYI, we were able to migrate in ~2 days. I assume it will take more for this project, since it's a public library and you need to release a major version, since inevitably there'll be some public interface change.

alexvo87, 2021-05-07:
@dariuszseweryn I think migrating to RxJava3 will fix this https://github.com/Polidea/RxAndroidBle/wiki/FAQ:-UndeliverableException and "GATT_INSUF_AUTHORIZATION or GATT_CONN_TIMEOUT"

sandy-8925, 2021-08-07:
@dariuszseweryn - Is there any further progress planned for this? Do you need some help with it?

dariuszseweryn, 2021-08-08:
I am short on time at the moment. I first plan to make the lib usable for Android 12, then proceed with other things

dariuszseweryn, 2022-06-13:
After merging #793 RxJava 3 based dependency is now on Maven Central under `com.polidea.rxandroidble3:rxandroidble:1.15.0`. I am sorry for such a long wait and would like to thank you all for the support. RxJava 2 based dependency is still supported for the time being under the usual `com.polidea.rxandroidble2:rxandroidble:1.15.0`. Cheers!

closed 2022-06-13 by dariuszseweryn