dariuszseweryn/RxAndroidBle #692: Failed to read encrypted characteristic bonded device

Dachmian, 2020-05-26:
**Describe the bug**
I am having issues reading encrypted characteristics for a bonded bluetooth device.
When connecting to the device it is only advertising 2 services but by writing to one of the exposed characteristics, it sends a Services Changed Indication and loads the remaining Characteristics.

The phone recognises the services changed indication and the remaining services and characteristics are discoverable using the discoverServices() method.
However, when reading the newly loaded characteristics after successful discovery I either get:

I/RxBle#GattCallback: MAC=MAC     onCharacteristicRead(), status=1, value=[uuid=UUID hexValue=null]

or: 
OnErrorNotification[com.polidea.rxandroidble2.exceptions.BleGattCharacteristicException: GATT exception from MAC=MAC, status 1 (GATT_INVALID_HANDLE), type BleGattOperation{description='CHARACTERISTIC_READ'}. (Look up status 0x01 here https://android.googlesource.com/platform/external/bluetooth/bluedroid/+/android-5.1.0_r1/stack/include/gatt_api.h)]

errors.

I thought this was due to something wrong going on with the discovery and I used the suggestion here: https://github.com/Polidea/RxAndroidBle/pull/453#issuecomment-459787554
to refresh cache. However I am still unable to read the characteristics that are loaded during connection. But this time I get another error:

W/BluetoothGatt: onCharacteristicRead() failed to find characteristic!

even though discoverServices() shows that it has discovered the characteristics I am trying to read.

If I connect to the device via my app and then go into LightBlue BLE app, I can browse all services and characteristics and read them successfully. 

Please let me know what kind of logs would be helpful or suggestions for steps I should try to overcome this issue.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
Able to read newly loaded characteristics

**Smartphone (please complete the following information):**
 - Device: [e.g. Huawei P20]
 - OS: [e.g. Android 8.1.0]
 - Library version: [e.g. 1.11.1]

**Logs from the application when bug occurs (this will greatly help in quick understanding the problem)**
To turn on logs use:
```
RxBleClient.updateLogOptions(new LogOptions.Builder()
        .setLogLevel(LogConstants.DEBUG)
        .setMacAddressLogSetting(LogConstants.MAC_ADDRESS_FULL)
        .setUuidsLogSetting(LogConstants.UUIDS_FULL)
        .setShouldLogAttributeValues(true)
        .build()
);
```
```
// please paste the logs here
```

**Additional context**
Add any other context about the problem here.

Dachmian, 2020-05-26:
When loading custom firmware to my bluetooth device which has all the services and characteristics exposed solves the issue. I am able to interact with all the services and characteristics so it excludes the bonding / encryption of the characteristics to be the issue.

dariuszseweryn, 2020-05-26:
I do not have access to any peripheral that behaves like you described. You could also share how you have modelled your interaction with the peripheral. I do not expect invalid handle exception if you used `BluetoothGattCharacteristic` API directly. `UUID` API could use cached `BluetoothGattCharacteristic` under the hood.
A log of the connection could also help me understand what have happened.

Dachmian, 2020-05-26:
Hello @dariuszseweryn

I have a RxBleManager class which holds the RxBleClient, a BehaviorRelay which exposes the ble connection and ble device, and contains all the api calls the app uses. 
I also have a connection module which contains the connection logic which scans for devices, establishes connection, sends necessary commands and then exposes the ble connection and ble device to the RxBleManager Relay.

Pseudo code for connection procedure in the connection module:

val scannedDevice = rxBleManager. scanForDevices(services)

val connectViaScan = scannedDevice
            .flatMap { scannedDevice ->
                scannedDevice.scannedPeripheral.bleDevice.establishConnection(true)
                    .doOnError { error -> error.localizedMessage?.let { logE(it) } }
                    .delay(1, TimeUnit.SECONDS)
                    .map{Pair(scannedDevice.scannedPeripheral.bleDevice, it)}
            }
.flatMap{ /* Interact with open characteristics which loads the new services*/}

connectViaScan.subscribe{scannedDevice ->   rxBleManager.bleRelay.accept(scannedDevice)
}

RxBleManager Code Snippets

fun scanForDevices(withServices: List<AdvertisedServices>): Observable<ScannedDevice>{
        return rxBleClient.observeStateChanges().startWith(rxBleClient.state)
            .filter { it == RxBleClient.State.READY }
            .flatMap {
                retrievePeripherals(
                    withServices.map{it.serviceUuid}
                )
            }
    }

fun retrievePeripherals(uuids: List<String>): Observable<ScanResult> {
        val scanSettings =
            ScanSettings.Builder().setScanMode(ScanSettings.SCAN_MODE_BALANCED).build()
        val listOfScanFilter = mutableListOf<ScanFilter>()
        uuids.forEach {
            listOfScanFilter.add(ScanFilter.Builder().setServiceUuid(ParcelUuid.fromString(it)).build())
        }
        return rxBleClient.scanBleDevices(scanSettings, *listOfScanFilter.toTypedArray())
    }

fun readCharacteristic(characteristicIdentifier: BluetoothGattCharacteristic): Observable<ByteArray> {
        return bleRelay.map { it.second }.take(1).flatMap { rxBleConnection ->
            rxBleConnection.readCharacteristic(characteristicIdentifier).toObservable()
        }
    }

    fun readCharacteristicUuid(uuid: UUID): Observable<ByteArray> {
        return bleRelay.map { it.second }.take(1).flatMap { rxBleConnection ->
            rxBleConnection.readCharacteristic(uuid).toObservable()
        }
    }

Then later inside the app I try to read certain characteristics using either of the convenience methods I defined but all of them returns the errors mentioned.

rxBleManager.readCharacteristic(uuid).subscribe()
rxBleManager.readCharacteristic(characteristic).subscribe()

I will run the tests again and produce the logs shortly. Thank you for the quick answer!

Dachmian, 2020-05-26:
To read using the `BluetoothGattCharacteristic ` API directly I took your example from this comment: https://github.com/Polidea/RxAndroidBle/issues/635#issuecomment-552877862

Dachmian, 2020-06-03:
@dariuszseweryn I compiled a small test project to show the issue I am facing.

The code I am running is:

´´´
    private fun runTestApp(){
        val rxBleClient = RxBleClient.create(this.applicationContext)

        RxBleClient.updateLogOptions(
            LogOptions.Builder()
                .setLogLevel(LogConstants.DEBUG)
                .setMacAddressLogSetting(LogConstants.MAC_ADDRESS_FULL)
                .setUuidsLogSetting(LogConstants.UUIDS_FULL)
                .setShouldLogAttributeValues(true)
                .build()
        )

        val disposable = rxBleClient.observeStateChanges()
            .startWith(rxBleClient.state)
            .filter { it == RxBleClient.State.READY }
            .flatMap {
                rxBleClient.scanBleDevices(
                    ScanSettings.Builder().setScanMode(ScanSettings.SCAN_MODE_LOW_LATENCY).build(),
                    *mutableListOf<ScanFilter>(ScanFilter.Builder().setServiceUuid(ParcelUuid.fromString("0000fFf0-0000-1000-8000-00805f9b34fb")).build()).toTypedArray()
                )
            }
            .filter{it.bleDevice.macAddress == deviceMacAddress}
            .distinctUntilChanged{t1, t2 ->  t1.bleDevice.macAddress == t2.bleDevice.macAddress}
            .delay(3, TimeUnit.SECONDS)
            .flatMap{scanResult ->
                scanResult.bleDevice.establishConnection(false).map{Pair(scanResult.bleDevice, it)}
            }
            .delay(3, TimeUnit.SECONDS)
            .flatMap{data ->
                data.second.queue(bluetoothGattRefreshCustomOp).ignoreElements()
                    .andThen(data.second.queue(discoverServicesCustomOp))
                    .doAfterNext { Log.d("First Services from custom refresh","services: ${it.map{it.characteristics.map{it.uuid.toString()}}}")}
                    .toFlowable(BackpressureStrategy.BUFFER).toObservable().map{data}
            }
            .delay(3, TimeUnit.SECONDS)
            .flatMap{data ->
                data.second.discoverServices().doAfterSuccess { Log.d("First Services from RxAndroidBle","services: ${it.bluetoothGattServices.map{it.characteristics.map{it.uuid.toString()}}}")}.toObservable().map{data}
            }
            .delay(3, TimeUnit.SECONDS)
            .flatMap{data ->
                data.second.readCharacteristic(
                    UUID.fromString("0000fef2-0000-1000-8000-00805f9b34fb")
                ).toObservable().map{data}
            }
            .delay(3, TimeUnit.SECONDS)
            .flatMap{data ->
                data.second.writeCharacteristic(
                    UUID.fromString("0000fef2-0000-1000-8000-00805f9b34fb"),
                    byteArrayOf(0x39,0x33,0x31,0x61,0x64,0x61,0x34,0x37,0x45,0x30,0x32,0x41,0x33,0x42,0x39,0x44)
                ).toObservable().map{data}
            }
            .delay(3, TimeUnit.SECONDS)
            .flatMap{data ->
                data.second.writeCharacteristic(
                    UUID.fromString("0000fef1-0000-1000-8000-00805f9b34fb"),
                    byteArrayOf(0x01)
                ).toObservable().map{data}
            }
            .delay(3, TimeUnit.SECONDS)
            .flatMap{data ->
                data.second.queue(bluetoothGattRefreshCustomOp).ignoreElements()
                    .andThen(data.second.queue(discoverServicesCustomOp))
                    .delay(3, TimeUnit.SECONDS)
                    .flatMapSingle{services ->
                        Log.d("Second Services from custom refresh","services: ${services.map{it.characteristics.map{it.uuid.toString()}}}")
[... 175 lines cut ...]
        at com.polidea.rxandroidble2.internal.serialization.ConnectionOperationQueueImpl.onConnectionUnsubscribed(ConnectionOperationQueueImpl.java:162)
        at com.polidea.rxandroidble2.internal.connection.ConnectorImpl$1$1.run(ConnectorImpl.java:65)
        at io.reactivex.internal.operators.observable.ObservableDoFinally$DoFinallyObserver.runFinally(ObservableDoFinally.java:142)
        at io.reactivex.internal.operators.observable.ObservableDoFinally$DoFinallyObserver.dispose(ObservableDoFinally.java:98)
        at io.reactivex.internal.disposables.DisposableHelper.dispose(DisposableHelper.java:124)
        at io.reactivex.internal.operators.observable.ObservableSubscribeOn$SubscribeOnObserver.dispose(ObservableSubscribeOn.java:73)
        at io.reactivex.internal.operators.observable.ObservableUnsubscribeOn$UnsubscribeObserver$DisposeTask.run(ObservableUnsubscribeOn.java:95)
        at io.reactivex.internal.schedulers.ScheduledDirectTask.call(ScheduledDirectTask.java:38)
        at io.reactivex.internal.schedulers.ScheduledDirectTask.call(ScheduledDirectTask.java:26)
        at java.util.concurrent.FutureTask.run(FutureTask.java:266)
        at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:301)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1162)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:636)
        at java.lang.Thread.run(Thread.java:784)
D/BluetoothManager: getConnectionState()
D/BluetoothManager: getConnectedDevices
D/BluetoothGatt: cancelOpen() - device: 54:6C:0E:A0:44:45
D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=9 device=54:6C:0E:A0:44:45
I/RxBle#GattCallback: MAC='54:6C:0E:A0:44:45'  onConnectionStateChange(), status=0, value=0
I/Process: Sending signal. PID: 12413 SIG: 9
Disconnected from the target VM, address: 'localhost:8620', transport: 'socket'

You can clearly see the new service "fee0" after it has been unlocked. However when trying to read using the connection I get an error. It happens at this line:

´return@flatMapSingle data.second.readCharacteristic(services[0].characteristics[0].uuid)´

Dachmian, 2020-06-17:
I managed to solve the issue by storing the characteristics discovered by the custom gatt refresh operation and use this for BLE interactions. Will use this as a workaround now.

closed 2020-06-17 by Dachmian