dariuszseweryn/RxAndroidBle #696: Information on reaching OS Bluetooth connection limit

dariuszseweryn, 2020-06-08:
**Is your feature request related to a problem? Please describe.**
[Original post](https://github.com/Polidea/RxAndroidBle/issues/688#issuecomment-640531222).  

**Describe the solution you'd like**
It would be awesome if the library would warn the user whenever the OS reaches maximum established Bluetooth connections.

**Describe alternatives you've considered**
Maybe not an alternative but a doubt as maximum Bluetooth connections may vary not only by API level but from vendor to vendor or even from chip manufacturer to another which may decrease usability of such a feature.

**Additional context**
According to this SO post the maximum OS level Bluetooth connections is limited by `MAX_L2CAP_LINKS` value which is:
- [13 on Android 10.0](https://cs.android.com/android/platform/superproject/+/android-10.0.0_r30:system/bt/internal_include/bt_target.h;l=477)
- [13 on Android 9.0](https://cs.android.com/android/platform/superproject/+/android-9.0.0_r1:system/bt/internal_include/bt_target.h;l=484)
- [7 on Android 8.0](https://cs.android.com/android/platform/superproject/+/android-8.0.0_r1:system/bt/include/bt_target.h;l=510)
- [7 on Android 5.1](https://cs.android.com/android/platform/superproject/+/android-5.1.1_r38:external/bluetooth/bluedroid/include/bt_target.h;l=1127)
- [7 on Android 4.4](https://cs.android.com/android/platform/superproject/+/android-4.4.4_r2:external/bluetooth/bluedroid/include/bt_target.h;l=1227)
- [7 on Android 4.3](https://cs.android.com/android/platform/superproject/+/android-4.3.1_r1:external/bluetooth/bluedroid/include/bt_target.h;l=1219)

remonh87, 2020-07-23:
@dariuszseweryn  that is a good idea however I know from experience that for some Huawei (e.g. P20) and Xaomi devices the threshold is lower( I even saw a Android 9 phone that had issues when connecting more then 6) , so your doubt is indeed justified :). However a warning with this disclaimer can still be good one or add it to the FAQ.

dariuszseweryn, 2020-07-23:
There are two things:
1. The OS limits
2. The actual Bluetooth Chip limits — there are numerous Bluetooth Chip manufacturers there and capabilities of the chips and firmwares that those chips run can be widely different. YMMV :(