dariuszseweryn/RxAndroidBle #697: Reading most characteristics are, reading Blood Pressure Measurement fails, why?

smkarber, 2020-06-08:
I can read almost all characteristics on my Omron BP7000 Blood Pressure device, except the actual Blood Pressure Measurement.

The code I have is: 

`val found = gattCharacteristics.any { t -> t.uuid ==
                UUID.fromString("00002a35-0000-1000-8000-00805f9b34fb") }

        if (found) {
            val u: UUID = UUID.fromString(CommonUUID.BloodPressureMeasurement.uuid)
            val v: UUID = UUID.fromString(CommonUUID.BloodPressureFeature.uuid)
            deviceConnection =
                device.establishConnection(true)
                    .flatMap {
                        Log.d("SETUP NOTIFICATION", "Setting notification now.")
                        it.setupNotification(u)
                    }
                    .observeOn(AndroidSchedulers.mainThread())
                    .take(1)
                    .subscribe(
                        { o->
                            Log.d("SETUP SUBSCRIPTION", "Setting subscription now.")
                            o.observeOn(AndroidSchedulers.mainThread())
                                .take(1)
                                .subscribe(
                                    {
                                        Log.d("HANDLE OBSERVATION", "Handling observation now.")
                                      for (c in characteristics) {
                                          if (c.uuid == CommonUUID.BloodPressureMeasurement.uuid) {
                                              c.value = processByteArray(c.uuid, it)
                                          }
                                          serviceDetailsAdapter.notifyDataSetChanged()
                                      }
                                    },
                                    {
                                        onConnectionFailure(it)
                                    }
                                )
                        },
                        {

                        }
                    )
        } else {
            deviceConnection =
                device.establishConnection(true)
                    .flatMapSingle {
                        for (g in gattCharacteristics) {
                            singles.add(it.readCharacteristic(g.uuid))
                        }
                        Single.zip(
                            singles
                        ) { varargs ->
                            val values: MutableList<ByteArray> = mutableListOf()
                            for (v in varargs) {
                                values.add(v as ByteArray)
                            }
                            return@zip values
                        }
                    }
                    .observeOn(AndroidSchedulers.mainThread())
                    .take(1)
                    .subscribe(
                        {
                            characteristics.forEachIndexed { index, c ->
                                c.value = processByteArray(c.uuid, it[index])
                            }
                            serviceDetailsAdapter.notifyDataSetChanged()
                        },
                        {
                            onConnectionFailure(it)
[... 41 lines cut ...]
2020-06-08 15:34:32.709 23561-23600/com.optum.ect.ble.hd D/BluetoothGatt: onSearchComplete() = Device=EC:21:E5:34:E7:5E Status=0
2020-06-08 15:34:32.710 23561-23600/com.optum.ect.ble.hd I/RxBle#GattCallback: MAC='EC:21:E5:34:E7:5E'     onServicesDiscovered(), status=0
2020-06-08 15:34:32.714 23561-23686/com.optum.ect.ble.hd W/RxBle#IllegalOperationChecker: Characteristic 00002a35-0000-1000-8000-00805f9b34fb supports properties: [ INDICATE ] (32) does not have any property matching [ NOTIFY ] (16)
2020-06-08 15:34:32.717 23561-23685/com.optum.ect.ble.hd I/RxBle#ConnectionOperationQueue: RUNNING  DescriptorWriteOperation{MAC='EC:21:E5:34:E7:5E', descriptor=[uuid='00002902-0000-1000-8000-00805f9b34fb', hexValue=[01, 00]]}
2020-06-08 15:34:32.737 23561-23600/com.optum.ect.ble.hd I/RxBle#GattCallback: MAC='EC:21:E5:34:E7:5E'        onDescriptorWrite(), status=3, value=[uuid='00002902-0000-1000-8000-00805f9b34fb']
2020-06-08 15:34:32.739 23561-23686/com.optum.ect.ble.hd D/BluetoothGatt: setCharacteristicNotification() - uuid: 00002a35-0000-1000-8000-00805f9b34fb enable: false
2020-06-08 15:34:32.740 23561-23685/com.optum.ect.ble.hd I/RxBle#ConnectionOperationQueue: RUNNING  DescriptorWriteOperation{MAC='EC:21:E5:34:E7:5E', descriptor=[uuid='00002902-0000-1000-8000-00805f9b34fb', hexValue=[00, 00]]}
2020-06-08 15:34:32.742 23561-23612/com.optum.ect.ble.hd I/RxBle#ClientOperationQueue: RUNNING  DisconnectOperation{MAC='EC:21:E5:34:E7:5E'}
2020-06-08 15:34:32.743 23561-23685/com.optum.ect.ble.hd W/RxBle#ConnectionOperationQueue: Queue's awaitRelease() has been interrupted abruptly while it wasn't released by the release() method.
    java.lang.InterruptedException
        at java.lang.Object.wait(Native Method)
        at java.lang.Object.wait(Object.java:442)
        at java.lang.Object.wait(Object.java:568)
        at com.polidea.rxandroidble2.internal.serialization.QueueSemaphore.awaitRelease(QueueSemaphore.java:15)
        at com.polidea.rxandroidble2.internal.serialization.ConnectionOperationQueueImpl$1.run(ConnectionOperationQueueImpl.java:74)
        at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:462)
        at java.util.concurrent.FutureTask.run(FutureTask.java:266)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
        at java.lang.Thread.run(Thread.java:919)
2020-06-08 15:34:32.744 23561-23687/com.optum.ect.ble.hd D/BluetoothGatt: cancelOpen() - device: EC:21:E5:34:E7:5E
2020-06-08 15:34:32.745 23561-23600/com.optum.ect.ble.hd D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=9 device=EC:21:E5:34:E7:5E
2020-06-08 15:34:32.746 23561-23600/com.optum.ect.ble.hd I/RxBle#GattCallback: MAC='EC:21:E5:34:E7:5E'  onConnectionStateChange(), status=0, value=0
2020-06-08 15:34:32.747 23561-23687/com.optum.ect.ble.hd D/BluetoothGatt: close()
2020-06-08 15:34:32.747 23561-23687/com.optum.ect.ble.hd D/BluetoothGatt: unregisterApp() - mClientIf=9`

dariuszseweryn, 2020-06-08:
Have you read the logs?

smkarber, 2020-06-08:
Yes, my thought is that the issue is it wants an INDICATION instead of a NOTIFICATION, but I read in one of the older issues that you had/were switching it to auto-select whichever it needed. Or was that never implemented?

dariuszseweryn, 2020-06-08:
No — it would require new API which was not created yet

smkarber, 2020-06-08:
Oh, I understand. So if I change it to an Indication instead, I don't get that first error, but it still hits the InterruptedException.

smkarber, 2020-06-08:
Is that also something I am doing wrong?

dariuszseweryn, 2020-06-08:
Yes, you prematurely dispose the connection and newly setup notification/indication

smkarber, 2020-06-08:
When it sets up the notification, a popup appears asking me to pair, then it errors. Maybe its hitting onPause() which calls deviceConnection.dispose() when the popup appears.

smkarber, 2020-06-08:
So I removed the `.take(1)` calls from the setup notification branch of my if statement and it goes through without issue, but it still never gets here:

`
.subscribe(
                                    {
                                        Log.d("HANDLE OBSERVATION", "Handling observation now.")
                                      for (c in characteristics) {
                                          if (c.uuid == CommonUUID.BloodPressureMeasurement.uuid) {
                                              c.value = processByteArray(c.uuid, it)
                                          }
                                          serviceDetailsAdapter.notifyDataSetChanged()
                                      }
                                    },
                                    {
                                        onConnectionFailure(it)
                                    }
                                )
`

dariuszseweryn, 2020-06-09:
It is most likely because your peripheral does not send anything

smkarber, 2020-06-09:
Does it require something I don't have in that code above to get data from an indication? After I set the indication, does the first data come through automatically, or do I have to acknowledge it first?

smkarber, 2020-06-09:
Is there an example of how to use an indication in the RxAndroidBle documentation?

dariuszseweryn, 2020-06-09:
Notifications and indications are treated by the application code absolutely the same way. There is no app side ACK whatsoever.

You have to consult your peripheral specification for details.

smkarber, 2020-06-09:
Thanks for the help. I'll see if I can figure this out.

closed 2020-06-09 by smkarber