dariuszseweryn/RxAndroidBle #699: Possible data loss when enabling notifications

froqstar, 2020-06-23:
**Describe the bug**
When enabling notifications on a characteristic on the mobile, and the peripheral starts notifying changes right after receiving the descriptor write (in the same connection interval?), the mobile may miss the first notification(s) sent by the peripheral.

**To Reproduce**
Steps to reproduce the behavior:
1. An iPhone is acting as BLE peripheral.
2. The Android phone connects to the iPhone and enables notifications on a characteristic. 
3. The iPhone starts notifying changes on that characteristic immediately after receiving the descriptor write.
4. The first notification sent by the iPhone is not received by the app running on the Android phone.
5. Following notifications are received by the app running on the Android phone.

**Expected behavior**
The Android phone should receive all notifications from the peripheral.

**Smartphone (please complete the following information):**
 - Device: Samsung Galaxy S9
 - OS: Android 9
 - Library version: 1.11.1

- Device: iPhone X
- OS: iOS 13.2

```
2020-06-22 16:43:40.475 18745-18869/com.bosch.ebike.onebikeapp D/RxBle#ClientOperationQueue: QUEUED   ConnectOperation(4806158)
2020-06-22 16:43:40.476 18745-18813/com.bosch.ebike.onebikeapp D/RxBle#ClientOperationQueue: STARTED  ConnectOperation(4806158)
2020-06-22 16:43:40.477 18745-18813/com.bosch.ebike.onebikeapp I/RxBle#ClientOperationQueue: RUNNING  ConnectOperation{MAC='61:98:FF:76:C5:DC', autoConnect=true}
2020-06-22 16:43:40.493 18745-18858/com.bosch.ebike.onebikeapp D/BluetoothAdapter: STATE_ON
2020-06-22 16:43:40.495 18745-18858/com.bosch.ebike.onebikeapp D/BluetoothGatt: connect() - device: 61:98:FF:76:C5:DC, auto: true
2020-06-22 16:43:40.495 18745-18858/com.bosch.ebike.onebikeapp D/BluetoothAdapter: isSecureModeEnabled
2020-06-22 16:43:40.495 18745-18858/com.bosch.ebike.onebikeapp D/BluetoothGatt: registerApp()
2020-06-22 16:43:40.495 18745-18858/com.bosch.ebike.onebikeapp D/BluetoothGatt: registerApp() - UUID=8ba9c8e6-8e02-4bce-9af6-18d1695ff809
2020-06-22 16:43:40.499 18745-18813/com.bosch.ebike.onebikeapp D/RxBle#ClientOperationQueue: FINISHED ConnectOperation(4806158) in 23 ms
2020-06-22 16:43:40.501 18745-18759/com.bosch.ebike.onebikeapp D/BluetoothGatt: onClientRegistered() - status=0 clientIf=11
2020-06-22 16:43:44.680 18745-18759/com.bosch.ebike.onebikeapp D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=11 device=61:98:FF:76:C5:DC
2020-06-22 16:43:44.718 18745-18759/com.bosch.ebike.onebikeapp I/RxBle#GattCallback: MAC='61:98:FF:76:C5:DC'  onConnectionStateChange(), status=0, value=2
2020-06-22 16:43:44.755 18745-18810/com.bosch.ebike.onebikeapp D/RxBle#ConnectionOperationQueue: QUEUED   MtuRequestOperation(9489980)
2020-06-22 16:43:44.757 18745-18868/com.bosch.ebike.onebikeapp D/RxBle#ConnectionOperationQueue: STARTED  MtuRequestOperation(9489980)
2020-06-22 16:43:44.760 18745-18868/com.bosch.ebike.onebikeapp I/RxBle#ConnectionOperationQueue: RUNNING  MtuRequestOperation{MAC='61:98:FF:76:C5:DC', mtu=517}
2020-06-22 16:43:44.768 18745-18858/com.bosch.ebike.onebikeapp D/BluetoothGatt: configureMTU() - device: 61:98:FF:76:C5:DC mtu: 517
2020-06-22 16:43:45.662 18745-18759/com.bosch.ebike.onebikeapp D/BluetoothGatt: onConfigureMTU() - Device=61:98:FF:76:C5:DC mtu=517 status=0
2020-06-22 16:43:45.672 18745-18759/com.bosch.ebike.onebikeapp I/RxBle#GattCallback: MAC='61:98:FF:76:C5:DC'             onMtuChanged(), status=0, value=517
2020-06-22 16:43:45.723 18745-18868/com.bosch.ebike.onebikeapp D/RxBle#ConnectionOperationQueue: FINISHED MtuRequestOperation(9489980) in 929 ms
2020-06-22 16:43:45.759 18745-18759/com.bosch.ebike.onebikeapp D/BluetoothGatt: onPhyUpdate() - status=0 address=61:98:FF:76:C5:DC txPhy=2 rxPhy=2
2020-06-22 16:43:45.817 18745-18822/com.bosch.ebike.onebikeapp D/RxBle#ConnectionOperationQueue: QUEUED   ServiceDiscoveryOperation(212000114)
2020-06-22 16:43:45.828 18745-18868/com.bosch.ebike.onebikeapp D/RxBle#ConnectionOperationQueue: STARTED  ServiceDiscoveryOperation(212000114)
16:56
2020-06-22 16:43:45.839 18745-18868/com.bosch.ebike.onebikeapp I/RxBle#ConnectionOperationQueue: RUNNING  ServiceDiscoveryOperation{MAC='61:98:FF:76:C5:DC'}
2020-06-22 16:43:45.854 18745-18858/com.bosch.ebike.onebikeapp D/BluetoothGatt: discoverServices() - device: 61:98:FF:76:C5:DC
2020-06-22 16:43:45.867 18745-18759/com.bosch.ebike.onebikeapp D/BluetoothGatt: onSearchComplete() = Device=61:98:FF:76:C5:DC Status=0
2020-06-22 16:43:45.887 18745-18759/com.bosch.ebike.onebikeapp I/RxBle#GattCallback: MAC='61:98:FF:76:C5:DC'     onServicesDiscovered(), status=0
2020-06-22 16:43:45.915 18745-18869/com.bosch.ebike.onebikeapp D/RxBle#ConnectionOperationQueue: QUEUED   DescriptorWriteOperation(24408441)
2020-06-22 16:43:45.920 18745-18868/com.bosch.ebike.onebikeapp D/RxBle#ConnectionOperationQueue: FINISHED ServiceDiscoveryOperation(212000114) in 97 ms
2020-06-22 16:43:45.925 18745-18868/com.bosch.ebike.onebikeapp D/RxBle#ConnectionOperationQueue: STARTED  DescriptorWriteOperation(24408441)
2020-06-22 16:43:45.931 18745-18868/com.bosch.ebike.onebikeapp I/RxBle#ConnectionOperationQueue: RUNNING  DescriptorWriteOperation{MAC='61:98:FF:76:C5:DC', descriptor=[uuid='00002902-0000-1000-8000-00805f9b34fb', hexValue=[01, 00]]}
2020-06-22 16:43:46.008 18745-18759/com.bosch.ebike.onebikeapp I/RxBle#GattCallback: MAC='61:98:FF:76:C5:DC'        onDescriptorWrite(), status=0, value=[uuid='00002902-0000-1000-8000-00805f9b34fb']
2020-06-22 16:43:46.021 18745-18869/com.bosch.ebike.onebikeapp D/BluetoothGatt: setCharacteristicNotification() - uuid: 424f5343-4820-4d43-5350-20204d49534f enable: true
2020-06-22 16:43:46.027 18745-18868/com.bosch.ebike.onebikeapp D/RxBle#ConnectionOperationQueue: FINISHED DescriptorWriteOperation(24408441) in 97 ms
2020-06-22 16:43:46.057 18745-18759/com.bosch.ebike.onebikeapp I/RxBle#GattCallback: MAC='61:98:FF:76:C5:DC'  onCharacteristicChanged(), value=[uuid='424f5343-4820-4d43-5350-20204d49534f', hexValue=[10, 06, 02, 01, 00, 00, 27, 10, 10, 06, 02, 02, 00, 00, 27, 10, 10, 06, 02, 03, 00, 00, 27, 10, 10, 06, 02, 04, 00, 00, 27, 10, 10, 06, 02, 05, 00, 00, 27, 10, 10, 06, 02, 06, 00, 00, 27, 10, 10, 06, 02, 07, 00, 00, 27, 10]]
```
The data received here is the *second* notification sent by the iPhone.

**Root cause**

This problem seems to be caused by the RxAndroidBle library only enabling notifications on the local characteristic when the descriptor write has finished. 
However, if a notification is already delivered within the same connection interval, the library will miss that notification because notifications are not yet fully enabled locally. 

This may be a samsung-only bug, we didn't test yet with other phones.

**Fix**

To fix this bug, I inserted a custom operation *before* the call to `setupNotification()` that just calls `BluetoothGatt.setCharacteristicNotification()` on the respective characteristic.
With this in place, all notifications are received.

dariuszseweryn, 2020-06-23:
Check [`RxBleConnection.setupNotification(UUID/BluetoothGattCharacteristic, SetupMode)`](https://github.com/Polidea/RxAndroidBle/blob/cb2690615c6c8babd6e1a0736473c8c429f6de04/rxandroidble/src/main/java/com/polidea/rxandroidble2/RxBleConnection.java#L296) with `SetupMode.QUICK_SETUP`

closed 2020-06-23 by dariuszseweryn

froqstar, 2020-06-23:
Seems to work :+1: 
Thanks a lot.