dariuszseweryn/RxAndroidBle #700: Can't get connection for a peripheral connected from OS

nateridderman-lilly, 2020-06-24:
Similar to #680, I'm running into problems when the user connects to a peripheral via the OS settings. We can detect these peripherals in the app with [`BluetoothManager#getConnectedDevices()`](https://developer.android.com/reference/android/bluetooth/BluetoothManager.html#getConnectedDevices(int)) and would then like to connect to them via RxAndroidBle and query some characteristics. But the Android API returns a `BluetoothDevice` and it seems there is no way to get acquire a `RxBleConnection` connection via the library. The best we can do is tell the user to disconnect and reconnect via the app and this is not a good UX. 

If we try calling `establishConnection()` then we get a BleAlreadyConnectedException, even though we didn't connect through the app. I believe the comment on #680 is wrong.

Is it possible to get a `RxBleConnection` for a device that's already connected via the OS?

Side note: I'm not sure if it's possible to connect to previously paired or unpaired BLE devices via the OS settings, but it's definitely possible to pair and this establishes a connection. I did not dive deep into all of the scenarios. I saw some talk about this in previous tickets as well.

dariuszseweryn, 2020-06-24:
Hello,
Thanks for using the library.
> I believe the comment on #680 is wrong.

Unfortunately this shouldn't be something to believe in. The code is fully auditable, please dive in and find if there is a bug there.

> Is it possible to get a RxBleConnection for a device that's already connected via the OS?

Yes, you need to connect to it inside the app.

nateridderman-lilly, 2020-06-25:
Thanks for the reply. I will investigate this further, hopefully in the next few weeks.

dariuszseweryn, 2020-06-25:
Please do. [Here](https://github.com/Polidea/RxAndroidBle/blob/cb2690615c6c8babd6e1a0736473c8c429f6de04/rxandroidble/src/main/java/com/polidea/rxandroidble2/internal/RxBleDeviceImpl.java#L84) is the code where a `BleAlreadyConnectedException` is created.

In the meantime I will close this issue and mark it as `awaiting feedback`. Please come back with more info when available and I will reopen it.

Best regards

closed 2020-06-25 by dariuszseweryn