dariuszseweryn/RxAndroidBle #701: com.polidea.rxandroidble2.exceptions.BleAlreadyConnectedException: Already connected to device with MAC address

SeloSlav, 2020-07-09:
**Problem:**

I'm listening for a notification using the following code:

```
bleDevice.establishConnection(false)
                .flatMap { rxBleConnection -> rxBleConnection.setupNotification(charUUID) }
                .doOnNext { }
                .flatMap { notificationObservable -> notificationObservable } // <-- Notification has been set up, now observe value changes.
                .subscribe(
                        { bytes ->
                            run {
                                // Log.i("Notification!", bytes!!.contentToString())
                                // println(bytes.toHex())
                                sp?.play(pool?.get(mRandom.nextInt(pool!!.size))!!, 1F, 1F, 0, 0, 1F)
                            }
                        },
                        { throwable -> Log.i(TAG, throwable.toString())}
                )
```

This notification works. I am able to see the value of the notification change when my device's sensor is activated.

Now, I want to click a button and send a write operation using the following code:

```
bleDevice.establishConnection(false)
                .flatMapSingle({ rxBleConnection -> rxBleConnection.writeCharacteristic(charUUID, bytesToWrite) })
                .subscribe(
                        { characteristicValue ->
                            run {
                                Log.d(TAG, "Write Command Succeeded")
                            }
                        },
                        { throwable ->
                            run {
                                Log.d(TAG, "Write Command Failed")
                                Log.d(TAG, throwable.toString())
                            }
                        }
                )
```

When I click the button I get the error message in the log output below. It says I am already connected. How can I send a write operation without attempting to connect again?

**Expected behavior**
I am expecting to be able to listen to notifications and also send write operations in the same Activity.

**Log Output**

```
D/ColorsFragment: Write Command Failed
    com.polidea.rxandroidble2.exceptions.BleAlreadyConnectedException: Already connected to device with MAC address 34:81:F4:C6:09:0F
```

akhatriST, 2020-07-09:
You are trying to establish the connection to an already connected device, Below line is the issue in your write operation

`bleDevice.establishConnection(false)`

You need to store the connection in an Observable so that you can use it in future.

Have a look at this example
https://github.com/Polidea/RxAndroidBle/blob/master/sample-kotlin/src/main/kotlin/com/polidea/rxandroidble2/samplekotlin/example4_characteristic/CharacteristicOperationExampleActivity.kt

SeloSlav, 2020-07-09:
Thanks. I ended up using that sample application but I am running into the same problem. Is it possible to pass connection observables from an activity to a fragment?

akhatriST, 2020-07-09:
Why are you initializing the connection in Activity and then in Fragment? Any particular reason

What I would suggest is, you should pass the device mac address (that you want to connect) to your Activity or Fragment, and try to initialize the connection only there. This way you will connect only once and do the intended work 

Also, I wouldn't recommend passing connection Observables between Activity and Fragment.

SeloSlav, 2020-07-09:
I ended up using the code in the sample application:
https://stackoverflow.com/q/62788480/4077513

I initialized the connection observable in my MainActivity because I wanted
the notification change subscription to be available when the user
navigates to either Fragment.

The app is fairly simple with one Activity and two Fragments. I listen to
Notifications in the Activity and send write operations from the Fragments.

I will try only initializing the observable in the first Fragment and try
both operations there.

dariuszseweryn, 2020-07-09:
The issues on github are meant for discussing bugs mostly. This is not a bug but a question related to usage. The place for this kind of questions is on Stackoverflow and this particular problem has been discussed there many times. Closing this.

Best regards

closed 2020-07-09 by dariuszseweryn