dariuszseweryn/RxAndroidBle #702: Unable to read data from the BLE device

akhatriST, 2020-07-09:
**Describe the bug**
Unable to read data from the BLE device. 

Use Case, I have 2 characteristics (A and B), one for writing and one for reading the data from the BLE device. In order to read data on the device, I need to write data on Characteristic A and enable notifications on Characteristic B. 

**Expected behavior**
Upon writing the Characteristic A, I should receive the data on the device via notifications enabled on Characteristic B. 

**Smartphone (please complete the following information):**
 - Device: [e.g. Samsung S7]
 - OS: [e.g. Android 8.0]
 - Library version: [e.g. 1.11.1]

**Logs from the application when bug occurs (this will greatly help in quick understanding the problem)**
To turn on logs use:
```
D/RxBle#ClientOperationQueue: QUEUED   ConnectOperation(262548174)
D/RxBle#ClientOperationQueue: STARTED  ConnectOperation(262548174)
I/RxBle#ClientOperationQueue: RUNNING  ConnectOperation{MAC='MAC-ADDRESS', autoConnect=false}
V/RxBle#BleConnectionCompat: Connecting without reflection
D/BluetoothGatt: connect() - device: MAC-ADDRESS, auto: false
D/BluetoothAdapter: isSecureModeEnabled
D/BluetoothGatt: registerApp()
    registerApp() - UUID=eb7a11d1-6524-4cca-a133-f033abc0b5ec
D/BluetoothGatt: onClientRegistered() - status=0 clientIf=8
D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=8 device=MAC-ADDRESS
I/RxBle#GattCallback: MAC='MAC-ADDRESS'  onConnectionStateChange(), status=0, value=2
V/MyApp#-: Connection successful with My Device   
D/RxBle#ClientOperationQueue: FINISHED ConnectOperation(262548174) in 980 ms
D/com.myapp.sample.ui.devicessearch.DeviceSearchFragment: onDeviceConnected
D/RxBle#ConnectionOperationQueue: QUEUED   ServiceDiscoveryOperation(165329419)
D/RxBle#ConnectionOperationQueue: STARTED  ServiceDiscoveryOperation(165329419)
I/RxBle#ConnectionOperationQueue: RUNNING  ServiceDiscoveryOperation{MAC='MAC-ADDRESS'}
D/BluetoothGatt: discoverServices() - device: MAC-ADDRESS
V/MyApp#: My Device    is Connected
D/BluetoothGatt: onConnectionUpdated() - Device=MAC-ADDRESS interval=6 latency=0 timeout=500 status=0
I/RxBle#GattCallback: MAC='MAC-ADDRESS'      onConnectionUpdated(), status=0, interval=6 (7.50 ms), latency=0, timeout=500 (5000 ms)
D/BluetoothGatt: onSearchComplete() = Device=MAC-ADDRESS Status=0
I/RxBle#GattCallback: MAC='MAC-ADDRESS'     onServicesDiscovered(), status=0
V/RxBle#ServiceDiscoveryOperation: Preparing services description
V/RxBle#ServiceDiscoveryOperation: --------------- ====== Printing peripheral content ====== ---------------
    MAC='MAC-ADDRESS'
    PERIPHERAL NAME: My Device   
D/BluetoothGatt: setCharacteristicNotification() - uuid: CHARACTERSTIC-B enable: true
E/MyApp#Connection: Notification has been setup
D/RxBle#ConnectionOperationQueue: QUEUED   DescriptorWriteOperation(260820541)
D/RxBle#ConnectionOperationQueue: FINISHED ServiceDiscoveryOperation(165329419) in 996 ms
D/RxBle#ConnectionOperationQueue: STARTED  DescriptorWriteOperation(260820541)
I/RxBle#ConnectionOperationQueue: RUNNING  DescriptorWriteOperation{MAC='MAC-ADDRESS', descriptor=[uuid='Client Characteristic Configuration', hexValue=[01, 00]]}
D/BluetoothGatt: onConnectionUpdated() - Device=MAC-ADDRESS interval=39 latency=0 timeout=500 status=0
I/RxBle#GattCallback: MAC='MAC-ADDRESS'      onConnectionUpdated(), status=0, interval=39 (48.75 ms), latency=0, timeout=500 (5000 ms)
I/RxBle#GattCallback: MAC='MAC-ADDRESS'        onDescriptorWrite(), status=0, value=[uuid='Client Characteristic Configuration']
D/RxBle#ConnectionOperationQueue: FINISHED DescriptorWriteOperation(260820541) in 117 ms
D/RxBle#ConnectionOperationQueue: QUEUED   CharacteristicWriteOperation(40916903)
D/RxBle#ConnectionOperationQueue: STARTED  CharacteristicWriteOperation(40916903)
I/RxBle#ConnectionOperationQueue: RUNNING  CharacteristicWriteOperation{MAC='MAC-ADDRESS', characteristic=[uuid='CHARACTERISTIC-A', hexValue=[01, 00, 00, 09, 08, 00, 00, 00, 00, 00, 00, 00, 00]]}
I/RxBle#GattCallback: MAC='MAC-ADDRESS'    onCharacteristicWrite(), status=0, value=[uuid='CHARACTERISTIC-A']
V/MyApp#-: Write on device successful
D/RxBle#ConnectionOperationQueue: FINISHED CharacteristicWriteOperation(40916903) in 119 ms
```

Here is the code 
```
 private Observable<byte[]> writeToGetNotification(byte[] data) {
        return rxBleConnectionObservable
                .flatMap(rxBleConnection -> rxBleConnection.writeCharacteristic(CHARACTERISTIC-A-UUID, data).toObservable());
    }

    private void setupNotifications() {

        final Disposable disposable = rxBleConnectionObservable
                .flatMap(rxBleConnection -> rxBleConnection.setupNotification(CHARACTERISTIC-B-UUID, NotificationSetupMode.QUICK_SETUP))
                .doOnNext(this::notificationHasBeenSetUp)
                .flatMap(notificationObservable -> notificationObservable)
                .subscribe(this::onNotificationReceived, this::onNotificationSetupFailure);

        compositeDisposable.add(disposable);
    }

    private void notificationHasBeenSetUp(Observable<byte[]> observable) {
        Log.i(TAG, "Notification has been setup");
    }

    private void onNotificationSetupFailure(Throwable throwable) {
       Log.e(TAG, "Notification error: " + throwable);
    }

    private void onNotificationReceived(byte[] abData) {
        Log.i(TAG, "Notification received");
    }
```
I call the ```setupNotifications()``` function after successfully establishing a connection. Also I have tried out other Notification setup modes as well, but no success.

dariuszseweryn, 2020-07-09:
As you can see everything on the app side is successful and there are no logs that the app got any notification. This points to your peripheral not behaving as you would expect. Consult peripheral documentation.

akhatriST, 2020-07-10:
Thanks, I got it working.

closed 2020-07-10 by akhatriST