dariuszseweryn/RxAndroidBle #713: Problem updating the sample project

RobLewis, 2020-08-16:
**Describe the bug**
Attempt to update my sample project from the GitHub repository gives errors. 

**To Reproduce**
Have an old copy of the sample project in local Android Studio. 
Attempt to update project from GitHub. 

**Expected behavior**
An error-free project should be created. 

**Actual behavior**
`RxBleGattCallback` has compiler errors: 
`import bleshadow.…` statements give "Cannot resolve symbol `bleshadow`" errors 
(l. 59): `return Observable.error(bleGattException);` gives error that required type is `android.database.Observable<?>`; provided type is `io.reactivex.Observable` 
(l. 288): `errorMapper` function is the wrong type 
(l. 302): `disconnectionRouter` is the wrong type 
(l. 309): `getOnConnectionStateChange` is the wrong type 
(l. 314 and several following): cannot resolve method `delay` in Observable
(l. 329): `getOnCharacteristicChanged` is the wrong type

**Additional context**
Android Studio v. 4.0.1

RobLewis, 2020-08-16:
I updated all the dependencies and then noticed that the following classes are also generating errors: 
NotificationAndIndicationManager 
CharacteristicWriteOperation 
ConnectionOperationQueueImpl 
Errors include "Cannot resolve method `filter` in Observable", and inability to resolve the `@Inject` and `@Named` annotations.

dariuszseweryn, 2020-08-24:
Hello Rob!
I think this is related to an old #627 issue. Dagger is shadowed in this project and shadow configuration cannot be done well enough for Android Studio. After building the project `DaggerClientComponent` should get generated.

RobLewis, 2020-08-31:
After updating the project it appears that I'm down to a single error: "Cannot resolve symbol 'DaggerClientComponent'" in the class `RxBleClient`.

RobLewis, 2020-08-31:
Sorry, but I don't know what "shadowed in this project" means. Could you briefly explain?

dariuszseweryn, 2020-09-01:
[Shadowed](https://github.com/johnrengelman/shadow) i.e. renaming a library's original package so multiple copies of this library could be used in a single application without a package clash (which may lead to runtime crashes if newer version of the lib is not backwards compatible). This was [an issue quite a long time ago](https://github.com/Polidea/RxAndroidBle/issues/342).

RobLewis, 2020-09-02:
Damn, now it seems we're headed down a rabbit hole because I don't know what a "package clash" is. Here's my best guess: can you confirm/deny/explain? 

- A given application bundle may require multiple copies of a library, and these different copies/versions may have incompatibilities. 

- To enable the build system to correctly link up the various copies, we rename them. This seems to be what is meant by "shadowing", but how does this actually take place? Is it automatic in some sense? Does the list of dependencies have to be changed in some way? 

- I'm guessing that if all versions of libraries were backward-compatible, none of this would be necessary. True? 

I'm sure I have more questions, but this will do for now. I appreciate any help! (I realize that questions like this might be better for Stack Overflow, but any quick tips might solve my problem.)

dariuszseweryn, 2020-09-04:
> A given application bundle may require multiple copies of a library, and these different copies/versions may have incompatibilities.

Correct. If in newer version of the library breaking change will appear then all dependencies that used this library as their dependency will stop working.

> To enable the build system to correctly link up the various copies, we rename them. This seems to be what is meant by "shadowing", but how does this actually take place? Is it automatic in some sense? Does the list of dependencies have to be changed in some way?

Correct. There is a "Shadow" library I linked in the previous post which does this renaming. I did not do the setup but it seems to be automatic — only the source code needs to have references to shadowed classes (which get generated only after project will build successfully) and Android Studio does not always pick them up correctly

> I'm guessing that if all versions of libraries were backward-compatible, none of this would be necessary. True?

Correct.

dariuszseweryn, 2022-05-28:
With e8fdb96e9313c12a6bb2e928f7535b5619b6b232 this problem should finally be a thing of the past 🎉

closed 2022-05-28 by dariuszseweryn