dariuszseweryn/RxAndroidBle #720: Disconnecting and closing BLE connection stops classic Blutetooth profile too

gary0707, 2020-09-08:
I have a setup with two Android devices: an android phone and a custom hardware running Android 10. The latter acts as a BLE Server, which is discoverable always with the same static MAC address. Also when that custom HW is paired with the phone it has access to phone's contacts and calls history. That HW can also play media (sound) streamed from the phone.

I use RxAndroidBle 1.11.0 library for BLE communication. As soon as I exchange some data via BLE Characteristic I unsubscribe from the RX observable so the library effectively calls: `bluetoothGatt.disconnect()` then `blutetoohGatt.close()` (all those inside the `DisconnectOperation` class).

**My problem** is the fact that based on my observations, calling `blutetoohGatt.disconnect()` disconnects the classic BT profile (PBAP) as well.

Is it an expected behaviour?

I have investigated the code of generic Android P framework and it looks that calling` bluetoothGatt.disconnect() `-> `bluetoothGatt.close()` completely disconnects the device described by the given MAC Address: [BluteoothGatt#disconnect()](https://android.googlesource.com/platform/frameworks/base/+/master/core/java/android/bluetooth/BluetoothGatt.java#898) 
[BlutetoothGatt#close()](https://android.googlesource.com/platform/frameworks/base/+/master/core/java/android/bluetooth/BluetoothGatt.java#712) 
[BlutetoothGatt#unregisterApp()](https://android.googlesource.com/platform/frameworks/base/+/master/core/java/android/bluetooth/BluetoothGatt.java#824)

I would risk a statement that I possibly have quite unusual BLE use-case scenario where the phone (Central/Client) and the peripheral (custom Android HW/Server) are connected simultaneously via Classic BT profile and via BLE. I suspect that disconnecting/closing device using its BT MAC address disconnect both profiles, hence the contacts sharing stops working. Is it possible to disconnect BLE connection only, but not affect the classic BT (SPP/PBAP) connections between the two devices?

dariuszseweryn, 2020-09-18:
Sorry for the delay — I was on vacation.

> I would risk a statement that I possibly have quite unusual BLE use-case scenario where the phone (Central/Client) and the peripheral (custom Android HW/Server) are connected simultaneously via [...]

I would expect that calling `BluetoothGatt.disconnect()` & `BluetoothGatt.close()` would close only the `BluetoothGatt` which is merely a client of the underlaying "physical" connection. On that connection one may have multiple opened `BluetoothGatt` objects and operating on the (almost) independently. Some Android BLE stacks postpone physical connection closure for several seconds sometimes (you can check by listening to `BluetoothDevice.ACTION_ACL_DISCONNECTED` broadcasts).
I would expect other Bluetooth Profiles to operate as these semi-independent clients as well.
I am unaware of how the stack does operate with dual mode peripherals — it may work differently depending on a manufacturer. I would expect it to be a misbehaviour of the BLE stack but if you would find different explanation — feel free to attach more info.
Unfortunately I do not have any available dual mode devices to perform my own tests.

gary0707, 2020-09-20:
I hope you had a fantastic holidays.
Thank you for the answer. I did a ton of other tests and I still suspect that having the same single (non-randomized) MAC address in my custom HW is related to the "disconnection operation" disconnecting both profiles - the BLE one and the classic one. 

For testing purposes I have used a BLE SIM app from Google Play store, running on my custom HW. The app advertises battery level profile, but uses an another (randomised) MAC address for the the BLE advertisement. Connecting/Disconnecting the client app running on the phone, does NOT affect the BT profile. I will investigate that path. Maybe the problem is related to having the same single/constant MAC address of the HW running the BLE Server. Will keep you posted.

gary0707, 2020-09-21:
That's what I can see in the system report (via `adb shell dumpsys bluetooth_manager` ):

> 09-21 08:34:59.289--BluetoothRemoteDevices -- ACTION_ACL_CONNECTED, device is FC77XY, linktype is 1
>09-21 08:34:59.893--BluetoothRemoteDevices -- AbstractionLayer.BT_PROPERTY_UUIDS, device is FC77XY, Handsfree,ObexObjectPush,PANU,NAP,AudioSink,AvrcpController,MNS,**309a00XYX-REST-OF-MY-SECERT_CHARACTERISTIC**

Doesn't it look like a single BluetoothRemoteDevice, determined by the same single MAC address, advertising listed protocols (mix of "handsgree, PANU and my BLE UUID of characteristic I broadcast.

gary0707, 2020-09-22:
I did manage to advertise my custom service using randomised BT MAC address and that did the trick. 
The same custom HW behaves correctly when the Gatt profile is bound to the MAC address different that the one assigned to Classic BT profiles.

What I did on my custom HW? A have reverted the order of operations, namely, as a user:
1) At the very beginning I do perform the BLE connect & bond
2) Then I connect and pair classic BT profile

Long Story short: even with a single radio module, the advertised MAC address does matter. 
Case closed.

closed 2020-09-22 by gary0707

dariuszseweryn, 2020-09-22:
I just made a quick query for "ble dual mode mac address" and it turns out people [had problems with it](https://www.silabs.com/community/wireless/bluetooth/forum.topic.html/bt121_dual_mode_pair-ed7L) — but this seems to be a bit different problem than you encountered.
Curiously though some other manufacturers [use different MAC addresses for their dual-mode devices](https://www.waveshare.com/wiki/Dual-mode_Bluetooth).

gary0707, 2020-09-23:
Thanks for the response and posted links. In those case, the "duality" seems to be related to fact, that the BT HW can act as a both: master and slave. That is not my case probably. 
I am still not sure that I have fully resolve my problems, but for sure the order of pairing (BLE first then Classic) is crucial in my case. All in all the topic I have raised in this ticket seems to be resolved hence I have closed it.

dariuszseweryn, 2020-09-23:
From what I read in the above mentioned links I see that the hardware is dual-mode i.e. it uses SPP (Classic Bluetooth) and GATT (BLE). Judging from the screens / text and more links they also used different MAC addresses for their BT and BLE functionalities.
Anyway — I am glad you resolved the problem. Unfortunately as far as I know the library cannot do anything to help — this is an OS BLE stack problem.