dariuszseweryn/RxAndroidBle #721: Support for Companion device pairing?

fabiendem, 2020-09-09:
**Is your feature request related to a problem? Please describe.**
Google is actively discouraging the use of location access, in the case of BLE, `ACCESS_FINE_LOCATION`.

This is detailed over there: https://support.google.com/googleplay/android-developer/answer/9799150

And as well here:
https://developer.android.com/privacy/best-practices#location
> If your app needs to pair the user's device with a nearby device over Bluetooth or Wi-Fi, use the companion device manager, which doesn't require location permissions.

They now provide an alternative to the classic scan which requires `ACCESS_FINE_LOCATION` in the form of a "Companion device pairing" feature.
See https://developer.android.com/guide/topics/connectivity/companion-device-pairing

**Describe the solution you'd like**

I wonder if you plan to integrate this in RxAdroidBle for the connection flow, and also in the react-native library.

**Describe alternatives you've considered**
If not, we will have to add a custom bridge and our own implementation.

**Additional context**
N/A

Cheers!

fabiendem, 2020-12-02:
Sorry to bump this, but has anyone any insights on Google's plans wrt the BLE connection procedure.
Reading https://developer.android.com/guide/topics/connectivity/bluetooth#FindDevices
They really seem to push for the use of this "Companion device pairing".

I work on a project where I actually don't bother pairing since it's possible, and simply scan for devices and if the one needed is found, I connect to it.
This move towards "Companion device pairing" for the connection procedure seems to enforce the pairing of a device, which has firmware/hardware implications.

Like in the initial message I also wonder if the Polidea's library will wrap the Companion device pairing of if we should do things home made.

Cheers

dariuszseweryn, 2020-12-04:
Hello,

It is very likely that it will be added at some point but it is not a priority at the moment. I still struggle to get consistent time to maintain the library (or even review PRs...). The highest priority is to develop a new API that will allow for fixing the `UndeliverableException` which will allow for integrating the newest version of the library into other modules like the react-native lib.

As far as I can tell the Companion Device Pairing feature is fairly simple to use and integrating with RxAndroidBle shouldn't be a problem even if the feature itself is not yet added to RxAndroidBle / react-native-ble-plx.

fabiendem, 2020-12-07:
> As far as I can tell the Companion Device Pairing feature is fairly simple to use and integrating with RxAndroidBle shouldn't be a problem even if the feature itself is not yet added to RxAndroidBle / react-native-ble-plx.

Yeah, I realise that. Background story is that I would need the wrapper for a React Native project, so in `react-native-ble-plx`, and I understand that having it in `RxAndroidBle` is a first step.
I will play around with the custom way for now.