dariuszseweryn/RxAndroidBle #726: Fail Tried to Re-connect

iqbalprabu, 2020-09-25:
hi, i have some problem when i try re-connect when device lost connection

this is error
`onConnectionFailure: Disconnected from MAC='32:2D:55:47:2B:4E' with status 133 (GATT_ERROR)`

this is my code
`private fun updateConnection() {

        macAddress?.let { address ->
            bleDevice = CamarApps.rxBleClient.getBleDevice(address)
            characteristicUuid = UUID.fromString(NOTIFY_UUID)
            connectionObservable = prepareConnectionObservable()

            connectionObservable
                ?.flatMapSingle { it.discoverServices() }
                ?.flatMapSingle { it.getCharacteristic(characteristicUuid) }
                ?.observeOn(AndroidSchedulers.mainThread())
                ?.doOnSubscribe {  }
                ?.subscribe(
                    { characteristic ->
                        updateUI(characteristic)
                    },
                    { onConnectionFailure(it) },
                    { updateUI(null) }
                )
                .let { obs -> obs?.let { connectionDisposable.add(it) } }

            connectionObservable
                ?.flatMap { it.setupNotification(characteristicUuid) }
                ?.doOnNext { }
                ?.flatMap { it }
                ?.observeOn(AndroidSchedulers.mainThread())
                ?.subscribe({ onNotificationReceived(it) }, { onNotificationSetupFailure(it) })
                .let { obs-> obs?.let { connectionDisposable.add(it) } }

            bleDevice?.observeConnectionStateChanges()
                ?.subscribe { connectionState ->
                   
                    if(connectionState == RxBleConnection.RxBleConnectionState.CONNECTED) {
                        snackbar?.dismiss()
                        myTimer?.cancel()
                    } else if(connectionState == RxBleConnection.RxBleConnectionState.DISCONNECTED) {
                        showSnackBarLostConnection()
                        tyringReConnect()
                    }
                }.let { obs -> obs?.let { connectionDisposable.add(it) } }
        }
    }
    private fun prepareConnectionObservable(): Observable<RxBleConnection>? =
        bleDevice
            ?.establishConnection(false)
            ?.takeUntil(disconnectTriggerSubject)
            ?.compose(ReplayingShare.instance())

    private fun tyringReConnect() {
        if(myTimer == null) {
            var time = 0
            myTimer = Timer();
            myTimer?.schedule(object : TimerTask() {
                override fun run() {
                    if (time == 10) {
                        time = 0
                        disconnectTriggerSubject.onNext(Unit)
                        connectionDisposable.dispose()
                        bleDevice = null
                        myTimer?.cancel()
                        myTimer = null
                        updateConnection()
                    }
                    time++
                }
            }, 0, 1000)
        }
    }`

please help.

dariuszseweryn, 2020-09-25:
Exactly as it is described in [the Wiki](https://github.com/Polidea/RxAndroidBle/wiki/FAQ:-Cannot-connect)

iqbalprabu, 2020-09-26:
so the solution is to re-scan? can re-scan and connect without user interaction?

iqbalprabu, 2020-09-26:
i have try like this, but still can't connect to devices

`private fun updateConnection() {

        macAddress?.let { address ->

            bleDevice = CamarApps.rxBleClient.getBleDevice(address)

            characteristicUuid = UUID.fromString(NOTIFY_UUID)
            connectionObservable = prepareConnectionObservable()

            connectionObservable
                ?.flatMapSingle { it.discoverServices() }
                ?.flatMapSingle { it.getCharacteristic(characteristicUuid) }
                ?.observeOn(AndroidSchedulers.mainThread())
                ?.subscribe(
                    { characteristic ->
                        updateUI(characteristic)
                    },
                    { onConnectionFailure(it) },
                    { updateUI(null) }
                )
                .let { obs -> obs?.let { connectionDisposable.add(it) } }

            connectionObservable
                ?.flatMap { it.setupNotification(characteristicUuid) }
                ?.doOnNext { }
                ?.flatMap { it }
                ?.observeOn(AndroidSchedulers.mainThread())
                ?.subscribe({ onNotificationReceived(it) }, { onNotificationSetupFailure(it) })
                .let { obs-> obs?.let { connectionDisposable.add(it) } }

            bleDevice?.observeConnectionStateChanges()
                ?.subscribe { connectionState ->
                    if(connectionState == RxBleConnection.RxBleConnectionState.CONNECTED) {
                        snackbar?.dismiss()
                    } else if(connectionState == RxBleConnection.RxBleConnectionState.DISCONNECTED) {
                        showSnackBarLostConnection()
                        scanAndReconectDevice()
                    }
                }.let { obs -> obs?.let { connectionDisposable.add(it) } }
        }
    }
    private fun prepareConnectionObservable(): Observable<RxBleConnection>? =
        bleDevice
            ?.establishConnection(false)
            ?.takeUntil(disconnectTriggerSubject)
            ?.compose(ReplayingShare.instance())

    private fun scanAndReconectDevice() {

        connectionDisposable.dispose()
        connectionDisposable.clear()

        if (bluetoothAdapter?.isEnabled == false) {
            enableBluetooth()
            showFailedReConnect()
            return
        }

        scanBleDevices()
            .observeOn(AndroidSchedulers.mainThread())
            .timeout(15, TimeUnit.SECONDS)
            .delaySubscription(10, TimeUnit.SECONDS)
            .doFinally {
                scanDisposable?.dispose()
            }
            .subscribe({
                if(it.bleDevice.name == deviceName) {
                    deviceName = it.bleDevice.name
                    macAddress = it.bleDevice.macAddress
                    scanDisposable?.dispose()
                    aisViewModel.macAddressAis.value = macAddress
                }
            }, {
                showFailedReConnect()
            })
            .let { scanDisposable = it }
    }

    private fun showFailedReConnect() {
        runOnUiThread {
            snackbar?.setText("Failed Re-connect Device")
            snackbar?.setAction("Try Again") {
                Handler().postDelayed({
                    scanAndReconectDevice()
                    showSnackBarLostConnection()
                }, 1000)
            }
        }
    }

    private fun scanBleDevices(): Observable<ScanResult> {
        val scanSettings = ScanSettings.Builder()
            .setScanMode(ScanSettings.SCAN_MODE_LOW_LATENCY)
            .setCallbackType(ScanSettings.CALLBACK_TYPE_ALL_MATCHES)
            .build()

        val scanFilter = ScanFilter.Builder()
            .setDeviceName(deviceName)
            .build()

        return CamarApps.rxBleClient.scanBleDevices(scanSettings, scanFilter)
    }`

dariuszseweryn, 2020-09-26:
Sorry to hear that. Unfortunately (most probably) this is not an issue with the library but with the convoluted logic you have implemented. For usage difficulties I do encourage checking [Stackoverflow for tag `rxandroidble`](https://stackoverflow.com/questions/tagged/rxandroidble) (and/or possibly `rxjava`) because that is a better suited place for usage questions.

closed 2020-11-23 by dariuszseweryn