dariuszseweryn/RxAndroidBle #730: pairing request dialog appears on every connection

tolot27, 2020-11-07:
In our application we get every five minutes (at each communication with BLE devices) the bonding request dialog as shown in https://github.com/NightscoutFoundation/xDrip/issues/1494.

This behavior appears on devices with Android 10 (i. e. Pixel 3a, Samsung A50, S10, etc.) were the Android 2020-11-01 security patch level is installed (see here: https://github.com/NightscoutFoundation/xDrip/issues/1494#issuecomment-722176016_).

Is there an advice how to avoid this?

dariuszseweryn, 2020-11-21:
Hello,
As you have linked to original issue. This does not seem to be a problem with the library but rather with the OS which manages bonding. I am afraid there is nothing can be done from the programmatic side of things.

tolot27, 2020-11-21:
@dariuszseweryn Yes, you are right. Thx!

closed 2020-11-21 by tolot27