dariuszseweryn/RxAndroidBle #732: `scanDevices` in foreground service doesn't work / times out unless `android:foregroundServiceType="location"` is defined in AndroidManifest.xml

j-roskopf, 2021-01-02:
This almost certainly isn't a problem with the library, but I felt it worth mentioning anyways.

I am relatively new to BLE and I have inherited a project that utilizes this library extensively for BLE work. When I took code I knew was working in my Activity and tried utilizing it in a Service and saw it not working, my immediate location to look for answers / support was this Github. I imagine others might be in the same boat (being relatively new to Android BLE and looking to this library as the first entry point into diagnosing issues)

I am hoping that maybe a blurb could be added in the documentation about using this library in a service.

RobLewis, 2021-01-02:
A while back I created this Gist to make it easier to use Services in a reactive application. It might help you: https://gist.github.com/RobLewis/fe7fad0381b761bae2842dc0d2cd8f2c 

Also, I had to do some things to keep the Service running, like using a WakeLock, and posting a Foreground Notification.

j-roskopf, 2021-01-02:
Yeah, that is what I ended up getting working was using a foreground service. I don't need the connection to stay alive forever, just long enough to scan, connected, and pull from data from the device. So I used work manager to schedule starting of the foreground service every hour or so to do that task. Once I added the foregroundServieType to the manifest, all was working perfectly then. Just took me a few days to come to that conclusion :)

Thank you again for the gist!

dariuszseweryn, 2021-04-02:
Added a mention to [the background operations Wiki page](https://github.com/Polidea/RxAndroidBle/wiki/Tutorial:-Background-operation)

Hope that helps! Cheers!

closed 2021-04-02 by dariuszseweryn

j-roskopf, 2021-04-02:
awesome - appreciate it :)