dariuszseweryn/RxAndroidBle #733: GATT 22 Error handling

georgeRaduNeusoft, 2021-01-05:
Hi! I have been using this library for some years but this is the first time when I get to the "connection failed com.polidea.rxandroidble2.exceptions.BleDisconnectedException: Disconnected from MAC='XX:XX:XX:XX:XX:XX' with status 22 (GATT_CONN_TERMINATE_LOCAL_HOST)" error.

I use a Whoop strap (heart rate device). This issue is present just on a certain device that we have installed our apk on but not on other device which works as expected. The difference between the two devices with Android OS is that they are from different in terms of manufacturing / Bluetooth adapter / components.

Do I have a chance somehow to dig more in order to understand the root of the issue? I know that others say that there exists a "Sniffer trace" device / tool or something like that.

Thank you in advance!

dariuszseweryn, 2021-01-06:
> Do I have a chance somehow to dig more in order to understand the root of the issue? I know that others say that there exists a "Sniffer trace" device / tool or something like that.

They probably refer to HCI snoop log. There seems to be a quite up-to-date blogpost about it [here](https://medium.com/@charlie.d.anderson/how-to-get-the-bluetooth-host-controller-interface-logs-from-a-modern-android-phone-d23bde00b9fa).

georgeRaduNeusoft, 2021-01-06:
@dariuszseweryn thank you very much! I will study the article mentioned, try to debug using their approach and come up with a feedback.

dariuszseweryn, 2021-04-02:
Closing since there was no activity for quite a while.

closed 2021-04-02 by dariuszseweryn