dariuszseweryn/RxAndroidBle #736: RxBleScanRecordMock missing in maven artifact

SwDevElatec, 2021-02-04:
**Describe the bug**
The class RxBleScanRecordMock is missing and the example at https://github.com/Polidea/RxAndroidBle/tree/master/mockrxandroidble#example does not compile.

**To Reproduce**
Steps to reproduce the behavior:
1. Include MockRxAndroidBle as described under Download via Gradle
2. See error "Cannot resolve symbol 'RxBleScanRecordMock'"

**Expected behavior**
All classes under src/main/java/com/polidea/rxandroidble2/mockrxandroidble/ should be in the aar artifact.

**Smartphone (please complete the following information):**
N/A, unit tests

**Logs from the application when bug occurs (this will greatly help in quick understanding the problem)**
N/A
```

**Additional context**
Downloaded mockclient-1.11.1.aar from https://mvnrepository.com/artifact/com.polidea.rxandroidble2/mockclient/1.11.1 and inspected it. RxBleScanRecordMock is missing.

nrbrook, 2021-02-08:
I believe this is because there has not been a recent release. In addition, there [hasn't been a snapshot](https://oss.sonatype.org/content/repositories/snapshots/com/polidea/rxandroidble2/mockclient/) recently either. I could really do with this too!

SwDevElatec, 2021-02-09:
@nrbrook Many thanks! I am actually trying to use RxAndroidBle for testing only, so this could be a workaround for me. @dariuszseweryn Excuse my confusion, but the artifact version (mockclient-1.11.1.aar) suggests that the classes in the maven repo are the same as the version with the same tag (1.11.1) in the source code repo. If there will be a release from 1.11.1 source code what version will the artifact have?

rfrowe, 2021-03-04:
@SwDevElatec Hi, you can check out my comment [here](https://github.com/Polidea/RxAndroidBle/pull/703#issuecomment-790074565) on how to get these classes via gradle!

nrbrook, 2021-03-08:
https://twitter.com/dseweryn/status/1369016230095511554?s=21

dariuszseweryn, 2021-03-31:
There is now a `1.12.0-SNAPSHOT` release available (see readme for snapshots). I expect 1.12.0 being released "soon" (as there is not much waiting in the pipeline.

dariuszseweryn, 2021-04-13:
Just released a regular `1.12.0`. Cheers!

closed 2021-04-13 by dariuszseweryn