dariuszseweryn/RxAndroidBle #739: Is undocumented scan throttle still relevant?

balakzde, 2021-03-11:
Hi,
I have been using this library for almost 2 years. I have been trying to get rid of all unnecessary libraries, so I also god rid of this one. One think I remembered is that I was making sure I only start bluetooth scanning once every 6050ms, because of undocumented scan throttle that is supposed to appear on android 7 and above. Anyway before I managed to reimplement this feature without this library, I noticed I can start ble scanning as much as I want and I have never experienced any problem. I told our testers to test all devices including android 7 (We do not have many of those left) and they also never experienced any problem.

So my question is, is this still relevant? Do you know any specific phone that suffers from this problem, or is it now a ghost, that only haunts users of this library?

iSWORD, 2021-03-22:
It's relevant, otherwise I wouldn't have stumbled upon your issue while looking up an undocumented scan exception on my Android 11 phone.

I don't see how **this** library is "unnecessary". It saves a ton of development & maintenance time. Replacing it with native Android code seems to be a waste of manpower.

balakzde, 2021-03-23:
> It's relevant, otherwise I wouldn't have stumbled upon your issue while looking up an undocumented scan exception on my Android 11 phone.
>
> I don't see how **this** library is "unnecessary". It saves [...]

I don't think you understand my question. I am not asking if it is part of this library - I know it is. If you are trying to figure out what it means - it means you tried to start scanning more than 5 times in 30s and that resulted in bluetooth scan fail on android 7, which was never documented in android documentation (that's why it is called undocumented scan throttle).

So this library is trying to prevent this from happening by implementing its own safety mechanism, so it is impossible for you to break your scanning without you knowing what the hell happened.

My question is not about this library, it is for android ble scanning in general. I am not able to break bluetooth scanning as much as I try and given this was not in official documentation I am asking if anyone is able to reproduce this behaviour - meaning is undocumented scan throttle error in this library still needed?

Sources: https://stackoverflow.com/questions/43114913/android-7-0-ble-scan-no-result
https://blog.classycode.com/undocumented-android-7-ble-behavior-changes-d1a9bd87d983

(I will not discuss if the library is or is not "unnecessary", If you are mainaining a big project for paying customers, it is always better not to use many libraries if not needed, so you have full control - for example i now can start more than 5 scanning in 30s and there is no problem so far)

iSWORD, 2021-03-23:
Oh I see your point. Makes sense then. Sorry I misunderstood. Good luck with your projects!

dariuszseweryn, 2021-03-31:
From what I see — [it is still relevant](https://android.googlesource.com/platform/packages/apps/Bluetooth/+/refs/tags/android-11.0.0_r33/src/com/android/bluetooth/gatt/AppScanStats.java#108).

closed 2021-04-13 by dariuszseweryn