dariuszseweryn/RxAndroidBle #741: [Question] - Number of messages that can be received in 1 second interval

VinnyScorsatto, 2021-03-17:
Hi, this is more of a question than a bug. 

Our device sends some status through bluetooth every second (5 messages of 20 bytes), and also if we change a setting it also sends a response (at least 3 messages of 20bytes). It seems to me that I am losing at least one of those. I asked our firmware team to increase the interval to 3 seconds and I got all messages with no problem.. Is there a limitation in the queue for receiving messages and if yes, is there a way to check if the queue is "full". Thanks a lot

dariuszseweryn, 2021-03-30:
> Is there a limitation in the queue for receiving messages and if yes, is there a way to check if the queue is "full".

There is no limitation in the queue that I know of. The highest throughput I remember I have seen was ~80 kBps but I think it was bottlenecked by BLE and not the library. This is way above amount of data you are receiving. I would like not to point anyone but my experience is that firmware is the hard part of BLE programming nowadays.

closed 2021-03-30 by dariuszseweryn

VinnyScorsatto, 2021-03-30:
Thanks @dariuszseweryn , don't know what is happening then.. I use the Polidea library for iOS as well and it works fine... When we increased the interval of some messages that we get periodically it works fine, but if we set it again to 1 second I am missing some bytes... thanks again

VinnyScorsatto, 2021-03-30:
Hi @dariuszseweryn , just to give additional information. I added the library logs with VERBOSE and I have 2 cases here, one that Failed and one successful... I will try to show the difference here. We use COBS encoding a full response ends in 00, right? I have added spaces between the messages to make it easier and highlighted in bold the lines were I believe we lost one message. For the library to handle all incoming messages well should we add any delay between the messages? 

Failed

I/RxBle#GattCallback: MAC='CC:93:4A:0A:50:BC'  onCharacteristicChanged(), value=[uuid='ec5aec02-5c84-4375-87b7-b18cddf9c66b', hexValue=[0C, 5C, 0A, 05, 58, 29, C0, 02, 10, 50, 01, 58, 06, A0, 01, 01, A8, 01, 06, F0]]
I/RxBle#GattCallback: MAC='CC:93:4A:0A:50:BC'  onCharacteristicChanged(), value=[uuid='ec5aec02-5c84-4375-87b7-b18cddf9c66b', hexValue=[01, 01, F8, 01, 05, 1A, 05, EE, C8, 00]]

I/RxBle#GattCallback: MAC='CC:93:4A:0A:50:BC'  onCharacteristicChanged(), value=[uuid='ec5aec02-5c84-4375-87b7-b18cddf9c66b', hexValue=[1C, 5D, 0A, 05, 58, 53, C0, 02, 2C, 08, 01, 12, 0A, 53, 58, 4B, 45, 59, 50, 41]]
I/RxBle#GattCallback: MAC='CC:93:4A:0A:50:BC'  onCharacteristicChanged(), value=[uuid='ec5aec02-5c84-4375-87b7-b18cddf9c66b', hexValue=[44, 30, 31, 18, 01, 50, 02, 58, 1D, 60, 80, 84, AC, 08, A0, 01, 01, A8, 01, 83]]
I/RxBle#GattCallback: MAC='CC:93:4A:0A:50:BC'  onCharacteristicChanged(), value=[uuid='ec5aec02-5c84-4375-87b7-b18cddf9c66b', hexValue=[01, B0, 01, C8, 98, 3D, B8, 01, 04, C0, 01, AE, 54, 7F, 38, 7D, 35, 00]]

I/RxBle#GattCallback: MAC='CC:93:4A:0A:50:BC'  onCharacteristicChanged(), value=[uuid='ec5aec02-5c84-4375-87b7-b18cddf9c66b', hexValue=[0C, 5E, 0A, 05, 58, 29, C0, 02, 10, 50, 01, 58, 06, A0, 01, 01, A8, 01, 06, F0]]
I/RxBle#GattCallback: MAC='CC:93:4A:0A:50:BC'  onCharacteristicChanged(), value=[uuid='ec5aec02-5c84-4375-87b7-b18cddf9c66b', hexValue=[01, 01, F8, 01, 05, 15, C9, D5, 11, 00]]

I/RxBle#GattCallback: MAC='CC:93:4A:0A:50:BC'  onCharacteristicChanged(), value=[uuid='ec5aec02-5c84-4375-87b7-b18cddf9c66b', hexValue=[1C, 5F, 0A, 05, 58, 53, C0, 02, 2C, 08, 01, 12, 0A, 53, 58, 4B, 45, 59, 50, 41]]
I/RxBle#GattCallback: MAC='CC:93:4A:0A:50:BC'  onCharacteristicChanged(), value=[uuid='ec5aec02-5c84-4375-87b7-b18cddf9c66b', hexValue=[44, 30, 31, 18, 01, 50, 02, 58, 1D, 60, 80, 84, AC, 08, A0, 01, 01, A8, 01, 83]]
I/RxBle#GattCallback: MAC='CC:93:4A:0A:50:BC'  onCharacteristicChanged(), value=[uuid='ec5aec02-5c84-4375-87b7-b18cddf9c66b', hexValue=[01, B0, 01, C8, 98, 3D, B8, 01, 04, C0, 01, AE, 54, 65, C0, E4, E6, 00]]

I/RxBle#GattCallback: MAC='CC:93:4A:0A:50:BC'  onCharacteristicChanged(), value=[uuid='ec5aec02-5c84-4375-87b7-b18cddf9c66b', hexValue=[0C, 60, 0A, 05, 58, 29, C0, 02, 10, 50, 01, 58, 06, A0, 01, 01, A8, 01, 06, F0]]
I/RxBle#GattCallback: MAC='CC:93:4A:0A:50:BC'  onCharacteristicChanged(), value=[uuid='ec5aec02-5c84-4375-87b7-b18cddf9c66b', hexValue=[01, 01, F8, 01, 05, BC, 26, 58, F2, 00]]
D/RxBle#ConnectionOperationQueue: QUEUED   CharacteristicWriteOperation(62807294)
D/RxBle#ConnectionOperationQueue: STARTED  CharacteristicWriteOperation(62807294)
I/RxBle#ConnectionOperationQueue: RUNNING  CharacteristicWriteOperation{MAC='CC:93:4A:0A:50:BC', characteristic=[uuid='87c886ab-9757-4cd8-870f-533d371e27c6', hexValue=[01, 11, 01, 0C, 58, 0E, C0, 02, 09, C8, 02, EC, B8, DC, DF, 0A, F0, 01, 03, F8, 01, 03, 80, 02, 05, A6, 70, B4, 03, 00]]}
I/RxBle#GattCallback: MAC='CC:93:4A:0A:50:BC'  onCharacteristicChanged(), value=[uuid='ec5aec02-5c84-4375-87b7-b18cddf9c66b', hexValue=[1C, 61, 0A, 05, 58, 53, C0, 02, 2C, 08, 01, 12, 0A, 53, 58, 4B, 45, 59, 50, 41]]
I/RxBle#GattCallback: MAC='CC:93:4A:0A:50:BC'  onCharacteristicChanged(), value=[uuid='ec5aec02-5c84-4375-87b7-b18cddf9c66b', hexValue=[44, 30, 31, 18, 01, 50, 02, 58, 1D, 60, 80, 84, AC, 08, A0, 01, 01, A8, 01, 83]]
I/RxBle#GattCallback: MAC='CC:93:4A:0A:50:BC'    onCharacteristicWrite(), status=13, value=[uuid='87c886ab-9757-4cd8-870f-533d371e27c6']
I/RxBle#GattCallback: MAC='CC:93:4A:0A:50:BC'  onCharacteristicChanged(), value=[uuid='ec5aec02-5c84-4375-87b7-b18cddf9c66b', hexValue=[01, B0, 01, C8, 98, 3D, B8, 01, 04, C0, 01, AE, 54, 5F, 42, 35, CF, 00]]
D/RxBle#ConnectionOperationQueue: FINISHED CharacteristicWriteOperation(62807294) in 261 ms
I/RxBle#GattCallback: MAC='CC:93:4A:0A:50:BC'  onCharacteristicChanged(), value=[uuid='ec5aec02-5c84-4375-87b7-b18cddf9c66b', hexValue=[0C, 63, 0A, 05, 58, 29, C0, 02, 10, 50, 01, 58, 06, A0, 01, 01, A8, 01, 06, F0]]
I/RxBle#GattCallback: MAC='CC:93:4A:0A:50:BC'  onCharacteristicChanged(), value=[uuid='ec5aec02-5c84-4375-87b7-b18cddf9c66b', hexValue=[01, 01, F8, 01, 05, 94, 8F, 46, AA, 00]]

I/RxBle#GattCallback: MAC='CC:93:4A:0A:50:BC'  onCharacteristicChanged(), value=[uuid='ec5aec02-5c84-4375-87b7-b18cddf9c66b', hexValue=[1C, 64, 0A, 05, 58, 53, C0, 02, 2C, 08, 01, 12, 0A, 53, 58, 4B, 45, 59, 50, 41]]
I/RxBle#GattCallback: MAC='CC:93:4A:0A:50:BC'  onCharacteristicChanged(), value=[uuid='ec5aec02-5c84-4375-87b7-b18cddf9c66b', hexValue=[44, 30, 31, 18, 01, 50, 02, 58, 1D, 60, 80, 84, AC, 08, A0, 01, 01, A8, 01, 83]]
I/RxBle#GattCallback: MAC='CC:93:4A:0A:50:BC'  onCharacteristicChanged(), value=[uuid='ec5aec02-5c84-4375-87b7-b18cddf9c66b', hexValue=[01, B0, 01, C8, 98, 3D, B8, 01, 04, C0, 01, AE, 54, 27, 48, BB, DA, 00]]

[... cut ...]
I/RxBle#GattCallback: MAC='CC:93:4A:0A:50:BC'  onCharacteristicChanged(), value=[uuid='ec5aec02-5c84-4375-87b7-b18cddf9c66b', hexValue=[01, B0, 01, C8, 98, 3D, B8, 01, 04, C0, 01, AE, 54, 81, C6, 8F, 07, 00]]

I/RxBle#GattCallback: MAC='CC:93:4A:0A:50:BC'  onCharacteristicChanged(), value=[uuid='ec5aec02-5c84-4375-87b7-b18cddf9c66b', hexValue=[0C, DF, 0A, 05, 58, 29, C0, 02, 10, 50, 01, 58, 06, A0, 01, 01, A8, 01, 06, F0]]
I/RxBle#GattCallback: MAC='CC:93:4A:0A:50:BC'  onCharacteristicChanged(), value=[uuid='ec5aec02-5c84-4375-87b7-b18cddf9c66b', hexValue=[01, 01, F8, 01, 05, 58, 59, 49, DD, 00]]

dariuszseweryn, 2021-04-01:
Thing is that you do not see this packet in the logs and the logs are put at the very first place the lib has access to the data hence the OS does not deliver it to the library/app at all. In my experience I have not seen missing notifications because of the OS bug at all — I have seen quite a lot missing notifications because of a peripheral firmware bug.

Btw. you cannot compare iOS and Android, these are different OSes and libraries there.

VinnyScorsatto, 2021-04-01:
Yeah.. agree, I have been doing a lot of tests and that is my perception as
well, thanks!!